BiancoRoyal/node-red-contrib-modbus #156: Modbus Read Node not working on Buster

braunku, 2019-10-09:
**I'm submitting a bug ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 modbus read pallete node does not work.  debug attached to either output pin does not trigger any flow data.  On previous versions a read block with debug will show a message for every read.  This is not the case.  Verified with two different Pi's running same software as decribed below (very stock installations)

 **Minimal reproduction of the problem with instructions**
 
   1.  Install fresh raspbian lite buster on rPi4
   2.  install nodeRED using the install script from NR website install instructions
   3. install modbus pallete using pallete manager
   4. configure modbus read node for slave server
   4. Connect the modbus read to a debug node and deploy
   5.  no output on debug, regardless of polling rate.

   
### Your Environment
**Please tell us about your environment:**

modbus TCP
  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x) Raspbian:  buster latest version linux 4.19.75
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : tried multiple, WAGO 352 coupler & modbus server in modbus pallete
     - Version: latest version 4.1.3
     - Vendor : you
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``
 4.1.3

* **Node:** (min. 6.x)
   `node --version` = 10.16.3, NPM 6.12.0

example flow:
[{"id":"3746ea60.e58016","type":"modbus-read","z":"8627d34b.a0a74","name":"","topic":"","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"5","rate":"100","rateUnit":"ms","delayOnStart":false,"startDelayTime":"5","server":"2e8d07cd.c09308","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":210,"y":460,"wires":[["726c99bb.a5c448"],[]]},{"id":"726c99bb.a5c448","type":"debug","z":"8627d34b.a0a74","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":450,"y":460,"wires":[]},{"id":"2e8d07cd.c09308","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.4.20","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]

biancode, 2019-10-09:
Hi Kurt, Node-RED v1.x?

braunku, 2019-10-09:
Yes, Node-RED core 1.0.1 
I installed about 2 hours ago. Oct 9, 2019 10:15am PST

biancode, 2019-10-09:
It is not v1.x compatible see #144

closed 2019-10-09 by biancode

braunku, 2019-10-09:
I did not realize that sorry.  Thank you!