BiancoRoyal/node-red-contrib-modbus #157: Modbus Server Error

luismiguelcelda, 2019-10-24:
**I'm submitting a Problem

## Problem
 - ( ) bug report =>Crash node-red when modbus server lost ip
 - ( ) unexpected behavior => node-red down
 
 **Current behavior**
I have a Modbus server on node-red in rpi. When I disconnect the cable of ethernet the node-red crash. The same behaviour occurs when the IP don't correspond to the ip of the device (rpi).
 
 **Expected behavior**
I expected that the communication was off, but not, node-red crashed.
 
 **Minimal reproduction of the problem with instructions**
   1. Disconnect ethernet 
   2. or use a different ip for modbus server than ip of device.
   
### Your Environment
Node red with modbus server and serveral modbus-flex-write. I use a ModScan like a client in a windows pc. Works fine, until I detach the cable.

  * ( x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS:  
    - ( ) Linux: x 
    - ( ) Windows:  
    - ( ) Raspbian: x

biancode, 2019-11-17:
Hi @luismiguelcelda could you provide some logs of that error or crash, please?

biancode, 2019-12-08:
please, test with v5.2.x and add some logs

biancode, 2020-03-04:
no response - reopen if there is more, please

closed 2020-03-04 by biancode