BiancoRoyal/node-red-contrib-modbus #160: Modbus TCP: "PortNotOpenError: Port Not Open"

mikeS7, 2019-12-05:
**I'm submitting a ...** 

## Problem
 - (x) unexpected behavior
 
 **Current behavior**
 
 My device doesn't maintain permanent TCP connection and close after 1.5 second or so. Single `FC 3` request rise the list of errors:
`TypeError: Cannot read property 'getMachineState' of undefined`
`PortNotOpenError: Port Not Open`
`FSM Reset On State [object Object]`
`Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`FSM Reset On State [object Object]`
`Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`

and log from console:
`5 Dec 09:08:04 - [error] [modbus-flex-getter:Shtyl] TypeError: Cannot read property 'getMachineState' of undefined`
`    at Object._inputCallback (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-flex-getter.js:123:61)`
`    at Object.Node._emitInput (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:199:18)`
`    at Immediate._onImmediate (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:179:33)`
`    at runCallback (timers.js:705:18)`
`    at tryOnImmediate (timers.js:676:5)`
`    at processImmediate (timers.js:658:5)`
`5 Dec 09:08:04 - [error] [modbus-flex-getter:Shtyl] PortNotOpenError: Port Not Open`
`    at ModbusRTU.writeFC4 (/data/node_modules/modbus-serial/index.js:501:24)`
`    at ModbusRTU.writeFC3 (/data/node_modules/modbus-serial/index.js:487:10)`
`    at /data/node_modules/modbus-serial/apis/promise.js:48:31`
`    at new Promise (<anonymous>)`
`    at ModbusRTU.readHoldingRegisters (/data/node_modules/modbus-serial/apis/promise.js:39:27)`
`    at Object.noneCommandSent [as callModbus] (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:550:7)`
`    at Object.sequentialDequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:140:13)`
`    at Timeout.r.dequeueCommand [as _onTimeout] (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:172:14)`
`    at ontimeout (timers.js:436:11)`
`    at tryOnTimeout (timers.js:300:5)`
`    at listOnTimeout (timers.js:263:5)`
`    at Timer.processTimers (timers.js:223:10)`
`5 Dec 09:08:04 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]`
`5 Dec 09:08:04 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`5 Dec 09:08:04 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]`
`5 Dec 09:08:04 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`5 Dec 09:08:05 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`5 Dec 09:08:05 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`5 Dec 09:08:05 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`

no response for request, request is lost. The only workaround, constantly send requests, in that case only first one is lost.

 **Expected behavior**
 
 Process a request. If port is closed, should open connection and process a request.

 **Minimal reproduction of the problem with instructions**
 
   1. Create simple flow with Getter or FlexGetter node with TCP connection.
   2. Device should close connection.
   3. Send FC 3 request
in result list of errors and no response.

### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
* **node-red-contrib-modbus version:** `5.1.0`

* **Node:** (min. 6.x)
   `node --version` = `v10.17.0`
Node-RED 1.0.3 docker on raspberry PI

mikeS7, 2019-12-07:
I tried the fix, doesn't help. List of errors and no answer to request.
Log after d0d3b23 commit:
`7 Dec 12:08:55 - [error] [modbus-flex-getter:Shtyl] PortNotOpenError: Port Not Open`
`    at ModbusRTU.writeFC4 (/data/node_modules/modbus-serial/index.js:501:24)`
`    at ModbusRTU.writeFC3 (/data/node_modules/modbus-serial/index.js:487:10)`
`    at /data/node_modules/modbus-serial/apis/promise.js:48:31`
`    at new Promise (<anonymous>)`
`    at ModbusRTU.converted [as readHoldingRegisters] (/data/node_modules/modbus-serial/apis/promise.js:39:27)`
`    at Object.ModbusClientNode.node.readModbus [as callModbus] (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:571:25)`
`    at ModbusClientNode.node.sequentialDequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:141:21)`
`    at Timeout.ModbusClientNode.node.dequeueCommand [as _onTimeout] (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:172:14)`
`    at ontimeout (timers.js:436:11)`
`    at tryOnTimeout (timers.js:300:5)`
`    at listOnTimeout (timers.js:263:5)`
`    at Timer.processTimers (timers.js:223:10)`
`7 Dec 12:08:55 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]`
`7 Dec 12:08:55 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`7 Dec 12:08:55 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]`
`7 Dec 12:08:55 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`7 Dec 12:08:55 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`7 Dec 12:08:55 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`
`7 Dec 12:08:55 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1`

biancode, 2019-12-07:
I'll give a fix to that in the next few days. That is a bug with the new state machine and it was not to see by testing. I have to add more tests here to take care about that problems with the FSM.

mikeS7, 2019-12-07:
Same error with previous FSM, readModbus should catch PortNotOpenError

biancode, 2019-12-07:
@mikeS7 could you add some logs, please? https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG

DEBUG=contribModbus*,modbus-serial

mikeS7, 2019-12-07:
@biancode done:
2019-12-07T10:22:18.090Z contribModbus:core:client Client -> new
2019-12-07T10:22:18.093Z contribModbus:core:client Client -> initialized
2019-12-07T10:22:19.200Z contribModbus:core:client Client -> init
7 Dec 13:22:19 - [warn] [modbus-client:Shtyl-dacha] Client -> reconnect in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1
7 Dec 13:22:19 - [info] Started flows
2019-12-07T10:22:19.823Z contribModbus:core:client Client -> TCP port TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.917Z modbus-serial TCP port: signal connect
2019-12-07T10:22:19.921Z contribModbus:core:client Client -> connected TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.923Z contribModbus:core:client Client -> activated TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.924Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.964Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.967Z contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:19.968Z contribModbus:core:client Client -> empty TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:22:21.392Z modbus-serial TCP port: signal close: false
2019-12-07T10:23:00.294Z contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"unitid":1,"fc":3,"address":501,"quantity":1,"messageId":"5deb7d8403d2563971aee44a"},"unit":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.295Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.297Z contribModbus:core:client Client -> {"info":"queue read msg","message":{"unitid":1,"fc":3,"address":501,"quantity":1,"messageId":"5deb7d8403d2563971aee44a"},"state":"empty","queueLength":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.299Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.300Z contribModbus:core:client Client -> {"type":"queue check","unitid":1,"sendAllowed":true,"queueLength":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.301Z contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":1,"queueLength":0,"sendAllowedForNext":false,"delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.305Z contribModbus:core:client Client -> {"info":"read msg","message":{"unitid":1,"fc":3,"address":501,"quantity":1,"messageId":"5deb7d8403d2563971aee44a"},"unitid":1,"timeout":1000,"state":"queueing"} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.314Z contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"unitid":1,"fc":3,"address":501,"quantity":1,"messageId":"5deb7d8403d2563971aee44a"}} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.315Z contribModbus:core:client Client -> activated TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.316Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.318Z contribModbus:flex:getter Port Not Open
7 Dec 13:23:00 - [error] [modbus-flex-getter:Shtyl] PortNotOpenError: Port Not Open
    at ModbusRTU.writeFC4 (/data/node_modules/modbus-serial/index.js:501:24)
    at ModbusRTU.writeFC3 (/data/node_modules/modbus-serial/index.js:487:10)
    at /data/node_modules/modbus-serial/apis/promise.js:48:31
    at new Promise (<anonymous>)
    at ModbusRTU.converted [as readHoldingRegisters] (/data/node_modules/modbus-serial/apis/promise.js:39:27)
    at Object.ModbusClientNode.node.readModbus [as callModbus] (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:571:25)
    at ModbusClientNode.node.sequentialDequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:141:21)
    at Timeout.ModbusClientNode.node.dequeueCommand [as _onTimeout] (/data/node_modules/node-red-contrib-modbus/modbus/modbus-client.js:172:14)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
2019-12-07T10:23:00.353Z contribModbus:core:client Client -> failed TCP@192.168.77.137:502 default Unit-Id: 1
7 Dec 13:23:00 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]
2019-12-07T10:23:00.357Z contribModbus:core:client Client -> broken TCP@192.168.77.137:502 default Unit-Id: 1
7 Dec 13:23:00 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.360Z contribModbus:core:client Client -> closed TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.362Z contribModbus:core:client Client -> broken TCP@192.168.77.137:502 default Unit-Id: 1
7 Dec 13:23:00 - [warn] [modbus-client:Shtyl-dacha] Client -> try to reconnect by init in 100 ms TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:00.371Z modbus-serial modbusErrorHandling:Port Not Open
2019-12-07T10:23:00.372Z contribModbus:core:client Client -> failed TCP@192.168.77.137:502 default Unit-Id: 1
7 Dec 13:23:00 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State [object Object]
[... cut ...]
2019-12-07T10:23:01.698Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:01.699Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:01.700Z contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:01.701Z contribModbus:core:client Client -> empty TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-07T10:23:03.216Z modbus-serial TCP port: signal close: false

biancode, 2019-12-07:
From the log, it should be OK to reconnect on the first request after connection gets lost, but if it gets no answer while timeout, then it reconnects again. It looks like your port is not ready to connect. What is the timeout value?

mikeS7, 2019-12-07:
Timeout 1 sec, tried 2 sec without success

mikeS7, 2019-12-08:
Unfortunately issue is not  solved in 5.2.0. Tomorrow will provide logs and working example with modbus-serial logger.

biancode, 2019-12-09:
Please, check the port settings in docker and firewall
hint:
sudo usermod -a -G tty $username
sudo usermod -a -G dailout $username

mikeS7, 2019-12-09:
> sudo usermod -a -G tty $username
> sudo usermod -a -G dailout $username

What is relation to TCP?

mikeS7, 2019-12-09:
I modified logger.js example a bit to add timing to console output:
```javascript
/* eslint-disable no-console, spaced-comment */

// create an empty modbus client
var ModbusRTU = require("modbus-serial");
// var ModbusRTU = require("../index");
var client = new ModbusRTU();
var timeoutRunRef = null;
var timeoutConnectRef = null;

var networkErrors = [
    "ESOCKETTIMEDOUT",
    "ETIMEDOUT",
    "ECONNRESET",
    "ECONNREFUSED",
    "EHOSTUNREACH",
    "ENETRESET",
    "ECONNABORTED",
    "ENETUNREACH",
    "ENOTCONN",
    "ESHUTDOWN",
    "EHOSTDOWN",
    "ENETDOWN",
    "EWOULDBLOCK",
    "EAGAIN"
];

// check error, and reconnect if needed
function checkError(e) {
    if(e.errno && networkErrors.includes(e.errno)) {
        console.log(new Date().toISOString(), "we have to reconnect");

        // close port
        client.close();

        // re open client
        client = new ModbusRTU();
        timeoutConnectRef = setTimeout(connect, 0);
    }
}

// open connection to a serial port
function connect() {
    // clear pending timeouts
    clearTimeout(timeoutConnectRef);

    // if client already open, just run
    if (client.isOpen) {
        run();
    }

    // if client closed, open a new connection
    client.connectTCP("192.168.77.137", { port: 502 })
        .then(setClient)
        .then(function() {
            console.log(new Date().toISOString(), "Connected"); })
        .catch(function(e) {
            checkError(e);
            console.log(new Date().toISOString(), e.message); });
}

function setClient() {
    // set the client's unit id
    // set a timout for requests default is null (no timeout)
    client.setID(1);
    client.setTimeout(1000);

    // run program
    run();
}

function run() {
    // clear pending timeouts
    clearTimeout(timeoutRunRef);

    // read the 4 registers starting at address 5
    client.readHoldingRegisters(200, 7)
        .then(function(d) {
            console.log(new Date().toISOString(), "Receive:", d.data);
            })
        .then(function() {
            timeoutRunRef = setTimeout(run, 1000); })
        .catch(function(e) {
            checkError(e);
            console.log(new Date().toISOString(), e.message); });
}

// connect and start logging
connect();
```

mikeS7, 2019-12-09:
Output with 1000 msec timeout:
```
/usr/src/app # node logger.js
2019-12-09T18:05:12.832Z Connected
2019-12-09T18:05:13.027Z Receive: [ 2375, 2381, 2393, 9, 8, 8, 501 ]
2019-12-09T18:05:14.215Z Receive: [ 2374, 2380, 2394, 9, 9, 8, 501 ]
2019-12-09T18:05:15.377Z Receive: [ 2374, 2383, 2395, 9, 9, 8, 501 ]
2019-12-09T18:05:16.545Z Receive: [ 2376, 2383, 2395, 9, 9, 8, 501 ]
2019-12-09T18:05:17.697Z Receive: [ 2373, 2380, 2392, 9, 9, 8, 501 ]
2019-12-09T18:05:18.864Z Receive: [ 2373, 2380, 2391, 9, 9, 8, 501 ]
2019-12-09T18:05:20.024Z Receive: [ 2373, 2380, 2391, 9, 9, 8, 501 ]
2019-12-09T18:05:21.176Z Receive: [ 2375, 2382, 2393, 9, 9, 8, 501 ]
2019-12-09T18:05:22.340Z Receive: [ 2371, 2375, 2387, 9, 9, 8, 501 ]
2019-12-09T18:05:23.504Z Receive: [ 2371, 2377, 2386, 9, 9, 8, 501 ]
2019-12-09T18:05:24.641Z Receive: [ 2368, 2376, 2387, 9, 9, 8, 501 ]
2019-12-09T18:05:25.790Z Receive: [ 2368, 2376, 2389, 9, 9, 8, 501 ]
2019-12-09T18:05:26.936Z Receive: [ 2371, 2383, 2392, 8, 8, 8, 501 ]
2019-12-09T18:05:28.028Z Receive: [ 2370, 2383, 2390, 8, 8, 8, 501 ]
2019-12-09T18:05:29.205Z Receive: [ 2373, 2386, 2394, 8, 8, 7, 501 ]
2019-12-09T18:05:30.360Z Receive: [ 2369, 2380, 2388, 8, 8, 7, 501 ]
2019-12-09T18:05:31.496Z Receive: [ 2372, 2382, 2390, 8, 8, 7, 501 ]
2019-12-09T18:05:32.656Z Receive: [ 2369, 2381, 2389, 8, 8, 7, 501 ]
2019-12-09T18:05:33.740Z Receive: [ 2372, 2381, 2390, 8, 8, 7, 501 ]
2019-12-09T18:05:34.890Z Receive: [ 2372, 2381, 2390, 8, 8, 7, 501 ]
2019-12-09T18:05:36.029Z Receive: [ 2371, 2383, 2391, 8, 8, 7, 501 ]
2019-12-09T18:05:37.177Z Receive: [ 2377, 2379, 2390, 8, 8, 7, 501 ]
2019-12-09T18:05:38.365Z Receive: [ 2379, 2380, 2393, 8, 8, 7, 501 ]
2019-12-09T18:05:39.446Z Receive: [ 2376, 2370, 2401, 8, 8, 7, 501 ]
2019-12-09T18:05:40.596Z Receive: [ 2378, 2369, 2402, 8, 8, 7, 501 ]
2019-12-09T18:05:41.602Z we have to reconnect
2019-12-09T18:05:41.606Z Port Not Open
2019-12-09T18:05:41.648Z Connected
2019-12-09T18:05:41.877Z Receive: [ 2376, 2369, 2401, 8, 8, 7, 501 ]
2019-12-09T18:05:43.040Z Receive: [ 2376, 2368, 2399, 8, 8, 7, 501 ]
2019-12-09T18:05:44.177Z Receive: [ 2377, 2368, 2399, 8, 8, 7, 501 ]
2019-12-09T18:05:45.348Z Receive: [ 2381, 2377, 2386, 8, 8, 7, 501 ]
2019-12-09T18:05:46.514Z Receive: [ 2379, 2375, 2394, 8, 8, 7, 501 ]
2019-12-09T18:05:47.669Z Receive: [ 2378, 2377, 2395, 8, 8, 7, 501 ]
2019-12-09T18:05:48.838Z Receive: [ 2381, 2377, 2397, 8, 8, 7, 501 ]
2019-12-09T18:05:50.000Z Receive: [ 2381, 2377, 2397, 8, 8, 7, 501 ]
```

mikeS7, 2019-12-09:
2000 msec timeout:
```
/usr/src/app # node logger.js
2019-12-09T18:08:00.678Z Connected
2019-12-09T18:08:00.874Z Receive: [ 2377, 2372, 2402, 8, 8, 7, 501 ]
2019-12-09T18:08:02.905Z we have to reconnect
2019-12-09T18:08:02.910Z Port Not Open
2019-12-09T18:08:02.982Z Connected
2019-12-09T18:08:03.241Z Receive: [ 2375, 2371, 2397, 8, 8, 7, 501 ]
2019-12-09T18:08:05.246Z we have to reconnect
2019-12-09T18:08:05.247Z Port Not Open
2019-12-09T18:08:05.291Z Connected
2019-12-09T18:08:05.529Z Receive: [ 2371, 2369, 2395, 8, 8, 7, 501 ]
2019-12-09T18:08:07.533Z we have to reconnect
2019-12-09T18:08:07.534Z Port Not Open
2019-12-09T18:08:07.592Z Connected
2019-12-09T18:08:07.794Z Receive: [ 2369, 2366, 2392, 8, 8, 7, 501 ]
```

mikeS7, 2019-12-09:
Node-Red settings:
[image]

[image]

simple payload:
```
{
    "fc": 3,
    "unitid": 1,
    "address": 200,
    "quantity": 7
}
```

I've got the following without actual response:
```
9 Dec 21:20:28 - [info] Stopping modified flows
9 Dec 21:20:28 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State {"value":"failed","actions":[],"changed":true}
9 Dec 21:20:28 - [warn] [modbus-client:Shtyl] Client -> try to reconnect by init in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:20:28 - [info] Stopped modified flows
9 Dec 21:20:28 - [info] Starting modified flows
9 Dec 21:20:28 - [warn] [modbus-client:Shtyl] Client -> reconnect in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:20:28 - [info] Started modified flows
9 Dec 21:20:32 - [error] [modbus-flex-getter:Shtyl] PortNotOpenError: Port Not Open
    at ModbusRTU.writeFC4 (/data/node_modules/modbus-serial/index.js:501:24)
    at ModbusRTU.writeFC3 (/data/node_modules/modbus-serial/index.js:487:10)
    at /data/node_modules/modbus-serial/apis/promise.js:48:31
    at new Promise (<anonymous>)
    at ModbusRTU.readHoldingRegisters (/data/node_modules/modbus-serial/apis/promise.js:39:27)
    at Object.callModbus (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-client-core.js:68:16)
    at Object.sendDataInParallel (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-queue-core.js:76:15)
    at Object.sequentialDequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-queue-core.js:32:4)
    at Object.dequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-queue-core.js:146:10)
    at Timeout._onTimeout (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:162:10)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
9 Dec 21:20:32 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State {"value":"failed","actions":[],"changed":true}
9 Dec 21:20:32 - [warn] [modbus-client:Shtyl] Client -> try to reconnect by init in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:20:32 - [warn] [modbus-flex-getter:Shtyl] FSM Reset On State {"value":"failed","actions":[],"changed":true}
9 Dec 21:20:32 - [warn] [modbus-client:Shtyl] Client -> try to reconnect by init in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:20:34 - [warn] [modbus-client:Shtyl] Client -> reconnect in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:20:34 - [warn] [modbus-client:Shtyl] Client -> reconnect in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
```

mikeS7, 2019-12-09:
single request with DEBUG=contribModbus*,modbus-serial:
```
9 Dec 21:31:49 - [warn] [modbus-client:Shtyl] Client -> try to reconnect by init in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1                                                                                                                                  [20/1959]
2019-12-09T18:31:49.811Z contribModbus:core:client Client -> closed TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:49.812Z contribModbus:core:client Client -> broken TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:31:49 - [warn] [modbus-client:Shtyl] Client -> try to reconnect by init in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:49.814Z contribModbus:core:client Client -> stopped TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:49.815Z contribModbus:core:client Client -> close node TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:31:49 - [info] Stopped modified flows
9 Dec 21:31:49 - [info] Starting modified flows
2019-12-09T18:31:49.834Z contribModbus:core:client Client -> new
2019-12-09T18:31:49.836Z contribModbus:core:client Client -> initialized
2019-12-09T18:31:49.863Z contribModbus:core:client Client -> init
9 Dec 21:31:49 - [warn] [modbus-client:Shtyl] Client -> reconnect in 2000 ms TCP@192.168.77.137:502 default Unit-Id: 1
9 Dec 21:31:49 - [info] Started modified flows
2019-12-09T18:31:50.365Z contribModbus:core:client Client -> TCP port TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.418Z modbus-serial TCP port: signal connect
2019-12-09T18:31:50.420Z contribModbus:core:client Client -> connected TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.423Z contribModbus:core:client Client -> activated TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.424Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.427Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.428Z contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:50.429Z contribModbus:core:client Client -> empty TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:51.812Z contribModbus:core:client Client -> stopped TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:51.815Z contribModbus:core:client Client -> stopped TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:51.918Z modbus-serial TCP port: signal close: false
2019-12-09T18:31:53.090Z contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"unitid":1,"fc":3,"address":200,"quantity":7,"messageId":"5dee9319876ec0aedcbb8f5c"},"unit":1,"sendingListLength":0} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.092Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.094Z contribModbus:core:client Client -> {"info":"queue read msg","message":{"unitid":1,"fc":3,"address":200,"quantity":7,"messageId":"5dee9319876ec0aedcbb8f5c"},"state":"empty","queueLength":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.095Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.098Z contribModbus:core:client Client -> {"type":"queue check","unitid":1,"sendAllowed":true,"queueLength":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.099Z contribModbus:core:client Client -> {"type":"serial sending and wait per unitid","unitid":1,"queueLength":0,"sendAllowedForNext":false,"delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.103Z contribModbus:core:client Client -> {"info":"read msg","message":{"unitid":1,"fc":3,"address":200,"quantity":7,"messageId":"5dee9319876ec0aedcbb8f5c"},"unitid":1,"timeout":1000,"state":"queueing"} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.111Z contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"unitid":1,"fc":3,"address":200,"quantity":7,"messageId":"5dee9319876ec0aedcbb8f5c"}} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.112Z contribModbus:core:client Client -> activated TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.113Z contribModbus:core:client Client -> queueing TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:53.115Z contribModbus:flex:getter Port Not Open
9 Dec 21:31:53 - [error] [modbus-flex-getter:Shtyl] PortNotOpenError: Port Not Open
    at ModbusRTU.writeFC4 (/data/node_modules/modbus-serial/index.js:501:24)
    at ModbusRTU.writeFC3 (/data/node_modules/modbus-serial/index.js:487:10)
    at /data/node_modules/modbus-serial/apis/promise.js:48:31
    at new Promise (<anonymous>)
    at ModbusRTU.readHoldingRegisters (/data/node_modules/modbus-serial/apis/promise.js:39:27)
    at Object.callModbus (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-client-core.js:68:16)
    at Object.sendDataInParallel (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-queue-core.js:76:15)
    at Object.sequentialDequeueCommand (/data/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-queue-core.js:32:4)
[... cut ...]
2019-12-09T18:31:58.672Z contribModbus:core:client Client -> {"state":"queueing","message":"dequeue command tcp","delay":1} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:58.674Z contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:31:58.675Z contribModbus:core:client Client -> empty TCP@192.168.77.137:502 default Unit-Id: 1
2019-12-09T18:32:00.179Z modbus-serial TCP port: signal close: false
```

mikeS7, 2019-12-09:
I'm ready to test any other cases

biancode, 2019-12-10:
@yaacov what do you think?

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode

xale12, 2020-04-01:
Hello, I am a new guy at here with the same issue.  I try to connect the modbus device with FC 3 but It is not connected . Can I get any help here .

tjunussov, 2020-05-31:
have same issue (using TCP connection not TTY), current version of node-red-contrib-modbus is 5.10.1
After some period of time ( maybe 2-4 hours ) of work it suddenly stops to connect to Modbus TCP Server, claiming what **"Modbus Failure On State sending Get More About It By Logging"** and after **'Port Not Open'**

biancode, 2020-05-31:
@tjunussov try v5.13.2 or above, please

winkelermatthias, 2020-06-17:
I am on v5.13.2 and I get  "Modbus Failure On State sending Get More About It By Logging" and after 'Port Not Open'. Strangely if I just open a webpage to nodered and trigger the modbus write from my webpage(mqtt into nodered) again then it will work. An hour later the port not open issue appears again.

pyrler, 2022-02-25:
Hi,

same Problem ModBus TCP, V 5.21.2

is it better to use a few modbus write nodes?

biancode, 2022-02-26:
Hi, some devices will close the Port if the communication is to slow. Why should the queue work with a delay per message of 1 sec. ?  Please, check the Modbus Specs or the information of the responding Modbus device in those cases.

biancode, 2022-02-26:
> Hi,
> 
> same Problem ModBus TCP, V 5.21.2
> 
> is it better to use a few modbus write nodes?

I use one read and one write if the device has enough possible connections for TCP.

Naim-Bijapure, 2022-06-07:
hi, I am also getting a port not found error on the TCP connection. 

actually, we are fetching plc data every 1 minute. the issue is  coming  after 6-8 hours. I am getting port not found error . 
and after node server restart it starts again !

is there any solution for this? 
i am on version   8.0.5

@biancode  please help

commit af27fe61 (2019-12-05): fix: #160 error on show activities true