BiancoRoyal/node-red-contrib-modbus #161: Issue with multiple modbus rtu slaves

joearkay, 2019-12-05:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 

Setting up multiple modbus rtu/rtu buffered slaves that utilise the same serial port cause issues. Setting up one at a time and testing works fine, which confirms there isn't an issue with main codebase or hardware setup. More info here, https://discourse.nodered.org/t/multiple-modbus-rtu-slaves-causes-error/18783/8
 
 **Expected behavior**
 

Multiple slaves should be accessible via the same serial port. Commands send concurrently should be queued per server, as not to choke up the comms bus. The same is true for initialisation, because as far as my tests show, these all try and happen at the same time, causing none of the servers to initialise.  
 
 **Minimal reproduction of the problem with instructions**
 

Hard to reproduce if you don't have the same hardware setup. However there are two modbus slaves on each of the two node red tabs copied below. Disabling one tab and deploying allows the other connection to work fine. Enabling both causes the error condition above, proving that it's an issue only with multiple modbus slaves on the same serial port. 

Flow 2A: 

`[
    {
        "id": "ae505081.3307",
        "type": "modbus-read",
        "z": "5484bbf1.3f12e4",
        "name": "digi",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "unitid": "",
        "dataType": "HoldingRegister",
        "adr": "14",
        "quantity": "1",
        "rate": "5",
        "rateUnit": "s",
        "delayOnStart": true,
        "startDelayTime": "1",
        "server": "7a2d3f7d.23a34",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 170,
        "y": 100,
        "wires": [
            [
                "d6f6caad.6f5908"
            ],
            []
        ]
    },
    {
        "id": "d6f6caad.6f5908",
        "type": "debug",
        "z": "5484bbf1.3f12e4",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 400,
        "y": 100,
        "wires": []
    },
    {
        "id": "7a2d3f7d.23a34",
        "type": "modbus-client",
        "z": "",
        "name": "Novus DigiRail 2A",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/tty.usbserial-14130",
        "serialType": "RTU",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "even",
        "serialConnectionDelay": "1000",
        "unit_id": "246",
        "commandDelay": "1000",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    }
]`
   

Flow 4C: 

`[
    {
        "id": "6265f722.682958",
        "type": "modbus-read",
        "z": "9e57cfa0.6fc53",
        "name": "rail",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "unitid": "",
        "dataType": "HoldingRegister",
        "adr": "512",
        "quantity": "1",
        "rate": "7",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "1ee5d4d5.ce5dab",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 230,
        "y": 100,
        "wires": [
            [
                "3bc7006b.b025f"
            ],
            []
        ]
    },
    {
        "id": "3bc7006b.b025f",
        "type": "debug",
        "z": "9e57cfa0.6fc53",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 470,
        "y": 100,
        "wires": []
    },
    {
        "id": "1ee5d4d5.ce5dab",
        "type": "modbus-client",
        "z": "",
        "name": "DigiRail 4C",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/tty.usbserial-14130",
        "serialType": "RTU",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "even",
        "serialConnectionDelay": "100",
        "unit_id": "247",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    }
]`

### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * (X) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - (X) macOS: 10.14.6
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other: Balena
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Novus
     - Version: USB-i485
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** `5.1.0`

* **Node:** (min. 6.x)
   `node --version` =

joearkay, 2019-12-05:
@biancode - in ref to our conversation on the Node-RED forums

joearkay, 2019-12-05:
FYI - I think I've found a workaround with this. If you leave the unit-id blank in the 'server' section but instead specify in the modbus-read node, all seems to work ok...

`[{"id":"9aabcd68.0d86e","type":"tab","label":"2A","disabled":false,"info":""},{"id":"cb189f69.91d6e","type":"tab","label":"4C","disabled":false,"info":""},{"id":"589ed34a.a517ac","type":"modbus-client","z":"","name":"Novus","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/tty.usbserial-14120","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"even","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"c4287c00.4379f","type":"modbus-read","z":"9aabcd68.0d86e","name":"DigiRail 2A","topic":"","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"246","dataType":"HoldingRegister","adr":"0","quantity":"1","rate":"1","rateUnit":"m","delayOnStart":false,"startDelayTime":"","server":"589ed34a.a517ac","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":140,"y":80,"wires":[["ed7b1f4b.7136c"],[]]},{"id":"ed7b1f4b.7136c","type":"debug","z":"9aabcd68.0d86e","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":300,"y":80,"wires":[]},{"id":"b87bf5b1.d55858","type":"modbus-read","z":"cb189f69.91d6e","name":"DigiRail 4C","topic":"","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"247","dataType":"HoldingRegister","adr":"0","quantity":"1","rate":"1","rateUnit":"m","delayOnStart":false,"startDelayTime":"","server":"589ed34a.a517ac","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":120,"y":60,"wires":[["69f0a8f4.f3d758"],[]]},{"id":"69f0a8f4.f3d758","type":"debug","z":"cb189f69.91d6e","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":300,"y":60,"wires":[]}]`

joearkay, 2019-12-05:
I have a slight workaround which I'm testing so far. The first time I deployed it I needed to do a hard restart of Node-RED. I'm using one modbus-getter and another modbus-getter (with the input that can be triggered by an inject node). It seems by staggering the timings of these, the initialisations happen one after the other, which causes all other commands to work smoothly. There's still a bug that needs addressing, but this further confirms the issue: 
`[
    {
        "id": "9aabcd68.0d86e",
        "type": "tab",
        "label": "2A",
        "disabled": false,
        "info": ""
    },
    {
        "id": "cb189f69.91d6e",
        "type": "tab",
        "label": "4C",
        "disabled": false,
        "info": ""
    },
    {
        "id": "589ed34a.a517ac",
        "type": "modbus-client",
        "z": "",
        "name": "Novus",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/tty.usbserial-14120",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "even",
        "serialConnectionDelay": "100",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    },
    {
        "id": "c4287c00.4379f",
        "type": "modbus-read",
        "z": "9aabcd68.0d86e",
        "name": "DigiRail 2A",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "unitid": "246",
        "dataType": "HoldingRegister",
        "adr": "0",
        "quantity": "30",
        "rate": "5",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "589ed34a.a517ac",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 120,
        "y": 80,
        "wires": [
            [
                "ed7b1f4b.7136c"
            ],
            []
        ]
    },
    {
        "id": "ed7b1f4b.7136c",
        "type": "debug",
        "z": "9aabcd68.0d86e",
        "name": "2A",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "x": 280,
        "y": 80,
        "wires": []
    },
    {
        "id": "69f0a8f4.f3d758",
        "type": "debug",
        "z": "cb189f69.91d6e",
        "name": "4C",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "x": 510,
        "y": 60,
        "wires": []
    },
    {
        "id": "e7b9f068.634ae",
        "type": "modbus-getter",
        "z": "cb189f69.91d6e",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "logIOActivities": false,
        "unitid": "247",
        "dataType": "HoldingRegister",
        "adr": "0",
        "quantity": "30",
        "server": "589ed34a.a517ac",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 320,
        "y": 60,
        "wires": [
            [
                "69f0a8f4.f3d758"
            ],
            []
        ]
    },
    {
        "id": "fcc1dcf1.05e85",
        "type": "inject",
        "z": "cb189f69.91d6e",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "5",
        "crontab": "",
        "once": true,
        "onceDelay": "2",
        "x": 130,
        "y": 60,
        "wires": [
            [
                "e7b9f068.634ae"
            ]
        ]
    }
]`

joearkay, 2019-12-05:
Lets see if this works with a third modbus slave...

joearkay, 2019-12-05:
I've left it set up with the two units and it goes back into the "Initialise -> reconnecting after 2000msec" loop...

joearkay, 2019-12-05:
@biancode I've done about as much testing as I can - let me know when you have something for me to test. Thanks.

joearkay, 2019-12-05:
It seems like a flex getter may be a workaround, as it gives you the ability to ensure that there are no requests being sent at the same time. Still, better queueing of use of the serial bus (perhaps FIFO) would be better, to abstract this issue from the user. 

See working example, where there is a 10 second delay on the first command to allow both to run at a period of 1 minute, but our of sync. 

`[
    {
        "id": "c08717cf.3f6758",
        "type": "modbus-flex-getter",
        "z": "730119a0.222898",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "logIOActivities": false,
        "server": "589ed34a.a517ac",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 600,
        "y": 180,
        "wires": [
            [
                "d243661a.c27368"
            ],
            []
        ]
    },
    {
        "id": "1cd25a5e.bc12b6",
        "type": "function",
        "z": "730119a0.222898",
        "name": "2A",
        "func": "msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 246, 'address': 0 , 'quantity': 1 };\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 350,
        "y": 120,
        "wires": [
            [
                "c08717cf.3f6758",
                "627ef0bb.2e23f"
            ]
        ]
    },
    {
        "id": "b2dffc0c.422a3",
        "type": "function",
        "z": "730119a0.222898",
        "name": "4C",
        "func": "msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 247, 'address': 0 , 'quantity': 1 };\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 350,
        "y": 280,
        "wires": [
            [
                "c08717cf.3f6758",
                "627ef0bb.2e23f"
            ]
        ]
    },
    {
        "id": "106caa7e.f9b3d6",
        "type": "inject",
        "z": "730119a0.222898",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "60",
        "crontab": "",
        "once": true,
        "onceDelay": "10",
        "x": 210,
        "y": 120,
        "wires": [
            [
                "1cd25a5e.bc12b6"
            ]
        ]
    },
    {
        "id": "d243661a.c27368",
        "type": "debug",
        "z": "730119a0.222898",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 860,
        "y": 200,
        "wires": []
    },
    {
        "id": "627ef0bb.2e23f",
        "type": "debug",
        "z": "730119a0.222898",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 680,
        "y": 340,
        "wires": []
    },
    {
        "id": "d51b1428.93ddd8",
        "type": "inject",
        "z": "730119a0.222898",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "60",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 210,
        "y": 280,
        "wires": [
            [
                "b2dffc0c.422a3"
            ]
        ]
    },
    {
        "id": "589ed34a.a517ac",
        "type": "modbus-client",
        "z": "",
        "name": "Novus",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/tty.usbserial-14120",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "even",
        "serialConnectionDelay": "100",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    }
]`

joearkay, 2019-12-05:
Alternatively, you can rate limit a flex getter which prevents 'collisions': 

`[
    {
        "id": "c08717cf.3f6758",
        "type": "modbus-flex-getter",
        "z": "730119a0.222898",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "logIOActivities": false,
        "server": "589ed34a.a517ac",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 860,
        "y": 180,
        "wires": [
            [
                "d243661a.c27368"
            ],
            []
        ]
    },
    {
        "id": "1cd25a5e.bc12b6",
        "type": "function",
        "z": "730119a0.222898",
        "name": "2A",
        "func": "msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 246, 'address': 0 , 'quantity': 1 };\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 350,
        "y": 120,
        "wires": [
            [
                "627ef0bb.2e23f",
                "ab2d10f4.dd6ae"
            ]
        ]
    },
    {
        "id": "b2dffc0c.422a3",
        "type": "function",
        "z": "730119a0.222898",
        "name": "4C",
        "func": "msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 247, 'address': 0 , 'quantity': 1 };\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 350,
        "y": 280,
        "wires": [
            [
                "627ef0bb.2e23f",
                "ab2d10f4.dd6ae"
            ]
        ]
    },
    {
        "id": "d243661a.c27368",
        "type": "debug",
        "z": "730119a0.222898",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 1170,
        "y": 180,
        "wires": []
    },
    {
        "id": "627ef0bb.2e23f",
        "type": "debug",
        "z": "730119a0.222898",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 370,
        "y": 200,
        "wires": []
    },
    {
        "id": "d51b1428.93ddd8",
        "type": "inject",
        "z": "730119a0.222898",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "15",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "x": 130,
        "y": 180,
        "wires": [
            [
                "b2dffc0c.422a3",
                "1cd25a5e.bc12b6"
            ]
        ]
    },
    {
        "id": "ab2d10f4.dd6ae",
        "type": "delay",
        "z": "730119a0.222898",
        "name": "",
        "pauseType": "rate",
        "timeout": "5",
        "timeoutUnits": "seconds",
        "rate": "1",
        "nbRateUnits": "5",
        "rateUnits": "second",
        "randomFirst": "1",
        "randomLast": "5",
        "randomUnits": "seconds",
        "drop": false,
        "x": 640,
        "y": 180,
        "wires": [
            [
                "c08717cf.3f6758"
            ]
        ]
    },
    {
        "id": "589ed34a.a517ac",
        "type": "modbus-client",
        "z": "",
        "name": "Novus",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/tty.usbserial-14120",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "even",
        "serialConnectionDelay": "100",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    }
]`

biancode, 2019-12-05:
The package needs an option/setting to do all with one queue if there is no gateway to handle multiple requests.

joearkay, 2019-12-05:
Yes I agree - something like this for each modbus client - an option to set whether there should be a queue, and if so of what type. 

[image]

biancode, 2019-12-05:
... and it needs an option at the client to not disconnect if a unit does not answer it should just drop the request and send a message to the output of the node about that drop.

joearkay, 2019-12-05:
Good idea

biancode, 2019-12-08:
please, test with v5.2.x

biancode, 2019-12-08:
[image]

biancode, 2019-12-08:
UnitId's in parallel has to be switched to off - you could also paly with reconnecting or not on timeoutes

joearkay, 2019-12-09:
I've tested this but am seeing exactly the same issues. Hardware setup is unchanged: 

Slave 1: 
Addr: 246
Port: novus usb-i485

Slave 2: 
Addr: 247
Port: novus usb-i485

novus usb-i485 is set up within Node-RED to point to the same Serial port (9600 8N1). It seems that both getters try and initialise at exactly the same time, using the same port, which causes the issue. 

[image]

joearkay, 2019-12-09:
[image]

gv11181989, 2019-12-11:
Hi,

I am currently stuck in a similar situation. I want to pull data from 13 slaves. I've added a delay of 100 milliseconds to every consecutive request. The flow works fine with 9 slaves. But if i add a 10 slaves it disconnects. 
Below is the code that worked for me, I've commented out the requests which makes the modbus to disconnect.

`[{"id":"bc4c0a36.57d488","type":"tab","label":"Flow 2","disabled":false,"info":""},{"id":"c55e8efd.8c457","type":"modbus-flex-getter","z":"bc4c0a36.57d488","name":"KVA","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"d2f89fb4.f5d7b","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":570,"y":40,"wires":[[],["b9092b3f.b6f9f8"]]},{"id":"e3efb736.162ec8","type":"function","z":"bc4c0a36.57d488","name":"Meters","func":"var adkva = 3900;\nvar adkvah = 3958;\nvar adkw = 3902;\nvar adkwh = 3960;\nvar admd = 3978;\nvar adpf = 3906;\nvar adpd = 3974;\nvar advolt = 3908;\nvar idplc = 1;\nvar idr301 = 2;\nvar idhwg = 3;\nvar idht = 4;\nvar idr201 = 5;\nvar idr101 = 6;\nvar idutility = 7;\nvar idlight = 8;\nvar idac1 = 9;\nvar idac2 = 10;\nvar idsc401 = 11;\nvar idr401 = 12;\nvar idpacking = 13;\nvar idstp = 14;\n\n\n\nhtkva= {\n    'fc': 3, \n    'unitid': idht, \n    'address': adkva, \n    'quantity': 2 \n}; \n\nhtkvah= {\n    'fc': 3, \n    'unitid': idht,\n    'address': adkvah, \n    'quantity': 2 ,\n\n};\n\nhtkw= {\n    'fc': 3, \n    'unitid': idht,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nhtkwh= {\n    'fc': 3, \n    'unitid': idht, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nhtmd= {\n    'fc': 3, \n    'unitid': idht,\n    'address': admd, \n    'quantity': 2 ,\n\n};\n\nhtpf= {\n    'fc': 3, \n    'unitid': idht, \n    'address': adpf, \n    'quantity': 2 ,\n\n};\n\nhtpd= {\n    'fc': 3, \n    'unitid': idht,\n    'address': adpd, \n    'quantity': 2 ,\n\n};\n\nhtvolt= {\n    'fc': 3, \n    'unitid': idht, \n    'address': advolt, \n    'quantity': 2 ,\n};\n\nac1kw= {\n    'fc': 3, \n    'unitid': idac1,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nac1kwh= {\n    'fc': 3, \n    'unitid': idac1, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nac2kw= {\n    'fc': 3, \n    'unitid': idac2,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nac2kwh= {\n    'fc': 3, \n    'unitid': idac2, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\nhwgkw= {\n    'fc': 3, \n    'unitid': idhwg,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nhwgkwh= {\n    'fc': 3, \n    'unitid': idhwg, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nlightkw= {\n    'fc': 3, \n    'unitid': idlight,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nlightkwh= {\n    'fc': 3, \n    'unitid': idlight, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\npackingkw= {\n    'fc': 3, \n    'unitid': idpacking,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\npackingkwh= {\n    'fc': 3, \n    'unitid': idpacking, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nr101kw= {\n    'fc': 3, \n    'unitid': idr101,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nr101kwh= {\n    'fc': 3, \n    'unitid': idr101, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nr201kw= {\n    'fc': 3, \n    'unitid': idr201,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nr201kwh= {\n    'fc': 3, \n    'unitid': idr201, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nr301kw= {\n    'fc': 3, \n    'unitid': idr301,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nr301kwh= {\n    'fc': 3, \n    'unitid': idr301, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nr401kw= {\n    'fc': 3, \n    'unitid': idr401,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nr401kwh= {\n    'fc': 3, \n    'unitid': idr401, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nsc401kw= {\n    'fc': 3, \n    'unitid': idsc401,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nsc401kwh= {\n    'fc': 3, \n    'unitid': idsc401, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nstpkw= {\n    'fc': 3, \n    'unitid': idstp,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nstpkwh= {\n    'fc': 3, \n    'unitid': idstp, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\nutilitykw= {\n    'fc': 3, \n    'unitid': idutility,\n    'address': adkw, \n    'quantity': 2 ,\n\n};\n\nutilitykwh= {\n    'fc': 3, \n    'unitid': idutility, \n    'address': adkwh, \n    'quantity': 2 ,\n\n};\n\n\na = {payload:r301kw};\nb = {payload:r301kwh,delay:100};\nc = {payload:hwgkw,delay: 200};\nd = {payload:hwgkwh,delay:300};\ne = {payload:htkva,delay:400};\nf = {payload:htkvah,delay:500};\ng = {payload:htkw,delay:600};\nh = {payload:htkwh,delay:700};\ni = {payload:htmd,delay:800};\nj = {payload:htpf,delay:900};\nk = {payload:htpd,delay:1000};\nl = {payload:htvolt,delay:1100};\nm = {payload:r201kw,delay:1200};\nn = {payload:r201kwh,delay:1300};\no = {payload:r101kw,delay:1400};\np = {payload:r101kwh,delay:1500};\nq = {payload:utilitykw,delay:1600};\nr = {payload:utilitykwh,delay:1700};\ns = {payload:lightkw,delay:1800};\nt = {payload:lightkwh,delay:1900};\nu = {payload:ac1kw,delay:2000};\nv = {payload:ac1kwh,delay:2100};\nw = {payload:ac2kw,delay:2200};\nx = {payload:ac2kwh,delay:2300};\n//y = {payload:sc401kw,delay:2400};\n//z = {payload:sc401kwh,delay:2500};\n//aa = {payload:r401kw,delay:2600};\n//ab = {payload:r401kwh,delay:2700};\n//ac = {payload:packingkw,delay:2800};\n//ad = {payload:packingkwh,delay:2900};\n//ae = { [... cut ...] },{"id":"d2f89fb4.f5d7b","type":"modbus-client","z":"","name":"ModbusRTU","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"COM4","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"even","serialConnectionDelay":"100","unit_id":"4","commandDelay":"100","clientTimeout":"2000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]`

gv11181989, 2019-12-11:
Hi,
So I've found out that there was a problem with my device. The 10 slave device was turned off. When i turned it on the flow worked without any issue.
So, the problem is modbus getting disconnected and get stuck in re-connection loop when ever it doesn't receive a reply from a slave.

biancode, 2019-12-11:
@gv11181989 In v5.2.0 you can switch off reconnect on timeout

closed 2019-12-14 by biancode

joearkay, 2019-12-17:
Morning @biancode - I had been away over the weekend so wasn't able to confirm that this resolved the problem. Did @gv11181989 's flow prove that the fix resolved the issue? Let me know if you need anything else.

biancode, 2019-12-18:
@gv11181989 does it fix your last point with v5.2.0 by switching off reconnect on timeout?

gv11181989, 2019-12-18:
> @gv11181989 does it fix your last point with v5.2.0 by switching off reconnect on timeout?

@biancode Hi,

Sorry for late reply,
Yes it resolved my issue. Thanks a lot for the help.
mfg.
Gourav

jcampino1, 2020-05-25:
Hi guys, I've been facing the same problem described above: can't connect to more than one device in the same serial port connection.

In my case I have two devices with 8N1 configuration and one with 8N2 configuration (unchangeable). Any ideas to solve this one? I've tried with node-red-contrib-serial-modbus but it doesn't work if I'm connected to the same bus, I can only read the 8N1 devices.

Thanks!

joearkay, 2020-05-25:
@jcampino1 I'll leave this up to the original author of the repo, however could I suggest you open a new issue as not related to the one above? 

At a quick glance, it sounds like you have two devices with differing configurations. I'm not expert, but I believe for modbus communication, all devices need the same config i.e. 8N1 etc

jcampino1, 2020-05-25:
Thanks for the advice and I'm gonna make another issue for the multi-configuration devices issue.

But I keep having problems to read the two devices with same configuration (leaving the other one out). Did you guys fix the problem of conecting to more than one device in the same bus? If so, can you specify how?

Thanks!

biancode, 2020-05-25:
@jcampino1 I tested with v5.13.1 and it works well to communicate with multiple Units over the same BUS (1 Port serial) - check that you do not have "parallel unitid" to on in the Modbus config node. If you need more help please open and fill out a new issue! ;)

kpm-at-hfi, 2020-06-01:
@biancode, being able to switch off reconnect on timeout is _extremely_ useful... thanks.

What does "UnitId's in parallel" do?

dhruvjhalani95, 2021-08-05:
Hi!
I checked this with v5.13.1 which @biancode suggested and it worked perfectly. However again with the newer version v5.14 the issue still persists

biancode, 2021-08-05:
> @biancode, being able to switch off reconnect on timeout is _extremely_ useful... thanks.
> 
> What does "UnitId's in parallel" do?

"UnitId' in parallel" works with a queue on each UnitId and sends them in parallel per cycle if possible.

The order of UnitIds depends on msgs.

Option off sending:
1
2
3
2
1
3

Option on sending:

1,2,3
2,1,3

shafikmuhereza, 2023-11-12:
i managed to go by this problem using below solution

[
    {
        "id": "fb73499b66b705c9",
        "type": "tab",
        "label": "Flow 4",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "0eb54c3a13e9a5b0",
        "type": "function",
        "z": "fb73499b66b705c9",
        "name": "function 25",
        "func": "msg.payload = { \n    value: msg.payload, \n    'fc': 3, 'unitid': 1, \n    'address': 0, \n    'quantity': 10 } \nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 150,
        "y": 100,
        "wires": [
            [
                "2d0357d13261c64a"
            ]
        ]
    },
    {
        "id": "743ab3603a44750f",
        "type": "inject",
        "z": "fb73499b66b705c9",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "1",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 130,
        "y": 160,
        "wires": [
            [
                "0eb54c3a13e9a5b0"
            ]
        ]
    },
    {
        "id": "2d0357d13261c64a",
        "type": "modbus-flex-getter",
        "z": "fb73499b66b705c9",
        "name": "",
        "showStatusActivities": false,
        "showErrors": true,
        "showWarnings": true,
        "logIOActivities": false,
        "server": "9b032cb12c86bdea",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 350,
        "y": 80,
        "wires": [
            [],
            [
                "429a8265dce886b5",
                "4e84aaece033c2a7"
            ]
        ]
    },
    {
        "id": "429a8265dce886b5",
        "type": "function",
        "z": "fb73499b66b705c9",
        "name": "function 26",
        "func": "msg.payload = { \n     'fc': 3, \n     'unitid': 2, \n     'address': 0,\n      'quantity': 10 \n      }\nreturn msg",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 350,
        "y": 180,
        "wires": [
            [
                "703f4bba32e3852f"
            ]
        ]
    },
    {
        "id": "703f4bba32e3852f",
        "type": "modbus-flex-getter",
        "z": "fb73499b66b705c9",
        "name": "",
        "showStatusActivities": false,
        "showErrors": true,
        "showWarnings": true,
        "logIOActivities": false,
        "server": "9b032cb12c86bdea",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 370,
        "y": 260,
        "wires": [
            [],
            [
                "3209b192c84e6f57",
                "4e84aaece033c2a7"
            ]
        ]
    },
    {
        "id": "3209b192c84e6f57",
        "type": "function",
        "z": "fb73499b66b705c9",
        "name": "function 27",
        "func": "msg.payload = { \n     'fc': 3, \n     'unitid': 3, \n     'address': 0,\n      'quantity': 10 \n      }\nreturn msg",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 350,
        "y": 340,
        "wires": [
            [
                "6f42cdb0aee59d89"
            ]
        ]
    },
    {
        "id": "6f42cdb0aee59d89",
        "type": "modbus-flex-getter",
        "z": "fb73499b66b705c9",
        "name": "",
        "showStatusActivities": false,
        "showErrors": true,
        "showWarnings": true,
        "logIOActivities": false,
        "server": "9b032cb12c86bdea",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 350,
        "y": 420,
        "wires": [
            [],
            [
                "4e84aaece033c2a7"
            ]
        ]
    },
    {
        "id": "4e84aaece033c2a7",
        "type": "debug",
        "z": "fb73499b66b705c9",
        "name": "debug 8",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 740,
        "y": 300,
        "wires": []
    },
    {
        "id": "9b032cb12c86bdea",
        "type": "modbus-client",
        "name": "",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "COM10",
        "serialType": "RTU",
        "serialBaudrate": "19200",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "100",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showWarnings": true,
        "showLogs": true
    }
]