BiancoRoyal/node-red-contrib-modbus #162: Read Node stopped working

PierreScerri, 2019-12-07:
**I'm submitting a ...**  (check one with "x")

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
Modbus Read node stopped working
 **Expected behavior**
 
 Gets readings from modes devices
 
### Your Environment
**Please tell us about your environment:**

RPi 3+ with Hassos 2.12. Hassio 0.102.3. Node-red add-on 5.0.5
SDM120 and SDM230 energy meters connected to Pi via USB.

```
 Add-on version: 5.0.5
 You are running the latest version of this add-on.
 System: HassOS 2.12  (armv7 / raspberrypi3)
 Home Assistant version: 0.102.3
 Supervisor version: 192
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] node-red.sh: executing... 
patching file nodes/ui_base.html
Hunk #1 succeeded at 1243 (offset 712 lines).
audited 200 packages in 6.917s
found 0 vulnerabilities
[cont-init.d] node-red.sh: exited 0.
[cont-init.d] user.sh: executing... 
[cont-init.d] user.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:09:46] INFO: Starting Node-RED...
> hassio-addon-node-red@ start /opt
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
7 Dec 10:09:51 - [info] 
Welcome to Node-RED
===================
7 Dec 10:09:51 - [info] Node-RED version: v1.0.3
7 Dec 10:09:51 - [info] Node.js  version: v10.16.3
7 Dec 10:09:51 - [info] Linux 4.14.98-v7 arm LE
7 Dec 10:09:53 - [info] Loading palette nodes
7 Dec 10:10:04 - [info] Dashboard version 2.14.0 started at /endpoint/ui
7 Dec 10:10:08 - [info] Settings file  : /etc/node-red/config.js
7 Dec 10:10:08 - [info] Context store  : 'default' [module=memory]
7 Dec 10:10:08 - [info] User directory : /config/node-red/
7 Dec 10:10:08 - [warn] Projects disabled : editorTheme.projects.enabled=false
7 Dec 10:10:08 - [info] Flows file     : /config/node-red/flows.json
7 Dec 10:10:08 - [info] Starting flows
7 Dec 10:10:09 - [info] Started flows
7 Dec 10:10:09 - [info] Server now running at http://127.0.0.1:46836/
[10:10:09] INFO: Starting NGinx...
nginx: [alert] detected a LuaJIT version which is not OpenResty's; many optimizations will be disabled and performance will be compromised (see https://github.com/openresty/luajit2 for OpenResty's LuaJIT or, even better, consider using the OpenResty releases from https://openresty.org/en/download.html)
7 Dec 10:10:11 - [info] [mqtt-broker:c28a2dad.22663] Connected to broker: mqtt://192.168.1.200:1883
7 Dec 10:10:13 - [info] [server:Home Assistant] WebSocket Connected to http://hassio/homeassistant
7 Dec 10:10:21 - [error] [modbus-read:SDM 230 Voltage] Timed out
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Voltage] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Current] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Power] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Energy Imp] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Energy Exp] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 120 Voltage] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 120 Current] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 120 Power] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 120 Energy Sol] FSM Reset On State [object Object]
7 Dec 10:10:21 - [error] [modbus-read:SDM 120 Voltage] Timed out
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Voltage] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Current] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Power] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Energy Imp] FSM Reset On State [object Object]
7 Dec 10:10:21 - [warn] [modbus-read:SDM 230 Energy Exp] FSM Reset On State [object Object]
```

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other: HASSOS on RPi3
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` = 5.1.0

biancode, 2019-12-07:
Hi, node-red-contrib-modbus version, please?

and your Node: (min. 6.x) node --version = 10.16.3

Please add some logs:
https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG

PierreScerri, 2019-12-07:
node-red-contrib-modbus version = 5.1.0

Don't know where to put the DEBUG command.  

Hassio has limited terminal commands.

PierreScerri, 2019-12-07:
Don't know if this helps:  Node-Red add-on set to debug level "info":

```
7 Dec 11:06:03 - [debug] Module: node-red-contrib-statistics 2.2.2
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-statistics
7 Dec 11:06:03 - [debug] Module: node-red-contrib-stoptimer 0.0.7
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-stoptimer
7 Dec 11:06:03 - [debug] Module: node-red-node-twitter 1.1.6
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-twitter
7 Dec 11:06:03 - [debug] Module: node-red-contrib-time-range-switch 0.6.3
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-time-range-switch
7 Dec 11:06:03 - [debug] Module: node-red-contrib-timecheck 1.1.0
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-timecheck
7 Dec 11:06:03 - [debug] Module: node-red-contrib-traffic 0.2.1
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-traffic
7 Dec 11:06:03 - [debug] Module: node-red-dashboard 2.19.1 *ignored due to local copy*
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-dashboard
7 Dec 11:06:03 - [debug] Module: node-red-node-base64 0.2.0
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-base64
7 Dec 11:06:03 - [debug] Module: node-red-node-email 1.7.3
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-email
7 Dec 11:06:03 - [debug] Module: node-red-node-feedparser 0.1.15
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-feedparser
7 Dec 11:06:03 - [debug] Module: node-red-node-geofence 0.1.2
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-geofence
7 Dec 11:06:03 - [debug] Module: node-red-node-msgpack 1.2.0
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-msgpack
7 Dec 11:06:03 - [debug] Module: node-red-node-pi-gpio 1.0.6
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-pi-gpio
7 Dec 11:06:03 - [debug] Module: node-red-node-ping 0.1.0
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-ping
7 Dec 11:06:03 - [debug] Module: node-red-node-random 0.1.4
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-random
7 Dec 11:06:03 - [debug] Module: node-red-node-rbe 0.2.6
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-rbe
7 Dec 11:06:03 - [debug] Module: node-red-node-sentiment 0.1.6
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-sentiment
7 Dec 11:06:03 - [debug] Module: node-red-node-serialport 0.10.0
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-serialport
7 Dec 11:06:03 - [debug] Module: node-red-node-smooth 0.1.2
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-smooth
7 Dec 11:06:03 - [debug] Module: node-red-node-suncalc 0.0.12
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-suncalc
7 Dec 11:06:03 - [debug] Module: node-red-node-tail 0.0.3
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-node-tail
7 Dec 11:06:03 - [debug] Module: node-red-contrib-sunevents 0.0.5
7 Dec 11:06:03 - [debug]         /opt/node_modules/node-red-contrib-sunevents
7 Dec 11:06:10 - [info] Dashboard version 2.14.0 started at /endpoint/ui
7 Dec 11:06:13 - [info] Settings file  : /etc/node-red/config.js
7 Dec 11:06:13 - [info] Context store  : 'default' [module=memory]
7 Dec 11:06:13 - [info] User directory : /config/node-red/
7 Dec 11:06:13 - [warn] Projects disabled : editorTheme.projects.enabled=false
7 Dec 11:06:13 - [info] Flows file     : /config/node-red/flows.json
7 Dec 11:06:13 - [info] Server now running at http://127.0.0.1:46836/
7 Dec 11:06:13 - [debug] loaded flow revision: e981dd1dbffdbbd983749a853774d4e2
7 Dec 11:06:13 - [debug] red/runtime/nodes/credentials.load : user provided key
7 Dec 11:06:13 - [debug] red/runtime/nodes/credentials.load : keyType=user
7 Dec 11:06:13 - [info] Starting flows
7 Dec 11:06:13 - [debug] red/nodes/flows.start : starting flow : global
7 Dec 11:06:13 - [debug] red/nodes/flows.start : starting flow : 922f2386.3a835
7 Dec 11:06:13 - [debug] red/nodes/flows.start : starting flow : bae2113e.d2105
7 Dec 11:06:13 - [debug] red/nodes/flows.start : starting flow : b4140577.2778e8
[11:06:13] INFO: Starting NGinx...
7 Dec 11:06:13 - [debug] [server:Home Assistant] instantiated node, name: Home Assistant
7 Dec 11:06:14 - [debug] [server-state-changed:Study Motion] instantiated node, name: Study Motion
7 Dec 11:06:14 - [debug] [server-state-changed:Pool Pump] instantiated node, name: Pool Pump
7 Dec 11:06:14 - [debug] [api-call-service:Study Light ON] instantiated node, name: Study Light ON
7 Dec 11:06:14 - [debug] [api-call-service:Study Light OFF] instantiated node, name: Study Light OFF
7 Dec 11:06:14 - [debug] [api-call-service:Tell Me] instantiated node, name: Tell Me
7 Dec 11:06:14 - [debug] [api-call-service:Garden Light ON] instantiated node, name: Garden Light ON
[... 9 lines cut ...]
7 Dec 11:06:18 - [debug] [server:Home Assistant] WebSocket Connecting http://hassio/homeassistant
7 Dec 11:06:18 - [debug] [server:Home Assistant] config server event listener connecting
7 Dec 11:06:18 - [info] [server:Home Assistant] WebSocket Connected to http://hassio/homeassistant
7 Dec 11:06:18 - [debug] [server:Home Assistant] config server event listener connected
7 Dec 11:06:26 - [error] [modbus-read:SDM 230 Voltage] Timed out
7 Dec 11:06:26 - [warn] [modbus-read:SDM 230 Voltage] FSM Reset On State [object Object]
[... cut ...]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 230 Energy Imp] FSM Reset On State [object Object]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 230 Energy Exp] FSM Reset On State [object Object]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 120 Voltage] FSM Reset On State [object Object]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 120 Current] FSM Reset On State [object Object]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 120 Power] FSM Reset On State [object Object]
7 Dec 11:06:26 - [warn] [modbus-read:SDM 120 Energy Sol] FSM Reset On State [object Object]
```

biancode, 2019-12-07:
@PierreScerri  could you add a log file, please?
https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG

DEBUG=contribModbus*,modbus-serial

PierreScerri, 2019-12-07:
Please excuse my ignorance, but I have looked at this page and still can't tell where to issue the command. Is it in terminal? is it in the add-on config? Where?
[image]

PierreScerri, 2019-12-07:
This is my add-on config:
```
{
  "credential_secret": "XXXXXXXXX",
  "dark_mode": true,
  "http_node": {
    "username": "",
    "password": ""
  },
  "http_static": {
    "username": "",
    "password": ""
  },
  "ssl": true,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem",
  "require_ssl": true,
  "system_packages": [],
  "npm_packages": [],
  "init_commands": [
    "DEBUG=contribModbus*,modbus-serial"
  ]
}
```

Now where is the log file?

biancode, 2019-12-08:
please, test with the new v5.2.x

biancode, 2019-12-08:
[image]

new options to control reconnect

PierreScerri, 2019-12-08:
The reconnect timeout is already there.

What is unitid's in parallel.  Don't have that.

Will try to update to v5.2.

[image]

PierreScerri, 2019-12-08:
Don't see a way to update or remove the Modbus pallette.  Even tried uninstalling and reinstalling the add-on.  No joy.

[image]

biancode, 2019-12-08:
I’ve got the E-Mail from libraries.io - now it should show v5.2.0

biancode, 2019-12-08:
https://libraries.io/npm/node-red-contrib-modbus

PierreScerri, 2019-12-08:
I saw that page.  

However I cannot use terminal commands like 

npm install node-red-contrib-modbus@5.2.0

Does not work on Hassio terminal via SSH.

[image]

PierreScerri, 2019-12-09:
Update works.  Only thing is that output frequency has slowed down from 10s to 90s as shown in the first image.  I have not changed any settings except that I've checked 'Reconnect on timeout' as previously instructed.

[image]
[image]
[image]

biancode, 2019-12-14:
Please, open a new issue if slow down is a failure. As I can see here it is to fix with a better node structure in the flow. If you open a new issue then send me a flow to test with, please.

closed 2019-12-14 by biancode