BiancoRoyal/node-red-contrib-modbus #164: Modbus still not working

PierreScerri, 2019-12-15:
**I'm submitting a ...**  (check one with "x")

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 After upgrading to Modbus 5.2, only one of two modes devices is working as before.  Second device is not giving any output.

 **Expected behavior**
 
 This flow was working until a short while ago.

### Your Environment
**Please tell us about your environment:**

RPi 3+ with Hassos 2.12. Hassio 0.102.3. Node-red add-on 5.0.6
Eastron SDM120 and SDM230 energy meters connected to Pi via USB.

Node-red add-on log, flow & video of problem:
[Archive.zip](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/3964762/Archive.zip)

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other HASSOS on RPi3:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X

5.2.0
* **Node:** (min. 10.x)
   `node --version` = v10.16.3

PierreScerri, 2019-12-15:
I have removed the section that was working, deployed and the remaining non working section started to work properly.

I then pasted the removed section back, deployed and everything was working.

Very confusing.

biancode, 2019-12-15:
Was the flow build with v4.x? Sometimes it is not working from major to major if some changes in the UI (HTML part of the node)

biancode, 2019-12-15:
The new option UnitId's in parallel could have some trouble with old flows, but there is mostly the only fix to export and import the flows to get the update fro Node-RED to the old nodes i the flow.

closed 2019-12-15 by biancode