BiancoRoyal/node-red-contrib-modbus #165: Queue freeze on multiple messages

bashlakov, 2019-12-27:
# Problem
 - ( x ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 
 **Current behavior**
Queue on Modbus Flex Getter freezes on recieving multiple messages in a few seconds. Sometimes it continue to work after recieving a single message.
 
 **Expected behavior**
Not freeze on multiple messages.
 
 **Minimal reproduction of the problem with instructions**
I've captured a video with problem reproduction:
[image]
   
### Your Environment

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  
  * Device :
    - (x) Linux: Debian 9.9, Kernel 4.9.0-9-amd64
   
   * Description of the connecting Modbus system:
   
     - TCP to RS485 converter (https://shop.nag.ru/catalog/00007.Avtomatizatsiya-i-monitoring-IoT/05629.ERD-Kontrollery/16094.SNR-ERD-3s - Russian docs only)
    - any modbus-connected device (plc, multimeter, etc..)
    
* **node-red-contrib-modbus version:** v7.7.3

* **modbus-serial version:** v5.2.0

* **Node:** v10.16.0

### Flow export
<details>
  <summary>Click to expand</summary>
  
```
[{"id":"cd13674e.c7dd28","type":"tab","label":"Modbus bug","disabled":false,"info":""},{"id":"387da264.60dcee","type":"modbus-flex-getter","z":"cd13674e.c7dd28","name":"","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"c1f6ce0b.34297","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":480,"y":220,"wires":[[],[]]},{"id":"f72f2449.0dca98","type":"inject","z":"cd13674e.c7dd28","name":"","topic":"","payload":"{\"fc\":3,\"unitid\":1,\"address\":35,\"quantity\":44}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":210,"y":220,"wires":[["387da264.60dcee"]]},{"id":"85ff4162.c7fa5","type":"modbus-queue-info","z":"cd13674e.c7dd28","name":"","topic":"","unitid":"1","queueReadIntervalTime":1000,"lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"c1f6ce0b.34297","errorOnHighLevel":false,"x":480,"y":320,"wires":[[]]},{"id":"c1f6ce0b.34297","type":"modbus-client","z":"","name":"ERDx","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.1.101","tcpPort":"5002","tcpType":"TELNET","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"300","clientTimeout":"10000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]
```
</details>

schefa11, 2020-02-14:
I am able to reproduce this bug, too.

modbus-client node settings:
rtu-buffered + default timeouts + reconnect on timeout: true + unitid's in parallel: false

Each time I want to read two 'packages' in a row the queue increases by one request. (The first one of each 'double request' is read successfully).
No matter how long I wait the queue stays on the same level if there is no further incoming request.
However, one seperate request releases the queue.

[image]
(all 'request' - functions contain the same code) 

I noticed a different behavior if I change the modbus-client node setting (unitid's in parallel: true) and there are requests with different unitId's in front of the flex getter. The node never becomes active again unless I restart my device.

[image]

biancode, 2020-04-01:
please, test v5.6.0 there is a whole new UnitId handling

schefa11, 2020-04-06:
The "freeze"-issue caused by multiple requests seems solved due to the new unit id handling (well done).
I could not check multiple unit id's so far, because I currently do not have more than one rtu slave.

Regarding the "reconnect-issue" I described before:
I don't know if it is the same "error-cause" but once the rtu-connection is broken, the node is still not able to reconnect until node-red restarts (deploy does not help). Can someone confirm this behavior?

Reproduce error:
1. Deploy a well configured node (node status is connected)
2. Disconnect at least one cable (D+ or D-)
3. Set read request (request will fail of course)
4. Connection becomes not reactivated

biancode, 2020-04-06:
@schefa11 with the option of reconnect on timeout it should reconnect on the next read with timeout

biancode, 2020-04-06:
with empty msg on failure you could check if there comes an error from Modbus or serialport

biancode, 2020-04-06:
@schefa11 check that the config node does not use the parallel mode - parallel is just working for TCP

schefa11, 2020-04-07:
@biancode: reconnect on timeout is activated and unit id in parallel is deactivated.
I can see that the node is trying to reconnect (status is switching between reconnect after 2000ms and initialize). And if I restart Node-Red, the status went to "connected" directly and I am able to read the data.

Conceivable scenario:
Seems that the connection keeps open even if the a read operation fails. The statemachine went to reconnecting and is not able to reestablish the connection because the port is still open?!

biancode, 2020-04-07:
interesting - it is only on odd modified deploys and yes - the state machine gets not the register info well to go back to new -> init - see de-register and register of nodes on the client

biancode, 2020-04-07:
it looks like node-red does not wait for the done function on close - it has to wait for the node call of done

biancode, 2020-04-09:
Please, test v5.9.0

biancode, 2020-04-10:
reopen if it is not fixed, please - test v5.10.1

closed 2020-04-10 by biancode

schefa11, 2020-04-15:
i can confirm: issue is solved.

kpm-at-hfi, 2020-06-01:
@biancode I just wanted to say thanks for the new "empty msg on failure" option. That's very useful.