BiancoRoyal/node-red-contrib-modbus #167: QIoT node-red-contrib-modbus Modbus-Flex-Getter "PortNotOpenError: Port Not Open"

LTDS, 2020-01-14:
Hi there !

I'm trying to use the QNAP QIoT Suite to read values on a Modbus probe. This Suite is using NodeRed. I'm using the node-red-contrib-modbus module. 

The probe is connected to a RS485 Module witch is itself connected on a RaspberryPi with the 192.168.1.22 IP adress.

There is my configuration :
Inject > Function > Modbus-Flex-Getter > Modbus Response
                                                               > Debug

Setup as following : 
- Inject : 
Chain [] 

- Function : 
 msg.payload = {
     'fc': 3,
     'unitid': 10,
     'address': 164,
     'quantity': 1};
 return msg; 

- Modbus-Flex-Getter : 
Modbus Client : 

Nœud | "a2c729b5.350318"
-- | --
Nom | test
Type | modbus-client
clienttype | "serial"
bufferCommands | true
stateLogEnabled | false
tcpHost | "192.168.1.22"
tcpPort | "502"
tcpType | "DEFAULT"
serialPort | "192.168.1.22/dev/ttyUSB0"
serialType | "RTU"
serialBaudrate | "9600"
serialDatabits | "8"
serialStopbits | "2"
serialParity | "none"
serialConnectionDelay | "100"
unit_id | 10
commandDelay | 1
clientTimeout | 1000
reconnectOnTimeout | false
reconnectTimeout | 2000
parallelUnitIdsAllowed | true

When I trigger the "Inject" node I get these messages : 
14/01/2020 à 15:50:26node: 23caa4eb.b96e1c
msg : PortNotOpenError
"PortNotOpenError: Port Not Open"
14/01/2020 à 15:50:26node: 23caa4eb.b96e1cmsg : string[65]
"FSM Reset On State {"value":"failed","actions":[],"changed":true}"
14/01/2020 à 15:50:26node: 23caa4eb.b96e1cmsg : string[65]
"FSM Reset On State {"value":"failed","actions":[],"changed":true}"

NodeRed is completely new for me and I don't quite get what is the issue here ! Any help ?

biancode, 2020-01-21:
maybe a duplicate of #160

biancode, 2020-01-21:
Btw, the serial port looks wrong for me "192.168.1.22/dev/ttyUSB0" should be "/dev/ttyUSB0" shouldn't it?

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode