BiancoRoyal/node-red-contrib-modbus #168: Modbus over RS485 - queue build up

rtborg, 2020-01-15:
**I'm submitting a ...**  

 - (x ) unexpected behavior 
 
 **Current behavior**
 
The modbus rtu queue keeps building up messages one after another and the poll rate is really slow.
 
 **Expected behavior**
 
The queue should be emptied faster than it's being filled.
 
 **Minimal reproduction of the problem with instructions**
 
I am using [this](https://flows.nodered.org/flow/75a4715fb6a3595057343a92abc493e4) example flow to pull data from SDM120 using [RS485 to TTL adapter](https://www.elecrow.com/uart-ttl-to-rs485-twoway-converter-p-1545.html), I've also tried using [USB to RS485 adapter](https://www.ebay.co.uk/itm/1x-USB-to-RS485-Modbus-converter-Supports-Win7-XP-Vista-Linux-Mac-OS-WinCE5-0/123737802444?hash=item1ccf5906cc:g:M0MAAOSwQwBZi3gC) with the same result. What happens is the queue object is showing increasing number of items and the flow is very slow. When I click on any of the modbus read nodes the queue seems to start moving for a short period. Illustration below:
[image]
   
The serial port queue delay is 100ms.

  * ( x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  
  * Device : Raspberry Pi Zero W
    - ( x) Raspbian GNU/Linux 10 (buster)
    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v10.18.0

fonske, 2020-01-18:
I had same problems with this V5 version after updating from 4.1.3.
I removed it and installed the older one again.

cd /root/.node-red
npm install -g node-red-contrib-modbus@4.1.3 --unsafe-perm --build-from-source

rtborg, 2020-01-18:
@fonske, thank you for the comment - I've reverted to 4.1.3 and it works as intended. 

Is that a known bug?

biancode, 2020-01-21:
Hi, for me it looks like the target is not fast enough to handle all requests. The queue info node shows it. You send too much and/or too fast. Mostly it is a configuration problem with that behaviour. Version 5 changed in handling queues so it potentiating the problem.

- check the speed of your requests
- think about the amount of requesting nodes
- use single requests and pick up what you need from the data

Now it looks like you send 4 requests every 5 sec., but maybe there are some more nodes to read more data from the SDM120. Did you check my example for the SDM120 and how to work with that device in my case?

gugani, 2020-02-14:
Posting here because I think it's related to the upgrade and to this issue. I'm having that strange behavior after upgrade from 4.1.3, the queue keeps growing and if I try to emtpy with `msg.resetQueue` = true I get `TypeError: a.initQueue is not a function`. I haven't change anything in what it was a already working Node Red project with 4.1.3.

Device : BeagleBone Black with Debian GNU/Linux 10 (buster)
Node-RED v1.0.2
node-red-contrib-modbus version: v5.2.0
node --version = v10.15.2

biancode, 2020-02-14:
@gugani if it is a major update from 4 to, then it has breaking-changes. Please, try the best way for major releases: uninstall the old version (or now your upgraded package) of Modbus before and install it again and after that export all the Modbus containing flows and import them again - deploy it. Not all nodes are backwards compatible on major releases and sometimes they have to be deleted and add again. From 4 to 5 it is also a breaking-change in the config node because it gots a new queue handling.

gugani, 2020-02-18:
Thanks @biancode, I made a clean install, I imported the previously exported nodes  and everything is working in 1.0.2. Now updating to the tested version 1.0.3. Thanks!!

gugani, 2020-02-19:
Hi, I have to add that communications are working, but the `msg.resetQueue = true` throws an `TypeError: a.initQueue is not a function` I open a new issue to avoid meshing more this thread.

closed 2020-03-28 by biancode