BiancoRoyal/node-red-contrib-modbus #169: possible memory leak

tasoss, 2020-01-21:
Hello.
I'm having the following problem and i need some help.

> 21 Jan 11:13:51 - [info] Node-RED version: v1.0.3
21 Jan 11:13:51 - [info] Node.js  version: v12.14.1
21 Jan 11:13:51 - [info] Linux 4.4.0-171-generic x64 LE
21 Jan 11:13:52 - [info] Loading palette nodes
21 Jan 11:13:54 - [info] Dashboard version 2.19.3 started at /ui

Please take a look [here](https://pastebin.com/juxBDvNd) (it's the output)
I know that given you a scenario would be the ideal case but i don't know if it's possible.
Is it possible to somehow found out why i get 'Port Not Open' ?
Any ideas on how i could possibly debug it?
Thanks!

biancode, 2020-01-21:
please check #160

biancode, 2020-01-21:
Hm, you have 1957.7 of 2084.0 MB in use and it is not possible to handle more at this point. With that, you got the errors. Is it growing over time from less MB in RAM? Why do you think about a memory leak here? That is not to see for me from the Pastebin, sorry.

tasoss, 2020-01-21:
Hello and thanks for replying.
I have 6GB of RAM.I don't know why it's using only 2???
In another crash i additionally got

> contribModbus:flex:getter FSM Not Ready To Read At State broken

I'm running it like this

> DEBUG=contribModbus:flex:*  node-red -v
Does it help if i use contribModbus* ?
So from what i can understand does #160 will solve this error?

Not something concrete about mem leak.Mainly because of

> FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

I think i should follow #160

closed 2020-01-21 by tasoss

biancode, 2020-01-21:
OK. If it does not help please send a log from contribModbus* by E-Mail ;)

tasoss, 2020-01-21:
hahahaha
ok thanks!

tasoss, 2020-01-24:
@biancode  hello again.I have sent you the log files i don't know if you got them.
btw i have noticed that the app is using more and more memory as the time passes.
Please take a look [here](https://pastebin.com/EYxz3RuN)

biancode, 2020-01-24:
@tasoss normally it should show some like a saw pattern (up and down) from the [nodejs garbage collector ](https://blog.risingstack.com/node-js-at-scale-node-js-garbage-collection/) behaviour on memory consumption over time. It is fine if it goes back down from time to time and is not growing endless, which would kill the process after a deterministic time.

biancode, 2020-01-24:
@tasoss why is the log file 2,8GB ?

tasoss, 2020-01-24:
@biancode  DEBUG=contribModbus*  node-red -v > logfile 2>&1
I have also setup node-red's settings.js level to 'trace'
It's growing and when it reaches 2GB and then it crashes.
In the last link that i have sent you , the memory lowers(from 1.7GB to 500MB) BUT it has already crashed(the pid changes and that's why it lowers-starts from the beginning).
Do you want me to log in a different way?
Thanks again!