BiancoRoyal/node-red-contrib-modbus #172: Modbus Flex Server - Fails after changing Unit-Id

MattBrittan, 2020-01-22:
**I'm submitting a ...** 

 - (x ) bug report
 
 **Current behavior**

Changing the ```Unit-Id``` in a 'Modbus Flex Server' prevents it from responding to any queries. Manually updating the node config (export, edit, import) to change ```Unit-Id``` from a string to number resolves the issue.
 
 **Expected behavior**

Changing the ```Unit-Id``` makes the server respond to queries targeting that unit id.
 
 **Minimal reproduction of the problem with instructions**

   1.  On new flow add a 'Modbus Flex Server'
   2.  Query with unit-id = 1 works successfully
   3.  Change the ```unit-id``` to 2 and deploy
   4.  Attempt a query (using unit-id = 2) and it times out
   5. Export flow and change ```"unitId": "2",```  to ```"unitId": 2,```, import the result and deploy
   6. Modbus query works as expected
 
### Your Environment

Created docker container with ```docker run -d -it -p 1880:1880 -p 10502:10502 --name mynodered nodered/node-red```; used ```docker exec``` to connect to it and ran ```npm install node-red-contrib-modbus```.

  * ( x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it

  * Device :

Docker image running under Windows (so linux vm).

   * Description of the connecting Modbus system:

N/A   

* **node-red-contrib-modbus version:** 

v5.2.0

* **Node:** (min. 10.x)

   `node --version` = v10.18.0

biancode, 2020-01-22:
Hi, I saw that problem with node-red v1.x and existing flows, too. It is a problem that number types of nodes (HTML setting part of a node) are strings if they are in quotes. As you see in your description, it changes the "2" into 2. Please, could you search the Node-RED forum or ask there for that? If you find some, then set a link here, please. The simple way here is also just to export and import the flows to get it working well again. A convert in the node could also help, but this is not the best way if we think about to change all code snippets inside nodes and so on.

MattBrittan, 2020-01-22:
Thanks - I found this issue when responding to a [stack over flow question](https://stackoverflow.com/questions/59844281/simulating-multiple-modbus-slave-devices-using-node-red)  for someone experiencing the same (I assume) issue and thought I should report it (could not find anything similar logged already). 

I tested a straight export and reimport and that does not fix the issue; it looks like you need to manually update the export and change the "2" to 2. I also note that this happens with other fields (but in most cases you run a parseInt at some point (e.g. [unit id in the clients](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/master/src/modbus-client.js#L52)) so it does not matter).

I have searched the Node-RED forum and could not find anything relevant (other than a [question](https://discourse.nodered.org/t/property-value-saved-as-string-to-flow-json-instead-of-number/7363) without an answer). Unfortunately as I don't use this node myself and have no experience with node-red development this is not a question I'm comfortable asking in the forum.

biancode, 2020-04-01:
please, test v5.6.0 it improves UnitId handling

MattBrittan, 2020-04-05:
Sorry - re-tested with 5.8.0 and the issue still exists. To retest I created the 'Modbus Flex Server', checked it worked (with a 'Modbus Getter') then changed the flex server unit-id to 2, deployed, changed unit-id back to 1, deployed and retested (failed to respond). Exporting, removing the quotes around the unitid, and reimporting fixes things.

biancode, 2020-04-10:
I'll add some parseInt steps, but it is not possible in all cases cause unitid can be none, but it is set from the client (config-node) to a fix UnitId as the best step in a flow

jbofill, 2020-04-12:
I think there is an additional issue here: when unit-id is 0, it will always be transformed to 1.
 
In line [https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/master/src/modbus-client.js#L60](modbus-client) If config.unit-id == 0 (or "0"), parseInt(0) will be false and thus assign default defaultUnitId.

PD: I'm new to node-red, excuse me if I'm off topic.

biancode, 2020-04-12:
Yes, that is the idea for serial it is min. 1 and on TCP it is not interesting. Am I wrong at this point?

jbofill, 2020-04-13:
I am monitoring a Huawei solar inverter Sun2000l. Using python it responds on unit-id 0. 
I checked it using tshark.

    # shark -i eth0 -f "src net 192.168.1.134" 
    Capturing on 'eth0' 

    1 0.000000000 192.168.1.134 → 192.168.1.81 TCP 60 502 → 45953 [SYN, ACK] Seq=0 Ack=1 Win=1500 Len=0 MSS=512
    2 1.508803601 192.168.1.134 → 192.168.1.81 Modbus/TCP 67 Response: Trans:     1; Unit:   0, Func:   3: Read Holding Registers

In my opinion unit zero is also valid.

biancode, 2020-04-13:
@jbofill on TCP it is. You're right at this point. Modbus TCP does not use the UnitID anymore as you can read in the Specs ;) With that you can send 0 or 1. I decided to use 1 every time cause of the compatible transform from TCP to Serial if there is some at the other side. Are you in trouble with the default of 1, please?

jbofill, 2020-04-13:
Sorry I have no knowledge of Modbus TPC internals.
But, yes, I am in technical trouble with the default of 1. I need to be able to use 0, please.

biancode, 2020-04-13:
@jbofill no problem, but I have to think about it. You're the first in four years and it has to be compatible. Please, open a new issue and describe your use case(s) and the needs for Node-RED.

jbofill, 2020-04-13:
Ok, opened new issue [https://github.com/BiancoRoyal/node-red-contrib-modbus/issues/188]()

closed 2022-04-21 by biancode