BiancoRoyal/node-red-contrib-modbus #174: Support Node-RED V1.0.3

heleon19, 2020-01-23:
Does anyone know if the library works reliably with the new Node-RED version 1.0.3? Can the new asynchronous processing of the flows not cause any problems?

biancode, 2020-01-23:
That should lay more in the flows instead of the nodes, doesn't it?

heleon19, 2020-01-27:
This is correct in principle, but the temporal behaviour is now changing.  It can now happen more often that two commands are written almost simultaneously.

Nxito, 2020-01-30:
- Node-RED version: v1.0.3
- Node.js version: v10.15.3
- NPM 6.4.1
- Windows 7

I am testing a beacon of the Patlite brand.

I updated node-red to 1.0.3 because I had 1.0.0 and it didnt works

after updating I receive other errors

log examples:
when I activate a modbus node it appears to me->

- 9 Jan 18:37:33 - [warn] [modbus-write: Write: Team leader advice] FSM Reset On State {"value": "failed", "actions": [], "changed": true}

After a while of working fine, it appears-> 

- 30 Jan 01:19:29 - [error] [modbus-write: Write: Team leader advice] Error: FSM Not Ready To Write At State stopped

and stop working
in versions prior to 1.0.0, the beacon works correctly

This node is very complete, but since 1.0.0 it can fails, so I leave this message so that other users are aware of it

biancode, 2020-01-30:
Hm, I tested all with Node-RED 1.0.3 for the last contribution package version and with the Modbus Simulator via TCP and Serial. It all worked very well. Failed to Stopped state changes should just happen if there is no node registered anymore or if the node is in the close mode as you can read in the code of the client node. It needs more logs and a reproducible flow.

Nxito, 2020-02-03:
Here is an example of the record:

29 Jan 17:22:41 - [info] Starting flows
29 Jan 17:22:42 - [info] Started flows
29 Jan 17:22:42 - [error] [OPCUA-IIoT-Read: 6e915dc4.0001c4] Error: Client Not SESSIONACTIVE On Read
29 Jan 17:22:42 - [error] [OPCUA-IIoT-Read: f9a382cf.2b206] Error: Client Not SESSIONACTIVE On Read
29 Jan 17:22:42 - [error] [OPCUA-IIoT-Read: fbdaf455.5d6498] Error: Client Not SESSIONACTIVE On Read
29 Jan 17:22:44 - [warn] [OPCUA-IIoT-Flex-Server: 95398011.c113c] Creating new address for OPC UA
29 Jan 17:22:44 - [warn] [OPCUA-IIoT-Flex-Server: 95398011.c113c] OPC UA address created
29 Jan 17:22:44 - [error] [OPCUA-IIoT-Read: 6e915dc4.0001c4] Error: Client Not SESSIONACTIVE On Read
29 Jan 17:22:44 - [error] [OPCUA-IIoT-Read: f9a382cf.2b206] Error: Client Not SESSIONACTIVE On Read
29 Jan 17:22:44 - [error] [OPCUA-IIoT-Read: fbdaf455.5d6498] Error: Client Not SESSIONACTIVE On Read
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m
RegisterServer to the LDS has failed during secure connection => please check that you server certificate is trusted by the LDS err: _socket has been disconnected by third party
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m
29 Jan 17:23:34 - [info] Stopping flows
29 Jan 17:23:34 - [warn] [modbus-write: Write: Team leader advice] FSM Reset On State {"value": "failed", "actions": [], "changed": true}
29 Jan 17:23:34 - [warn] [modbus-write: Write: Team leader advice] FSM Reset On State {"value": "failed", "actions": [], "changed": true}
29 Jan 17:23:34 - [warn] [modbus-write: Write: Team leader advice] FSM Reset On State {"value": "failed", "actions": [], "changed": true}
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m
RegisterServer to the LDS has failed during secure connection => please check that you server certificate is trusted by the LDS err: _socket has been disconnected by third party
[33mWarning: AddressSpace # constructEventData (eventType, options) cannot find property [39mAuditSecurityEventType => [36mstatusCodeId [39m

Here is the flow :
`[{"id":"9df96908.d81da8","type":"function","z":"9c5ab8d1.cd4228","name":"Call  team leader","func":"try{\nnode.status({fill:\"green\",shape:\"dot\",text:\"run\"});\nvar value=msg.payload[0].value.value\n\n\nif (value==0){\n    msg.payload=[256];\n    node.status({fill:\"grey\",shape:\"ring\",text:\"off\"});\n    global.set(\"callTeamLeaderHR1\", 0);\n}\nelse {\n    msg.payload=[262];\n    node.status({fill:\"blue\",shape:\"dot\",text:\"on\"});\n    global.set(\"callTeamLeaderHR1\", 0); \n}\nreturn msg;\n    \n}\n\ncatch(err){\n    node.error(err)\n    node.status({fill:\"red\",shape:\"ring\",text:\"error\"});\n    \n    \n}","outputs":1,"noerr":0,"x":760,"y":120,"wires":[["b9c20b87.e2cb28","2405cde6.272892"]]},{"id":"c6ae2be9.95e2b8","type":"OPCUA-IIoT-Read","z":"9c5ab8d1.cd4228","attributeId":"13","maxAge":1,"depth":1,"connector":"fdc25143.b20ab","name":"","justValue":true,"showStatusActivities":true,"showErrors":true,"parseStrings":false,"historyDays":1,"serverMaxItemsToRead":"","x":390,"y":120,"wires":[["4bf77a8.e877984","43dec47b.f29c4c"]]},{"id":"c2b6b734.00f9d8","type":"modbus-response","z":"9c5ab8d1.cd4228","name":"","registerShowMax":20,"x":1170,"y":160,"wires":[]},{"id":"174a62c.fd00e9d","type":"inject","z":"9c5ab8d1.cd4228","name":"Inject OpcUa","topic":"","payload":"{\"nodetype\":\"inject\",\"injectType\":\"inject\",\"addressSpaceItems\":[{\"name\":\"\",\"nodeId\":\"ns=1;s=balizaHR1-Call\",\"datatypeName\":\"Byte\"}]}","payloadType":"json","repeat":"1","crontab":"","once":false,"onceDelay":0.1,"x":160,"y":120,"wires":[["c6ae2be9.95e2b8"]]},{"id":"4bf77a8.e877984","type":"rbe","z":"9c5ab8d1.cd4228","name":"","func":"rbei","gap":"","start":"","inout":"out","property":"msg.payload[0].value.value","x":570,"y":120,"wires":[["9df96908.d81da8"]]},{"id":"b9c20b87.e2cb28","type":"debug","z":"9c5ab8d1.cd4228","name":"Send status Beacon HR1","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":1010,"y":80,"wires":[]},{"id":"43dec47b.f29c4c","type":"debug","z":"9c5ab8d1.cd4228","name":"Send msg","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":580,"y":80,"wires":[]},{"id":"2405cde6.272892","type":"modbus-write","z":"9c5ab8d1.cd4228","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"MHoldingRegisters","adr":"14","quantity":"1","server":"8aa59291.9dc38","x":980,"y":140,"wires":[[],["c2b6b734.00f9d8"]]},{"id":"fdc25143.b20ab","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://127.0.0.1:4840","keepSessionAlive":true,"loginEnabled":false,"securityPolicy":"None","securityMode":"NONE","name":"LOCAL DEMO SERVER","showErrors":true,"individualCerts":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"60000","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetr [... cut ...] sts":""},{"id":"8aa59291.9dc38","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectOnTimeout":true,"reconnectTimeout":2500,"parallelUnitIdsAllowed":false}]`

biancode, 2020-02-14:
Please, reopen if you are a sponsor.

closed 2020-02-14 by biancode