BiancoRoyal/node-red-contrib-modbus #175: Modbus queue - Optimize handling of multiple devices

schefa11, 2020-01-24:
**I'm submitting a ...**  

 - (x) feature request
 
 **What is the motivation / use case for changing the behavior?**
Optimization of 'Modbus queue handling' for multiple devices
 
 **Current behavior**
One 'Modbus Queue Info' node covers only one modbus server with a specific unit id at the same time. The unit id is not configurable via incoming message.
 
 **Desired behavior**
There are different possibilities to improve the behavior in my opinion. I don't know how this can be implemented (regarding to your latest archtecture), so a few suggestions on my part:
- New / modified 'Modbus Queue Info':
   a) Unit id (and maybe further node properties) configurable via incoming message (flex queue)
   b) One node handels the queues of all servers. Event-based --> e.g. Output is triggered if the queue of a specific server changes. Input only for resetting (does not trigger the Node itself).
   + Node Input: {"server": "MyRTUServer", "unitId" : 3, resetQueue : true}
   + Node Output: {"server": "MyRTUServer", "unitId" : 1, "queues" : [...]}
- Expanded 'Modbus Flex Getter' - Node:
   + additional Node Input: reset queue by "msg.resetQueue = true"
   + additional Node Output: Current Queue-Info at the output when data has been read

biancode, 2020-04-01:
please test v5.6.0

schefa11, 2020-04-02:
Hi biancode,
thanks for the update. Can you specify what I should test or rather what did you change exactly (regarding the queuing behavior/monitoring/resetting)?
I currently do not have a modbus rtu slave to build up a queue, but it seems that I cannot change the "linked" unitid of the "modbus Queue Info". The node status stays at preconfigured unitid after trying to override it with "msg.payload.unitid".

biancode, 2020-04-09:
Ah, I see. I'm sorry but that issue here, would be a very new node to handle all queues of all UnitId's via events, but I thought you are a perfect tester with multiple RTU Units :)

biancode, 2020-04-10:
v5.10.0 first version with an enhanced queue info

schefa11, 2020-04-16:
@biancode:
I'm not sure how the node should behave. E.g:
- Does **"An update of all unit queues can be watched by checking the option"** mean, that any change of the queue size result in a message? I don't see this effect.
- Is the info message **"Use just a single queue info per client if you activate that option."** related to the radio Box "**All Queues**"? If yes, I think it should be **...if you deactivate that option."** The lines of the info box and node configuration arn't on the same level so it is a little bit confusing.
- Can I read and reset one dedicated queue if I the "all queues" are activated.

[image]

Is there a reason why you don't set a message when lowLowLevel is reached?

What do you think about making the inputs configurable? E.g for "dynamic reset":
```
    node.on('input', function (msg) {
      let msgUnitId = Number(msg.payload.unitid) || node.unitid
      node.lowLowLevel = Number(msg.payload.lowLowLevel) || node.lowLowLevel
      node.lowLevel = Number(msg.payload.lowLevel) || node.lowLevel
      .
      .
```

biancode, 2020-05-17:
@schefa11 lowLowLevel is reached would be a new feature request, please

biancode, 2020-05-17:
@schefa11 support us on https://osi.bianco-royal.com/  or send your pull request to get more features, please

closed 2020-05-21 by biancode