BiancoRoyal/node-red-contrib-modbus #177: Using Modbus Flex Connector returns FSM Reset On State {"value":"failed","actions":[],"changed":true}

GamerWindow, 2020-02-12:
## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
Using 'Modbus Flex Connector' and 'Modbus Flex Getter' returns warning: `FSM Reset On State {"value":"failed","actions":[],"changed":true}`. The injected Payload to  'Modbus Flex Getter' is: 

```
msg.payload = {
    'value':msg.payload,
    'fc': 3,
    'unitid': 1,
    'address': 24 ,
    'quantity': 1
};
return msg;
```

What does the warning mean? Is it serious?

I do not get any Modbus output.
 
 **Expected behavior**
 
No warning.
 
 **Minimal reproduction of the problem with instructions**
 
   1. Connect Inject, function Node and Modbus Flex Connector
   2. Connect Inject, function Node and Modbus Flex Getter
   3. Inject to Modbus Flex Connector, then to Modbus Flex Getter

Demo Flow:
```
[{"id":"3ba35010.8a2fa","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"e6b7f53e.5d7b18","type":"modbus-flex-connector","z":"3ba35010.8a2fa","name":"","maxReconnectsPerMinute":4,"emptyQueue":false,"showStatusActivities":false,"showErrors":true,"server":"ebc49841.242f08","x":570,"y":120,"wires":[]},{"id":"e09e1c56.55e1c8","type":"function","z":"3ba35010.8a2fa","name":"","func":"msg.payload = {\n    'connectorType': 'TCP',\n    'tcpHost': '10.0.0.81',\n    'tcpPort': '502',\n    'unitId': 2\n};\nreturn msg;","outputs":1,"noerr":0,"x":380,"y":120,"wires":[["e6b7f53e.5d7b18"]]},{"id":"1faf2d77.4fb13b","type":"inject","z":"3ba35010.8a2fa","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":240,"y":120,"wires":[["e09e1c56.55e1c8"]]},{"id":"5945e4a3.b9208c","type":"inject","z":"3ba35010.8a2fa","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":260,"y":180,"wires":[["8fe16f06.445b88"]]},{"id":"8fe16f06.445b88","type":"function","z":"3ba35010.8a2fa","name":"","func":"msg.payload = {\n    'value':msg.payload,\n    'fc': 3,\n    'unitid': 1,\n    'address': 24 ,\n    'quantity': 1\n};\nreturn msg;","outputs":1,"noerr":0,"x":410,"y":180,"wires":[["c8c340d5.7666d8"]]},{"id":"c8c340d5.7666d8","type":"modbus-flex-getter","z":"3ba35010.8a2fa","name":"","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"ebc49841.242f08","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":590,"y":180,"wires":[[],["b3c606f1.ac3178"]]},{"id":"b3c606f1.ac3178","type":"debug","z":"3ba35010.8a2fa","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":770,"y":200,"wires":[]},{"id":"ebc49841.242f08","type":"modbus-client","z":"","name":"Test","clienttype":"tcp","bufferCommands":false,"stateLogEnabled":false,"tcpHost":"10.0.0.254","tcpPort":"20502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":false,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]
```
   
### Your Environment
**Please tell us about your environment:**

  * ( x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  
    
  * Device :
    
    - ( x ) Node-RED Docker Image v 1.0.2 
   
   * Description of the connecting Modbus system:
   
     - Name   : Acromag 967/8 EN 

    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v10.16.3

bakfunk, 2020-02-12:
same probleme here. any solutions?

frankzappa82, 2020-02-13:
Issue disappeared!!!!
I don't know why....
Just gamed with queue , and ,after deleted the three Modbus flex getter buttons, put it a new one, and OK!!!
bye.

GamerWindow, 2020-02-13:
Tried with various queue settings, worked at the first few requests but then i got the FSM warnings again. Sorry, no solution yet.

frankzappa82, 2020-02-13:
Try to set queue to 10 ms (with commands queue checked), applied to all the flow.
Then delete all the modbus nodes.
Then deploy.
If ok ==> add one modbus flex getter/flex write (it's preferred to use only these two nodes).
Then deploy.
If ok ==> add all the other ones.
Good look.

biancode, 2020-02-14:
Please, reopen if you want support as a sponsor.

biancode, 2020-02-14:
maybe duplicate with  #165

closed 2020-02-14 by biancode

PierreScerri, 2020-03-02:
I have the same error.  Using only Modbus Read node (9 instances) from 2 devices. Errors are duplicated ie same error twice for each node (see log)

System: Rpi3+ with Hassio and Node-Red addon (versions in log below)

Node-Red flow for one of the registers:
`[{"id":"30b8c211.4649be","type":"modbus-read","z":"1e6fb6a5.f40a31","name":"Energy Imp","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"72","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"5","server":"a0f341cb.3769f8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":120,"y":260,"wires":[["90c6871b.c44238"],[]]},{"id":"90c6871b.c44238","type":"function","z":"1e6fb6a5.f40a31","name":"Import","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = (parseFloat(fltView[0])) ;\nmsg.topic = \"import\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":250,"y":260,"wires":[["47aa5c7d.ef1f5c"]]},{"id":"47aa5c7d.ef1f5c","type":"range","z":"1e6fb6a5.f40a31","minin":"0","maxin":"6324.86","minout":"36481.26","maxout":"42819.97","action":"scale","round":false,"property":"payload","name":"Map","x":370,"y":260,"wires":[["d18f41e3.b0843"]]},{"id":"d18f41e3.b0843","type":"function","z":"1e6fb6a5.f40a31","name":".00","func":"\nmsg.payload = (parseFloat(msg.payload)).toFixed(2) ;\nmsg.topic = \"import\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":490,"y":260,"wires":[["9965468d.bbf198"]]},{"id":"9965468d.bbf198","type":"ha-entity","z":"1e6fb6a5.f40a31","name":"SDM230EnergyImport","server":"8d09c353.bd6828","version":1,"debugenabled":false,"outputs":1,"entityType":"sensor","config":[{"property":"name","value":"sdm230import"},{"property":"device_class","value":""},{"property":"icon","value":""},{"property":"unit_of_measurement","value":"kWh"}],"state":"payload","stateType":"msg","attributes":[],"resend":true,"outputLocation":"","outputLocationType":"none","inputOverride":"allow","x":660,"y":260,"wires":[["ca5d7b3e.2d4878"]]},{"id":"ca5d7b3e.2d4878","type":"ui_text","z":"1e6fb6a5.f40a31","group":"4781c607.4294d8","order":4,"width":0,"height":0,"name":"","label":"Imp","format":"{{msg.payload}} kWh","layout":"row-spread","x":830,"y":260,"wires":[]},{"id":"a0f341cb.3769f8","type":"modbus-client","z":"1e6fb6a5.f40a31","name":"Serial_9600_8_N_1","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"200","clientTimeout":"2000","reconnectOnTimeout":true,"reconnectTimeout":"5000","parallelUnitIdsAllowed":false},{"id":"8d09c353.bd6828","type":"server","z":"","name":"Home Assistant","legacy":false,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true},{"id":"4781c607.4294d8","type":"ui_group","z":"","name":"SDM230","tab":"fe9d9ab7.0983e8","disp":true,"width":4,"collapse":false},{"id":"fe9d9ab7.0983e8","type":"ui_tab","z":"","name":"Energy","icon":"dashboard","order":1,"disabled":false,"hidden":false}]`

Settings:
[image]

Startup log:
```
Log
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: Node-RED
 Flow-based programming for the Internet of Things
-----------------------------------------------------------
 Add-on version: 6.1.3
 You are running the latest version of this add-on.
 System: HassOS 3.11  (armv7 / raspberrypi3)
 Home Assistant Core: 0.106.2
 Home Assistant Supervisor: 206
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] node-red.sh: executing... 
patching file nodes/ui_base.html
Hunk #1 succeeded at 1243 (offset 712 lines).
audited 96 packages in 5.31s
13 packages are looking for funding
  run `npm fund` for details
found 0 vulnerabilities
[cont-init.d] node-red.sh: exited 0.
[cont-init.d] user.sh: executing... 
[cont-init.d] user.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:41:22] INFO: Starting Node-RED...
> addon-node-red@ start /opt
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
2 Mar 13:41:26 - [info] 
Welcome to Node-RED
===================
2 Mar 13:41:26 - [info] Node-RED version: v1.0.4
2 Mar 13:41:26 - [info] Node.js  version: v12.15.0
[... cut ...]
2 Mar 13:43:14 - [warn] [modbus-read:Power] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 13:43:14 - [warn] [modbus-read:Energy Imp] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 13:43:14 - [warn] [modbus-read:Energy Imp] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 13:43:14 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 13:43:14 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true})
```

biancode, 2020-03-02:
@PierreScerri that is not the flow you attached here. Where are all the other read nodes? Looks like you have the config node working twice or a flow-engine problem.

PierreScerri, 2020-03-02:
This is the whole flow:

```
[{"id":"e29a1087.4cfd8","type":"tab","label":"SDM","disabled":false,"info":""},{"id":"3d39863c.21583a","type":"comment","z":"e29a1087.4cfd8","name":"SDM230 Energy Meter","info":"","x":120,"y":40,"wires":[]},{"id":"1497c1d8.3154ee","type":"comment","z":"e29a1087.4cfd8","name":"SDM120 Energy Meter","info":"","x":120,"y":400,"wires":[]},{"id":"361e1a40.2a73e6","type":"modbus-read","z":"e29a1087.4cfd8","name":"Voltage","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"2","dataType":"InputRegister","adr":"0","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"c20cce23.34cdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":110,"y":440,"wires":[["540edeed.ba19b"],[]]},{"id":"540edeed.ba19b","type":"function","z":"e29a1087.4cfd8","name":"Voltage","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(0));\nmsg.topic = \"voltage\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":260,"y":440,"wires":[["55d57fac.e70be"]]},{"id":"2f29809b.43c12","type":"modbus-read","z":"e29a1087.4cfd8","name":"Current","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"2","dataType":"InputRegister","adr":"6","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"c20cce23.34cdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":110,"y":500,"wires":[["e97f0924.c75688"],[]]},{"id":"e97f0924.c75688","type":"function","z":"e29a1087.4cfd8","name":"Current","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"current\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":260,"y":500,"wires":[["b9825249.64ec2"]]},{"id":"3195bff0.5514c","type":"modbus-read","z":"e29a1087.4cfd8","name":"Power","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"2","dataType":"InputRegister","adr":"12","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"c20cce23.34cdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":110,"y":560,"wires":[["fd287e4b.88217"],[]]},{"id":"fd287e4b.88217","type":"function","z":"e29a1087.4cfd8","name":"Power","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"power\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":250,"y":560,"wires":[["e112ee1a.aa59"]]},{"id":"a561a8d4.12bc78","type":"modbus-read","z":"e29a1087.4cfd8","name":"Energy Sol","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"2","dataType":"InputRegister","adr":"72","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"c20cce23.34cdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":120,"y":620,"wires":[["c683f710.3212b8"],[]]},{"id":"db7c0106.907b2","type":"modbus-read","z":"e29a1087.4cfd8","name":"Energy Exp","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"74","quantity":"2","rate":"10","rateUnit":"s","delayOnStart":false,"startDelayTime":"5","server":"c20cce23.34cdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":120,"y":320,"wires":[["eb11067e.d96788"],[]]},{"id":"e283d939.5a1bc8","type":"function","z":"e29a1087.4cfd8","name":"Voltage","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(0));\nmsg.topic = \"voltage\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":260,"y":80,"wires":[["93773856.4a4b28"]]},{"id":"f75ef59d.4b03c8","type":"function","z":"e29a1087.4cfd8","name":"Current","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"current\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":260,"y":140,"wires":[["acdc5a46.0265d8"]]},{"id":"c80d1bbb.d86e78","type":"function","z":"e29a1087.4cfd8","name":"Power","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"pow [... cut ...] up","z":"","name":"SDM230","tab":"fe9d9ab7.0983e8","disp":true,"width":4,"collapse":false},{"id":"10356220.ced226","type":"ui_group","z":"","name":"SDM120","tab":"fe9d9ab7.0983e8","order":2,"disp":true,"width":4,"collapse":false},{"id":"8390dbcd.dbdd5","type":"ui_group","z":"","name":"Graph","tab":"fe9d9ab7.0983e8","order":3,"disp":true,"width":"6","collapse":false},{"id":"fe9d9ab7.0983e8","type":"ui_tab","z":"","name":"Energy","icon":"dashboard","order":1,"disabled":false,"hidden":false}]``
```

biancode, 2020-03-02:
ah ... you have two working queues - unit ids - check Logs with more [Debug](https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG), please

PierreScerri, 2020-03-02:
Does this help:

```
Log
13 packages are looking for funding
  run `npm fund` for details
found 0 vulnerabilities
[cont-init.d] node-red.sh: exited 0.
[cont-init.d] user.sh: executing... 
[cont-init.d] user.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[16:17:21] INFO: Starting Node-RED...
> addon-node-red@ start /opt
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
2 Mar 16:17:25 - [info] 
Welcome to Node-RED
===================
2 Mar 16:17:25 - [info] Node-RED version: v1.0.4
2 Mar 16:17:25 - [info] Node.js  version: v12.15.0
2 Mar 16:17:25 - [info] Linux 4.19.93-v7 arm LE
2 Mar 16:17:26 - [info] Loading palette nodes
2 Mar 16:17:27 - [debug] Module: node-red-contrib-home-assistant-websocket 0.21.3
2 Mar 16:17:27 - [debug]         /config/node-red/node_modules/node-red-contrib-home-assistant-websocket
2 Mar 16:17:27 - [debug] Module: node-red-node-timeswitch 0.0.7
2 Mar 16:17:27 - [debug]         /config/node-red/node_modules/node-red-node-timeswitch
2 Mar 16:17:27 - [debug] Module: node-red-contrib-actionflows 2.0.3
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-actionflows
2 Mar 16:17:27 - [debug] Module: node-red-contrib-alexa-home-skill 0.1.17
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-alexa-home-skill
2 Mar 16:17:27 - [debug] Module: node-red-contrib-bigtimer 2.2.6
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-bigtimer
2 Mar 16:17:27 - [debug] Module: node-red-contrib-cast 0.2.14
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-cast
2 Mar 16:17:27 - [debug] Module: node-red-contrib-counter 0.1.5
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-counter
2 Mar 16:17:27 - [debug] Module: node-red-contrib-home-assistant-websocket 0.21.3 *ignored due to local copy*
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-home-assistant-websocket
2 Mar 16:17:27 - [debug] Module: node-red-contrib-http-request 0.1.14
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-http-request
2 Mar 16:17:27 - [debug] Module: node-red-contrib-influxdb 0.4.0
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-influxdb
2 Mar 16:17:27 - [debug] Module: node-red-contrib-interval-length 0.0.4
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-interval-length
2 Mar 16:17:27 - [debug] Module: node-red-contrib-looptimer 0.0.8
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-looptimer
2 Mar 16:17:27 - [debug] Module: node-red-contrib-modbus 5.2.0
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-modbus
2 Mar 16:17:27 - [debug] Module: node-red-contrib-moment 3.0.3
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-moment
2 Mar 16:17:27 - [debug] Module: node-red-contrib-state-machine 1.2.0
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-state-machine
2 Mar 16:17:27 - [debug] Module: node-red-contrib-statistics 2.2.2
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-statistics
2 Mar 16:17:27 - [debug] Module: node-red-contrib-stoptimer 0.0.7
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-stoptimer
2 Mar 16:17:27 - [debug] Module: node-red-contrib-sunevents 2.0.2
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-sunevents
2 Mar 16:17:27 - [debug] Module: node-red-contrib-time-range-switch 1.0.0
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-time-range-switch
2 Mar 16:17:27 - [debug] Module: node-red-contrib-timecheck 1.1.0
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-timecheck
2 Mar 16:17:27 - [debug] Module: node-red-contrib-traffic 0.2.1
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-contrib-traffic
2 Mar 16:17:27 - [debug] Module: node-red-dashboard 2.19.4
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-dashboard
2 Mar 16:17:27 - [debug] Module: node-red-node-base64 0.2.1
2 Mar 16:17:27 - [debug]         /opt/node_modules/node-red-node-base64
2 Mar 16:17:27 - [debug] Module: node-red-node-email 1.7.7
[... 125 lines cut ...]
[02/Mar/2020:16:17:43 +0100] 200 -(192.168.1.120) POST /endpoint/ui/socket.io/?EIO=3&transport=polling&t=N2Rqmbu&sid=WvabNJgmFx4ujY3HAAAA HTTP/2.0 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:16:17:43 +0100] 200 -(192.168.1.120) GET /endpoint/ui/socket.io/?EIO=3&transport=polling&t=N2RqmaD&sid=WvabNJgmFx4ujY3HAAAA HTTP/2.0 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:16:17:44 +0100] 200 -(192.168.1.120) GET /endpoint/ui/socket.io/?EIO=3&transport=polling&t=N2Rqmf4&sid=WvabNJgmFx4ujY3HAAAA HTTP/2.0 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)

2 Mar 16:18:17 - [error] [modbus-read:Current] Timed out
2 Mar 16:18:17 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Current] FSM Reset On State {"value":"failed","actions":[],"changed":true}
[... cut ...]
2 Mar 16:18:17 - [warn] [modbus-read:Power] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Energy Imp] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Energy Imp] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true}
2 Mar 16:18:17 - [warn] [modbus-read:Voltage] FSM Reset On State {"value":"failed","actions":[],"changed":true}

```

biancode, 2020-03-02:
@PierreScerri sorry, nope ... add modbus-serial* to DEBUG env variable and try again, please. You should see all modbus in and out with that. I want to see if there are two failures or if it is the two working queues for the two unit-ids

PierreScerri, 2020-03-02:
For troubleshooting, I have disabled all flows except the modbus flow.
Seems to have stabilised.  The FSM reset on State errors only come when the flows are restarted.
Duplication has gone.

I am now getting this:

```
Log
13 packages are looking for funding
  run `npm fund` for details
found 0 vulnerabilities
[cont-init.d] node-red.sh: exited 0.
[cont-init.d] user.sh: executing... 
[cont-init.d] user.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[16:54:41] INFO: Starting Node-RED...
> addon-node-red@ start /opt
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"
2 Mar 16:54:45 - [info] 
Welcome to Node-RED
===================
2 Mar 16:54:45 - [info] Node-RED version: v1.0.4
2 Mar 16:54:45 - [info] Node.js  version: v12.15.0
2 Mar 16:54:45 - [info] Linux 4.19.93-v7 arm LE
2 Mar 16:54:46 - [info] Loading palette nodes
2 Mar 16:54:47 - [debug] Module: node-red-contrib-home-assistant-websocket 0.21.3
2 Mar 16:54:47 - [debug]         /config/node-red/node_modules/node-red-contrib-home-assistant-websocket
2 Mar 16:54:47 - [debug] Module: node-red-node-timeswitch 0.0.7
2 Mar 16:54:47 - [debug]         /config/node-red/node_modules/node-red-node-timeswitch
2 Mar 16:54:47 - [debug] Module: node-red-contrib-actionflows 2.0.3
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-actionflows
2 Mar 16:54:47 - [debug] Module: node-red-contrib-alexa-home-skill 0.1.17
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-alexa-home-skill
2 Mar 16:54:47 - [debug] Module: node-red-contrib-bigtimer 2.2.6
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-bigtimer
2 Mar 16:54:47 - [debug] Module: node-red-contrib-cast 0.2.14
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-cast
2 Mar 16:54:47 - [debug] Module: node-red-contrib-counter 0.1.5
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-counter
2 Mar 16:54:47 - [debug] Module: node-red-contrib-home-assistant-websocket 0.21.3 *ignored due to local copy*
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-home-assistant-websocket
2 Mar 16:54:47 - [debug] Module: node-red-contrib-http-request 0.1.14
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-http-request
2 Mar 16:54:47 - [debug] Module: node-red-contrib-influxdb 0.4.0
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-influxdb
2 Mar 16:54:47 - [debug] Module: node-red-contrib-interval-length 0.0.4
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-interval-length
2 Mar 16:54:47 - [debug] Module: node-red-contrib-looptimer 0.0.8
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-looptimer
2 Mar 16:54:47 - [debug] Module: node-red-contrib-modbus 5.2.0
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-modbus
2 Mar 16:54:47 - [debug] Module: node-red-contrib-moment 3.0.3
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-moment
2 Mar 16:54:47 - [debug] Module: node-red-contrib-state-machine 1.2.0
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-state-machine
2 Mar 16:54:47 - [debug] Module: node-red-contrib-statistics 2.2.2
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-statistics
2 Mar 16:54:47 - [debug] Module: node-red-contrib-stoptimer 0.0.7
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-stoptimer
2 Mar 16:54:47 - [debug] Module: node-red-contrib-sunevents 2.0.2
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-sunevents
2 Mar 16:54:47 - [debug] Module: node-red-contrib-time-range-switch 1.0.0
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-time-range-switch
2 Mar 16:54:47 - [debug] Module: node-red-contrib-timecheck 1.1.0
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-timecheck
2 Mar 16:54:47 - [debug] Module: node-red-contrib-traffic 0.2.1
2 Mar 16:54:47 - [debug]         /opt/node_modules/node-red-contrib-traffic
2 Mar 16:54:47 - [debug] Module: node-red-dashboard 2.19.4
[... 62 lines cut ...]
2 Mar 16:54:58 - [info] Started flows
2 Mar 16:55:01 - [info] [server:Home Assistant] WebSocket Connecting http://supervisor/core
2 Mar 16:55:02 - [info] [server:Home Assistant] WebSocket Connected to http://supervisor/core
2 Mar 16:55:02 - [debug] [server:Home Assistant] config server event listener connected
[02/Mar/2020:17:00:18 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /icons/node-red/debug.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:17:00:18 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /icons/node-red/debug.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:17:00:19 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /red/images/typedInput/expr.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:17:00:29 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /icons/node-red/debug.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[... cut ...] 3987.116 Safari/537.36)
[02/Mar/2020:17:00:55 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /icons/node-red/debug.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)
[02/Mar/2020:17:00:55 +0100] 200 192.168.1.120, 172.30.32.1(172.30.32.2) GET /icons/node-red/debug.svg HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.116 Safari/537.36)

```

xale12, 2020-04-01:
I have the same issue as you guy what should I do now T-T.

PierreScerri, 2020-04-01:
I have completely eliminated all Modbus errors by triggering the Getter Nodes sequentially:

[image]

The inject node triggers every 10s the first Modbus getter, this triggers the next and that triggers the next, etc.

The Sunset/sunrise node stops triggering unwanted Getters when it is dark.

xale12, 2020-04-01:
[{"id":"94fdec20.2c691","type":"modbus-write","z":"9a9b9e8d.7e73e","name":"Q0.0","showStatusActivities":true,"showErrors":true,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"1","server":"af4f96d3.3cb678","x":450,"y":740,"wires":[["74ad9436.0fb06c"],["74ad9436.0fb06c"]]},{"id":"af4f96d3.3cb678","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"192.168.1.39","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":true}]

xale12, 2020-04-01:
[{"id":"22233acf.4270e6","type":"function","z":"9a9b9e8d.7e73e","name":"16 Bit INT 16001","func":"msg.payload={};\nvar buf=Buffer.alloc(2);\nbuf.writeInt16BE(16001);\nmsg.payload.buffer=buf;\nreturn msg;","outputs":1,"noerr":0,"x":340,"y":1000,"wires":[["3fa5a00f.14015"]]}]

xale12, 2020-04-01:
I use the Modbus protocol to communicate between OpenPLC Master and Raspberry like a slave device ( I make raspberry like a PLC). When I control the simulation Modbus TCP /IP b MODSIM app  It run perfectly . But when I write to my open PLC analog PIN QW0, it does not accept my command.

PierreScerri, 2020-04-01:
Sorry can't help with that.  Don't know what it is.