BiancoRoyal/node-red-contrib-modbus #178: Uncaught Exception: Node-Red crashes after deploying incorrectly configured modbus read

schefa11, 2020-02-14:
**I'm submitting a ...**

## Problem
 - (x) bug report
 
 **Current behavior**
 Node-Red crashes after deploying incorrectly configured modbus read.
 
 **Expected behavior**
 Use default values or throw an error
 
 **Minimal reproduction of the problem with instructions**
1. Insert a Modbus Read Node.
2. Set up a valid Server (TCP or RTU).
3. Leave everything else (or at least on mandatory field) blank.
[image]
4. Ignore warnings and deploy.
5. Hit input of Read Node
6. Node-Red crashes:
14 Feb 16:32:25 - [red] Uncaught Exception:
14 Feb 16:32:25 - TypeError: this.internalDebug is not a function
    at Object.callModbus (XXXXXXX.node-red\node_modules\node-red-contrib-modbus\modbus\maps\core\core\modbus-client-core.js:154:10)
    at Object.sendDataInParallel (XXXXXXX\.node-red\node_modules\node-red-contrib-modbus\modbus\maps\core\core\modbus-queue-core.js:76:15)
    at Object.sequentialDequeueCommand (XXXXXXX\.node-red\node_modules\node-red-contrib-modbus\modbus\maps\core\core\modbus-queue-core.js:32:4)
    at Object.dequeueCommand (XXXXXXX\.node-red\node_modules\node-red-contrib-modbus\modbus\maps\core\core\modbus-queue-core.js:146:10)
    at Timeout._onTimeout (XXXXXXX\.node-red\node_modules\node-red-contrib-modbus\modbus\maps\modbus-client.js:162:10)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
  
  
  * Tested on following devices :
    
    - (x) Linux
    - (x) Windows
   
  * Description of the connecting Modbus system:
    
    - (x) No matter
    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v10.16.3

biancode, 2020-03-11:
[image]

closed 2020-03-28 by biancode

biancode, 2022-07-29:
With v3.0.x of node-red we decided to handle errors in our package from v5.22.x+

commit a0389e7b (2020-03-11): fix: #178 too get no crash on internal debug log

feat: #180 option to send empty msg on Modbus failure
chore: update version because of eflexsystems release clone
commit ff691343 (2020-03-11): fix: #178 debug per node to see where it is not correct or fallback core log

feat: link local development via NPM