BiancoRoyal/node-red-contrib-modbus #179: modbus-queue-info node is not resetting the queue when msg.resetQueue = true arrive

gugani, 2020-02-21:
**I'm submitting a ...**
## Feature
 - ( ) feature request
 
## Simple question to get some hint
 - ( ) simple question
 - ( ) help wanted

## Problem
 - (x) bug report
 - ( ) unexpected behavior
 
 **Current behavior**
 modbus-queue-info is not resetting the queue when `msg.resetQueue = true` arrive. Debug shows `TypeError: a.initQueue is not a function`
 
 **Expected behavior**
modbus-queue-info should reset the queue when `msg.resetQueue = true` arrive.
 
 **Minimal reproduction of the problem with instructions**
   1. Create a Modbus read over RTU and enable Queue commands
   2. Use modbus-queue-info node to check the items in the queue
   3. Send `msg.resetQueue = true` to the modbus-queue-info node
   4. Check debug tab messages.
   
### Your Environment
  * (x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device : BeagleBone Black
    
    - ( ) macOS: v 
    - (x) Linux: v Debian Buster 10.2
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Custom ARM boards
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v10.19.0

biancode, 2020-04-01:
please, test v5.6.0

biancode, 2020-04-09:
reopen if you find more, please

closed 2020-04-09 by biancode