BiancoRoyal/node-red-contrib-modbus #180: Asynchronous Functionality Breaks due to Timeouts

drowsylogic, 2020-02-21:
**I'm submitting a ...**

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
When attempting to setup an asynchronous Modbus Read to an RTU slave:
GlobalFunction (function-node) => Modbus-Flex-Getter-Node => CallbackNode (function-node)
CallGlobalFunction (function-node)

Node-RED Sample Workflow (working only is slave exists and doesn't timeout):
```
[
    {
        "id": "83baa70e.e903d8",
        "type": "tab",
        "label": "Flow 4",
        "disabled": false,
        "info": ""
    },
    {
        "id": "5117a231.9bda0c",
        "type": "function",
        "z": "83baa70e.e903d8",
        "name": "run callback",
        "func": "msg.callback(msg.payload);",
        "outputs": "0",
        "noerr": 0,
        "x": 650,
        "y": 140,
        "wires": []
    },
    {
        "id": "759653a.244dcac",
        "type": "inject",
        "z": "83baa70e.e903d8",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "",
        "crontab": "",
        "once": true,
        "x": 90,
        "y": 140,
        "wires": [
            [
                "5d0de017.290a9"
            ]
        ]
    },
    {
        "id": "5d0de017.290a9",
        "type": "function",
        "z": "83baa70e.e903d8",
        "name": "set readModbus",
        "func": "var readModbus = function(unitid, address, quantity, callback) {\n    fc = 3;\n    unitid = unitid || 1;\n    address = address || 0;\n    quantity = quantity || 1;\n    \n    \n    // modbus request\n    msg= {payload:{fc, unitid, address, quantity}};\n    \n    // assign callback to msg\n    msg.callback = callback;\n    \n    // return copy of msg\n    node.send(Object.assign({}, msg));\n};\n\n\n// make global\nglobal.set('readModbus', readModbus);",
        "outputs": 1,
        "noerr": 0,
        "x": 260,
        "y": 140,
        "wires": [
            [
                "9202417e.dbc51"
            ]
        ]
    },
    {
        "id": "dbe64ea3.f3e2e",
        "type": "inject",
        "z": "83baa70e.e903d8",
        "name": "",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "repeat": "0.5",
        "crontab": "",
        "once": false,
        "onceDelay": "",
        "x": 90,
        "y": 200,
        "wires": [
            [
                "866b7315.7bfe6"
            ]
        ]
    },
    {
        "id": "866b7315.7bfe6",
        "type": "function",
        "z": "83baa70e.e903d8",
        "name": "run readModbus(unitid, address, quantity, callback)",
        "func": "global.get('readModbus')(95, 26, 2, function (mbresponse) {\n    msg.payload = mbresponse;\n    node.send(msg);\n});",
        "outputs": 1,
        "noerr": 0,
        "x": 370,
        "y": 200,
        "wires": [
            [
                "c54cf931.be87c8"
            ]
        ]
    },
    {
        "id": "c54cf931.be87c8",
        "type": "debug",
        "z": "83baa70e.e903d8",
        "name": "",
        "active": true,
        "console": "false",
        "complete": "false",
        "x": 670,
        "y": 200,
        "wires": []
    },
    {
        "id": "9202417e.dbc51",
        "type": "modbus-flex-getter",
        "z": "83baa70e.e903d8",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "logIOActivities": false,
        "server": "4e882045.79912",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 450,
        "y": 140,
        "wires": [
            [
                "5117a231.9bda0c"
            ],
            []
        ]
    },
    {
        "id": "9a8025fd.e44728",
        "type": "modbus-queue-info",
        "z": "83baa70e.e903d8",
        "name": "",
        "topic": "",
        "unitid": "",
        "queueReadIntervalTime": "500",
        "lowLowLevel": 25,
        "lowLevel": 75,
        "highLevel": 150,
        "highHighLevel": 300,
        "server": "4e882045.79912",
        "errorOnHighLevel": true,
        "x": 450,
        "y": 80,
        "wires": [
            []
        ]
    },
    {
        "id": "4e882045.79912",
        "type": "modbus-client",
        "z": "",
        "name": "",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyM0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "115200",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "1000",
        "unit_id": "1",
        "commandDelay": "50",
        "clientTimeout": "100",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "200",
        "parallelUnitIdsAllowed": true
    }
]
```

The issue:
If the request reaches timeout or there is no response from the unit then no message gets sent out from the Modbus-Flex-Getter-Node. This event can happen because of a missing slave or because the slave took to long to respond (timeout).
 
**Expected behavior**
Message is sent from Modbus-Flex-Getter-Node but with an empty msg.payload and the msg.status with relevant information about the request.

 **Minimal reproduction of the problem with instructions**
 

1. Copy example workflow into node-red flow.
2. Click Inject node to create global function.
3. Click 2nd Inject node to call that global function.
4. Make sure that the values passed to the global modbus read function result in a timeout.
   
### Your Environment
**Please tell us about your environment:**

System: Debian 9
I am using the flex-getter as a modbus client.
Request is being sent to a Modbus RTU Slave Device over an RS485 Serial Interface. 
The slave device returns vibration levels.
Vendor: Machine Saver Inc.
Model: Tri-Vibe

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  * Device :
    - (X) Linux: v Kernel 4.4

   
   * Description of the connecting Modbus system:
   
     - Name   :  Tri-Vibe
     - Version:  112
     - Vendor : Machine Saver Inc
     - Link   : https://www.machinesaver.com
    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v12.2.0

drowsylogic, 2020-02-24:
Do you know how long it might take to implement this feature?
I have been given the okay from my company to sponsor this feature up to $200.

biancode, 2020-02-25:
@drowsylogic https://osi.bianco-royal.com/ - you need min. business support for that, please.

drowsylogic, 2020-02-26:
Payment has been made.
Please let me know the next steps.
Thanks!

biancode, 2020-02-26:
@drowsylogic you've got an e-mail for the next steps. Thank you for the support!

biancode, 2020-03-11:
[image]

biancode, 2020-03-11:
[image]

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode