BiancoRoyal/node-red-contrib-modbus #181: Flex-Getter not working expected

rucksman, 2020-03-06:
My flow looks like this: A single timestamp injects a (not relevant) payload into 4 function nodes every 10 seconds. Each of these function nodes genererates a payload for a holding register (like msg.payload = { 'fc': 3, 'unitid': 3, 'address': 30867, 'quantity': 2 } ). Each of these function nodes injects its payload into its own Flex-Getter. The result of each Flex-Getter gets into a debug node.

My expectation was that every 10 seconds I get all 4 values. But instead I get only one value every 10 seconds, so that after 40 seconds I finally received all 4 seconds. Is this the expected behaviour or do I do something worng?

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode

rucksman, 2020-04-14:
Tried it with 5.10.1. Now I get the error message:
`TypeError: Cannot read property 'status' of undefined`

biancode, 2020-04-14:
@rucksman I got an e-mail - was it a mistake or do you have the problem with "status"?

rucksman, 2020-04-20:
Kind of a mistake... when I tested after upgrade the problem seemed to persist, but after testing again everything worked fine. I already reopened this ticket after my first test but deleted my post right after my second test, so probably that's why you got en email. Sorry, but this ticket can be closed. Thanks for asking!