BiancoRoyal/node-red-contrib-modbus #182: Modbus TCP - Unexpected issue

O14P, 2020-03-10:
## Problem
 - unexpected behavior

 **Current behavior**
 
 I'm currently setting-up some Modbus Read nodes to read-out some values via Modbus TCP from differente devices. For one of the devices (Lennox HVAC machine - ReadLennox node below) no issues at all, but I can't read out any values from our PLC (Wago - ReadPLC node below). However I can read out both devices with a Modbus Poll tool on my PC using the same addresses.  

I'm getting following messages from the ReadPLC node:
`PortNotOpenError: Port Not Open`
`FSM Reset On State {"value":"failed","actions":[],"changed":true}`

 **Expected behavior**
 
Seeing that I'm able to read both Modbus slaves from a poll tool, I should be able to read out both slaves from the Modbus Read, that doesn't seem to be the case. Any tips on how to fix this? 

 **Test flow**
 
`[
    {
        "id": "8c891858.890aa8",
        "type": "tab",
        "label": "Flow 2",
        "disabled": false,
        "info": ""
    },
    {
        "id": "a221a0b3.b2558",
        "type": "modbus-read",
        "z": "8c891858.890aa8",
        "name": "ReadLennox",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "unitid": "110",
        "dataType": "HoldingRegister",
        "adr": "37",
        "quantity": "7",
        "rate": "5",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "7b4f9c2f.b0b814",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 150,
        "y": 80,
        "wires": [
            [],
            []
        ]
    },
    {
        "id": "273fcc14.078734",
        "type": "modbus-read",
        "z": "8c891858.890aa8",
        "name": "ReadPLC",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "unitid": "14",
        "dataType": "HoldingRegister",
        "adr": "0",
        "quantity": "1",
        "rate": "5",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "9fecc378.d3287",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "x": 140,
        "y": 140,
        "wires": [
            [],
            []
        ]
    },
    {
        "id": "7b4f9c2f.b0b814",
        "type": "modbus-client",
        "z": "",
        "name": "LennoxA",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "192.168.30.110",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "unit_id": 110,
        "commandDelay": 1,
        "clientTimeout": 1000,
        "reconnectOnTimeout": true,
        "reconnectTimeout": 2000,
        "parallelUnitIdsAllowed": true
    },
    {
        "id": "9fecc378.d3287",
        "type": "modbus-client",
        "z": "",
        "name": "PLC",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "192.168.30.14",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "unit_id": 14,
        "commandDelay": 1,
        "clientTimeout": 1000,
        "reconnectOnTimeout": true,
        "reconnectTimeout": 2000,
        "parallelUnitIdsAllowed": true
    }
]`
   
### Your Environment

Using via a Docker container, installed via nodered/node-red. Installed node-red-contrib-modbus 5.2.0 via the Palette Manager.
  
  * Device
    - Docker 19.03.07

   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.2.0

* **Node:** (min. 10.x)
   `node --version` = v10.19.0

schefa11, 2020-03-13:
Regarding to your error message I think another modbus client is already connected to your PLC's (-e.g. one of your poll tools). Common problem is that the connection is still established even if the tool is closed. Restart your PC and try it directly with Node-Red.

biancode, 2020-03-13:
try `DEBUG=contribModbus*,modbus-serial node-red` to see more about the Modbus problem, please see [DEBUG Wiki](https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG)

schefa11, 2020-03-13:
I tried to run node-red in the debug-mode, too. But unfortunately the commands did not work.
Do I have to take care of anything else? (I installed debug-package)

biancode, 2020-03-13:
try `DEBUG=contribModbus*,modbus-serial node-red -v > interesting.log`

schefa11, 2020-03-16:
Was my fault due to missing knowledge... .I have a Node-Red Installation on my Windows PC.
Running the debug-mode in command prompt requires the corresponding environment variable:
SET DEBUG=contribModbus*

Same for powershell:
$env:DEBUG = "contribModbus*"

biancode, 2020-03-16:
to get more see [debug](https://www.npmjs.com/package/debug)

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode