BiancoRoyal/node-red-contrib-modbus #184: Problem

mihnea0071, 2020-03-24:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (x ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - () unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
When i press the inject node from the ModBus Flex Writer it appears that data is send to my slave.
But when i read from that registers is empty... and lately i got a new error "FSM Reset On State {"value":"failed","actions":[],"changed":true}".
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
[image]
   2. 
[image]

   3. 
[image]
  
[image]
The number of the registers from the documentation of the device.

   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * (x) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: 10 Pro 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : AL1940
     - Version: 2.1.28
     - Vendor : ifm electronic
     - Link   : https://www.ifm.com/na/en/product/AL1940
    
* **node-red-contrib-modbus version:** vX.X.X

* **Node:** (min. 1.03)
   `node --js` = v12.6
The code from my flow:

[{"id":"3e103dd4.997482","type":"tab","label":"ModBus","disabled":false,"info":""},{"id":"d42c447c.9439d8","type":"inject","z":"3e103dd4.997482","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":280,"wires":[["c0b2d455.5cfec8"]]},{"id":"c0b2d455.5cfec8","type":"function","z":"3e103dd4.997482","name":"Adress","func":"msg.payload = {'fc': 3, \n'unitid': 1, \n'address': 600,\n'quantity': 2\n};\nreturn msg","outputs":1,"noerr":0,"x":300,"y":280,"wires":[["37624811.6c39e8"]]},{"id":"5911be9b.c537d","type":"debug","z":"3e103dd4.997482","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":670,"y":300,"wires":[]},{"id":"40386fb3.f65b","type":"inject","z":"3e103dd4.997482","name":"","topic":"","payload":"100","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":170,"y":140,"wires":[["70c2e64d.752688"]]},{"id":"16885620.2c8f0a","type":"function","z":"3e103dd4.997482","name":"Adress","func":"msg.payload = {value: msg.payload,\n   'fc': 16, \n   'unitid': 1, \n   'address': 600,\n   'quantity': 2 \n};\nreturn msg\n","outputs":1,"noerr":0,"x":520,"y":140,"wires":[["a3f3bc19.fd994"]]},{"id":"70c2e64d.752688","type":"join","z":"3e103dd4.997482","name":"","mode":"custom","build":"array","property":"payload","propertyType":"msg","key":"topic","joiner":"\\n","joinerType":"str","accumulate":false,"timeout":"","count":"2","reduceRight":false,"reduceExp":"","reduceInit":"","reduceInitType":"","reduceFixup":"","x":340,"y":140,"wires":[["16885620.2c8f0a"]]},{"id":"848e6e82.76015","type":"modbus-response","z":"3e103dd4.997482","name":"","registerShowMax":20,"x":970,"y":80,"wires":[]},{"id":"38a80424.fff55c","type":"modbus-response","z":"3e103dd4.997482","name":"","registerShowMax":20,"x":790,"y":240,"wires":[]},{"id":"a3f3bc19.fd994","type":"modbus-flex-write","z":"3e103dd4.997482","name":"","showStatusActivities":false,"showErrors":false,"server":"d9cd150a.8fb568","x":700,"y":140,"wires":[[],["848e6e82.76015"]]},{"id":"37624811.6c39e8","type":"modbus-flex-getter","z":"3e103dd4.997482","name":"","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"d9cd150a.8fb568","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":490,"y":280,"wires":[[],["5911be9b.c537d","38a80424.fff55c"]]},{"id":"d9cd150a.8fb568","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.0.250","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":true}]

biancode, 2020-03-26:
maybe duplicate in another case of issue #182

biancode, 2020-03-28:
Please, test v5.4.0 and reopen with new information if the problem is not fixed.

closed 2020-03-28 by biancode