BiancoRoyal/node-red-contrib-modbus #185: "Client -> stopped state without reconnecting" - on "Modified Flows/Nodes" Deploy

mikeS7, 2020-04-01:
**I'm submitting a ...**

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 Read Modbus nodes (checked with modbus-read & modbus-flex-getter) lock in stopped state on "Modified Flows/Nodes" Deploy. No issues with Full Deploy.

 **Expected behavior**
 
 Node should be in active state after deploy.

 **Minimal reproduction of the problem with instructions**
 
   1. Add and configure modbus-read node, I'm using ModbusTCP
   2. Deploy
   3. Change pull rate and select Modified Flows or Nodes deploy
   4. "Client -> stopped state without reconnecting" message is appeared in debug pane and console
   
As workaround second deploy can be used or disable/enable of configuration node with deploy is also helpful.

### Your Environment
**Please tell us about your environment:**

ModbusTCP is in use
  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.4.0

* **Node:** (min. 10.x)
   `node --version` = v10.19.0

biancode, 2020-04-01:
please, test 5.5.1 - there was an build error in that release ...

mikeS7, 2020-04-01:
@biancode 
Still something wrong:

- on fresh Node-RED start node has status "connected" instead of "active", but is working
[image]

- on any type of deploy there are warning
[image]
after several deploys last message repeats many times in a row
[image]

biancode, 2020-04-01:
Yes, there is a new state concept behind. It goes to active after the first Modbus request to the client and it is cleaned up to show the states step by step again. That was broken for a while.

TCP can now work in parallel with UnitIds or UnitId by UnitId as they are queued.
That brings all the changes to the FSM (StateMachine and Status handling)  ;)

[image]

mikeS7, 2020-04-01:
One more thing, my device supports only 4 connections, I've got repeating of "Client -> init in 2000 ms TCP@192.168.7.168:502 default Unit-Id: 1" after 5th deploy.
There are multiple connections in TIME_WAIT state
tcp        0      0 e3108ba6f498:58636      192.168.7.168:502       TIME_WAIT
tcp        0      0 e3108ba6f498:58662      192.168.7.168:502       TIME_WAIT
tcp        0      0 e3108ba6f498:58648      192.168.7.168:502       TIME_WAIT
tcp        0      0 e3108ba6f498:58664      192.168.7.168:502       TIME_WAIT
tcp        0      0 e3108ba6f498:58652      192.168.7.168:502       TIME_WAIT

mikeS7, 2020-04-01:
with "UnitId's in parallel" unchecked it doesn't work
my settings:
[image]

biancode, 2020-04-01:
> One more thing, my device supports only 4 connections, I've got repeating of "Client -> init in 2000 ms TCP@192.168.7.168:502 default Unit-Id: 1" after 5th deploy.
> There are multiple connections in [...]

could you try this with v5.6.0 - it is on the way - hope there is no compiling problem for the release

mikeS7, 2020-04-01:
unfortunately still have an issue, will test more tomorrow
also node send multiple messages instead one after several redeploy

looks like an issue with number of connections, will check tomorrow

thank you for your work

biancode, 2020-04-02:
please, Test v5.7.0 and request new issue if it has more issues. Thank you for testing ;)

closed 2020-04-02 by biancode

mikeS7, 2020-04-02:
Thank you,

looks good now, but create another one #187