BiancoRoyal/node-red-contrib-modbus #187: Several message in a row after redeploy

mikeS7, 2020-04-02:
**I'm submitting a ...**
## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 There are multiple output messages in a row after several "Modified Nodes" deploy

 **Expected behavior**
 
 Should be only one message per read request

 **Minimal reproduction of the problem with instructions**
 
   1. modbus-read (ModbusTCP) node configured with 2 seconds poll rate
   2. change pull rate to 1 second and perform "Modified Nodes" deploy
   3. there are one output message with another 2 in a row
   4. change pull rate to 2 seconds and perform "Modified Nodes" deploy
   5. there are one output message with another 3 in a row
   6. and so on
   I can record animated gif if you would prefer.

### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.7.0

* **Node:** (min. 10.x)
   `node --version` = v10.19.0

biancode, 2020-04-02:
hm, I can not reproduce it on three devices (Linux VM, macOS, RPI)

biancode, 2020-04-02:
Try to uninstall, call `npm cache verify` and install it again, please! A deploy sets the client to STOP state and from there it can not do anything. Is it a special device (power, cpu's?) I tested with nodejs 10.19.0 too.

biancode, 2020-04-02:
[see me loom](https://www.loom.com/share/f1f38e2c542c4a1dbd6c589e6bbab307)

mikeS7, 2020-04-02:
I see ... you just press FULL deploy, try Modified Nodes only
[image]

mikeS7, 2020-04-02:
I think something wrong with config node reinitialization on part deploy

biancode, 2020-04-08:
https://www.loom.com/share/34c5e50554414f4e9878c50f72833373

mikeS7, 2020-04-08:
I've got different behaviour, will record later today.
@biancode which version did you test on last loom? 5.8.0 or some from develop branch?

biancode, 2020-04-09:
should work with v5.9.0 as expected again

mikeS7, 2020-04-10:
Sorry, it doesn't. Freeze in wait state after deploy.

[image]

biancode, 2020-04-10:
Ah I had another node in testing which made the restart on read from mb-client events :(  :)

biancode, 2020-04-10:
It works now also with just a node. It is not the best practice in all cases to deploy partial it gets very complex with physical connection in some cases. I'd prefer the full deploy on every change.
Test v5.10.1, please

mikeS7, 2020-04-10:
Thanks a lot, it is working.
I don't want to open dispute about deploy types ;)

closed 2020-04-10 by mikeS7