BiancoRoyal/node-red-contrib-modbus #188: Modbus-client does not accept Unid-Id 0

jbofill, 2020-04-13:
(x) unexpected behavior 
 
 **Current behavior**

Connect to a device using modbus-tpc connection with unit-id 0.
When connecting, modbus-client will change the unit-id to 1 (default unit-id), and the device will not respond.

 **Expected behavior**

 Modbus-client should accept unid-id 0; device responds.

 **Environment** 

Raspbian 10 buster
Node-RED v1.0.5
node-red-contrib-modbus v5.10.1
Device connecting Modbus system:
   
     - Name   : Sun2000l
     - Version:
     - Vendor : Huawei 
     - Link   : https://support.huawei.com/enterprise/en/doc/EDOC1100011912

boraozgen, 2020-05-12:
I have the same issue and am working on a fix. A further issue is that broadcasting using Unit ID = 0 returns a timeout error when there is no response. It should not expect a response as it is a broadcast message.

This behavior is documented on page 7 of the spec:
http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

Will issue a pull request soon.

biancode, 2020-05-12:
@boraozgen in the last version released it is possible to set 0 for TCP, but to have a support for broadcasting is more a feature request here, because it was not in plan to support it until now ;)
Please open a new feature request for that!

boraozgen, 2020-05-12:
The broadcast address 0 is specified also for the serial mode.

biancode, 2020-05-17:
released with v5.11 for TCP

closed 2020-05-17 by biancode