BiancoRoyal/node-red-contrib-modbus #191: "TypeError: d.pushToQueueByUnitId(...).then(...).catch(...).finally is not a function" reported after node red modbus update

christianhenriksen, 2020-04-18:
After I updated the modbus on my node red some days ago, I got this error:

4/18/2020, 2:30:29 PMnode: d3b1b88d.921aa8
msg : error
"TypeError: d.pushToQueueByUnitId(...).then(...).catch(...).finally is not a function"

The only reason I am putting this in as a bug here is because this is the only place I could find the "d.pushToQueueByUnitId" in github. AFter I updated the modbus in my palette, I am not anymore able to read the ACTI9 modbus unit. I can see this unit was updated 17 days ago, so it migt be related to that release? I am way to much of a novice to add meaningfull comments, but if you let me know how to, I can poke into my system and see?

Thanks a lot. 
Christian

**I'm submitting a ...BUG?

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X

* **Node:** (min. 10.x)
   `node --version` = vX.X.X

schefa11, 2020-04-20:
Just a hint:
Did you try to deinstall the old version berfore installing the new one?
Once I have had a problem after an update, too.
Anyway you should describe your environment as you can see in your "bug-description template". Maybe you are still using the "old" Node-Red version (<1) or some packages does not match.

biancode, 2020-04-20:
It looks like the problem with a wrong node.js version. What is your node.js version, please?
see Node is min. v10.x!

biancode, 2020-04-20:
maybe duplicate of #190

christianhenriksen, 2020-04-21:
Hi, 
First of all thanks for all the answers I have got allready. I will try to answer as good as I can.
I am running this on a QNAP server in DOCKER.
I then went into the log window, restarted it and I think the problem must be the node.js version unfortunately:
21 Apr 11:10:31 - [info] Node-RED version: v0.20.8                                                                                                                                                        
21 Apr 11:10:31 - [info] Node.js  version: v8.16.1                                                                                                                                                        
21 Apr 11:10:31 - [info] Linux 4.14.24-qnap x64 LE                                                                                                                                                        
21 Apr 11:10:33 - [info] Loading palette nodes                                                                                                                                                            
21 Apr 11:10:38 - [warn] rpi-gpio : Raspberry Pi specific node set inactive                                                                                                                               
21 Apr 11:10:38 - [warn] rpi-gpio : Cannot find Pi RPi.GPIO python library                                                                                                                                
21 Apr 11:10:51 - [info] Settings file  : /data/settings.js                                                                                                                                               
21 Apr 11:10:51 - [info] Context store  : 'default' [module=memory]                                                                                                                                       
21 Apr 11:10:51 - [info] User directory : /data                                                                                                                                                           
21 Apr 11:10:51 - [warn] Projects disabled : editorTheme.projects.enabled=false                                                                                                                           
21 Apr 11:10:51 - [info] Flows file     : /data/flows.json                                                                                                                                                
21 Apr 11:10:51 - [info] Server now running at http://127.0.0.1:1880/                                                                                                                                     
21 Apr 11:10:51 - [warn]     

I will try and see if there is any way at all that I can update that and get back to here if it worked.

biancode, 2020-04-21:
21 Apr 11:10:31 - [info] Node-RED version: v0.20.8
21 Apr 11:10:31 - [info] Node.js version: v8.16.1
not supported anymore - you can install older version via npm i package@version

biancode, 2020-04-21:
Your flows are maybe not compatible from 0.20 to 1.x of Node-RED - search the issues to get more info ;)

closed 2020-04-21 by biancode