BiancoRoyal/node-red-contrib-modbus #192: Modbus reconnect problem

Proamp, 2020-04-22:
## Problem
After starting Node-Red, if modbus slave is not connected, Modbus node(Read, Write, Flex getter etc.) send the message to physical interface(Serial) only once, and never send it again. If modbus slave is connected on start  Node-Red Modbus node start communicating with the slave, but if I disconnect slave(from RS485 interface) and connect it again communication disappeared forever before restart or redeploy Node-Red flows.
  
   
### Your Environment
I'm using 3 serial ports Raspberry PI4 (AMA0, AMA2 and AMA4) with isolated RS485 interfaces) simultaniously. 
 
  
  * Device : Raspberry PI4 2Gb
    
    - ( ) Raspbian: v4.19.97 (Buster Lite) 

   
   * Description of the connecting Modbus system:
   
     - Name   : MODBUS MCU x 6
     - Version: 1.33
     - Vendor : IOTON
     - Link   : https://hackaday.io/project/167043-multifunctional-modbus-controller

    
   node-red-contrib-modbus version: v5.8.0
   Node: v12.16.1

biancode, 2020-04-22:
Should be fixed with 5.10.1

Proamp, 2020-04-22:
> Should be fixed with 5.10.1 see https://github.com/BiancoRoyal/node-red-contrib-modbus/commits/master [919db9b](https://github.com/BiancoRoyal/node-red-contrib-modbus/commit/919db9bad7edbb0532783b893a969d8725a44eb6)

Thank you!

biancode, 2020-04-23:
@Proamp does it work well for you with 5.10.1?

Proamp, 2020-04-23:
> @Proamp does it work well for you with 5.10.1?

I will try today.

Proamp, 2020-04-23:
> @Proamp does it work well for you with 5.10.1?

Yes, now reconnection works fine an fast. Thank you.

closed 2020-04-23 by biancode