BiancoRoyal/node-red-contrib-modbus #195: Broadcast support for serial mode

boraozgen, 2020-05-12:
**I'm submitting a ...**

## Feature
 - (x) feature request
 
 **What is the motivation / use case for changing the behavior?**
Broadcast messages with Unit ID = 0 should be supported for serial mode. The master should not expect a response in case of a broadcast message.

This behavior is documented on page 7 of the spec:
http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

biancode, 2020-05-12:
Todo: [Modbus-Serial-Lib Check for UnitID 0](https://www.npmjs.com/package/modbus-serial )

biancode, 2020-05-12:
Todo: UnitID checking on flex nodes ( false vs. 0 needs to be number for UnitID)

biancode, 2020-05-12:
Todo: unit tests for UnitID 0

biancode, 2020-05-12:
https://github.com/BiancoRoyal/node-red-contrib-modbus/projects/4#card-38085398

biancode, 2020-05-17:
Please, send unstable PR's to branch develop!

biancode, 2020-05-21:
first release with v5.13.0

biancode, 2020-05-26:
@boraozgen could you test with v5.13.2 on multiple devices or with the Modbus Simulator for PC and serial connections? I get no response from broadcasts via UnitId 0.

boraozgen, 2020-05-27:
You are not supposed to get any responses. From spec:

> In broadcast mode, the master can send a request to all slaves. No response is returned to broadcast requests sent by the master.  The broadcast requests are necessarily writing commands.  All devices must accept the broadcast for writing function.  The address 0 is reserved to identify a broadcast exchange.

boraozgen, 2020-05-27:
Closing the issue as it works as of v5.13.1 for my setup.

closed 2020-05-27 by boraozgen

biancode, 2020-05-27:
@boraozgen thanks for your contribution :)