BiancoRoyal/node-red-contrib-modbus #198: Modbus fails if in between http request and response

gumulka, 2020-05-14:
## Problem
 - (x ) bug report 
 
Not sure if this is a problem with node red, or with this package, so, feel free to close it, if it is a node red problem.

 **Current behavior**
I have a http in node, which gets a request, transforms this to a payload modbus-getter understands and then want to pass the modbus response on as the http-response.

Node now fails with 
```
"TypeError: Cannot set property listening of #<Server> which has only a getter"
```
Which seems to be related to the fact, that both http-in and modbus-getter set the msg.req object.

I see, that modbus sends out the message, and receives an answer, but when sending out the message, it fails.
If the input comes from an inject node, there is no problem. 

 **Minimal reproduction of the problem with instructions**
[image]

Create a flow like in the image (issue does not allow json upload). Inject node is working. HTTP get is not.

Transform function:
```
msg.payload = Object.assign({}, msg.req.params);
return msg;
```

### Your Environment
**Please tell us about your environment:**
   
   * Description of the connecting Modbus system:   
     - Name   : a somehow working python script, but not the issue here.
    
* **node-red-contrib-modbus version:** v5.8.0
21bb78a
Modified the finally statement to work with Node-8

* **Node:** (min. 10.x)
   `node --version` = v8.16.1 (I know, it is not supported, but I hope, that is not the problem here)

biancode, 2020-05-14:
@gumulka you need 5.11.0 and the new keep properties option - see the new example for HTTP via Node-RED import -> Examples -> ...

gumulka, 2020-05-15:
I don't see the 5.11 tag here on github, only 5.10.1. Is it missing?

biancode, 2020-05-15:
Oh it was - normally it gets pushed automatically. Fixed

gumulka, 2020-05-15:
Okay, I updated to newest master code, used the example flow provided by your package and just changed the modbus-serial connection to my own.

```
TypeError: Cannot set property listening of #<Server> which has only a getter    at _clone (/usr/lib/node/node-red/node_modules/clone/clone.js:162:16)    at _clone (/usr/lib/node/node-red/node_modules/clone/clone.js:162:18)    at _clone (/usr/lib/node/node-red/node_modules/clone/clone.js:162:18)    at _clone (/usr/lib/node/node-red/node_modules/clone/clone.js:162:18)    at _clone (/usr/lib/node/node-red/node_modules/clone/clone.js:162:18)    at clone (/usr/lib/node/node-red/node_modules/clone/clone.js:202:10)    at Object.cloneMessage (/usr/lib/node/node-red/node_modules/@node-red/util/lib/util.js:91:17)    at Object.node (/usr/lib/node/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:410:61)    at BREAK (/usr/lib/node/node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-getter.js:58:11)    at /usr/lib/node/node-red/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-client-core.js:66:11
```

gumulka, 2020-05-15:
My bad, did not install 5.11, but 5.8 again... Did a messup in git. It is now working with 5.11.0 Thank you

closed 2020-05-15 by gumulka