BiancoRoyal/node-red-contrib-modbus #200: MODBUS Read Node's output msg object has no "topic" property.

gemini86, 2020-05-15:
**I'm submitting a ...**  (check one with "x")

## Problem
 - (**x** ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
Using Modbus Read Node: the output msg object has no "topic" property, no matter what (if anything) is inserted into the "topic" field in the config.
 
 **Expected behavior**
msg.topic should reflect the string in the config field. If no string is entered, "topic" property should not exist.
 
 **Minimal reproduction of the problem with instructions**
   1. Open config for MODBUS Read node.
   2. type something into the "Topic" config field.
   3. read output msg in the debug panel or try to reference "msg.topic". 
   
### Your Environment
**Please tell us about your environment:**
Windows server 2016 node-red server. 
Reading any modbus device.

  * (**x**) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (**x**) Windows: **Server 2016**
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : AL1340
     - Version:
     - Vendor : IFM
     - Link   :  https://www.ifm.com/us/en/product/AL1340
    
* **node-red-contrib-modbus version:** v5.11.0

* **Node:** (min. 10.x)
   `node --version` = v13.6.0

biancode, 2020-05-17:
It was 'polling' if empty and I just will put it into msg here.

biancode, 2020-05-17:
released with v5.12

closed 2020-05-17 by biancode