BiancoRoyal/node-red-contrib-modbus #202: Bug Sequential de-queuing

cdriouech, 2020-05-20:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
Blocking serial messages on Timeout and sequential queue.
 
 **Expected behavior**
It should not block the queue and every response should release with error if blocked.
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (X ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Node RED
     - Version: v1.0.3
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.10.1

* **Node:** (min. x.x)
   `node --version` = v12.2.0

biancode, 2020-05-21:
#195 is also interesting here to send only one msg to request all units

mastrikGit, 2020-05-24:
I have same error with setup which worked for many monthes.
Started to appear after update to 5.12
Still happens with 5.13 after random time running 4hrs>

Here is part of flow:

`[{"id":"86604079.996eb","type":"mqtt out","z":"6cb719ba.d8bff8","name":"","topic":"","qos":"2","retain":"true","broker":"6f159f68.33fd","x":1130,"y":80,"wires":[]},{"id":"ad4a4b5f.87ae48","type":"rbe","z":"6cb719ba.d8bff8","name":"","func":"rbe","gap":"","start":"","inout":"out","property":"payload","x":910,"y":80,"wires":[["86604079.996eb"]]},{"id":"ec724b86.f33658","type":"inject","z":"6cb719ba.d8bff8","name":"","topic":"","payload":"","payloadType":"date","repeat":"0.2","crontab":"","once":true,"onceDelay":0.1,"x":110,"y":120,"wires":[["36d79240.bacaee"]]},{"id":"29ce1ddc.dd3252","type":"modbus-flex-getter","z":"6cb719ba.d8bff8","name":"","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"d54264c3.dde5e8","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":false,"x":550,"y":80,"wires":[["fbb339d2.bdbf18"],["f757d3c4.e0ab3"]]},{"id":"36d79240.bacaee","type":"function","z":"6cb719ba.d8bff8","name":"","func":"var msg1 = {topic:\"1-100\"};\nvar msg2 = {topic:\"101-200\"};\n\nmsg1.payload ={\n'fc': 3,\n'unitid': 1,\n'address': 1 ,\n'quantity': 100 }\n\nmsg2.payload ={\n'fc': 3,\n'unitid': 1,\n'address': 101 ,\n'quantity': 100 }\nreturn [[msg1,msg2]];","outputs":1,"noerr":0,"x":230,"y":80,"wires":[["7631ecad.7c8694"]]},{"id":"22d83040.2c9df","type":"comment","z":"6cb719ba.d8bff8","name":"Read from modbus","info":"","x":100,"y":40,"wires":[]},{"id":"fbb339d2.bdbf18","type":"function","z":"6cb719ba.d8bff8","name":"Map to local","func":"// build an array of msgs, mapping topics to registers\n\nswitch(msg.topic) {\n    case '1-100':\n    var msgs = [\n        { topic: \"Home/Light/Dimmer1/q\", payload: msg.payload[0] },\n        { topic: \"Home/Light/Dimmer1/command\", payload: msg.payload[1] },\n        { topic: \"Home/Light/Dimmer1/val\", payload: msg.payload[2] },\n        { topic: \"Home/Light/Dimmer2/q\", payload: msg.payload[3] },\n        { topic: \"Home/Light/Dimmer2/command\", payload: msg.payload[4] },\n        { topic: \"Home/Light/Dimmer2/val\", payload: msg.payload[5] },\n        { topic: \"Home/Light/Dimmer3/q\", payload: msg.payload[6] },\n        { topic: \"Home/Light/Dimmer3/command\", payload: msg.payload[7] },\n        { topic: \"Home/Light/Dimmer3/val\", payload: msg.payload[8] },\n        { topic: \"Home/Light/Dimmer4/q\", payload: msg.payload[9] },\n        { topic: \"Home/Light/Dimmer4/command\", payload: msg.payload[10] },\n        { topic: \"Home/Light/Dimmer4/val\", payload: msg.payload[11] },\n        { topic: \"Home/Light/Dimmer5/q\", payload: msg.payload[12] },\n        { topic: \"Home/Light/Dimmer5/command\", payload: msg.payload[13] },\n        { topic: \"Home/Light/Dimmer5/val\", payload: msg.payload[14] },\n        { topic: \"Home/Light/Dimmer6/q\", payload: msg.payload[15] },\n        { topic: \"Home/Light/Dimmer6/command\", payload: msg.payload[16] },\n        { topic: \"Home/Light/Dimmer6/val\", payload: msg.payload[17] },\n        { topic: \"Home/Light/Dimmer7/q\", payload: msg.payload[18] },\n        { topic: \"Home/Light/Dimmer7/command\", payload: msg.payload[19] },\n        { topic: \"Home/Light/Dimmer7/val\", payload: msg.payload[20] },\n        { topic: \"Home/Light/Dimmer8/q\", payload: msg.payload[21] },\n        { topic: \"Home/Light/Dimmer8/command\", payload: msg.payload[22] },\n        { topic: \"Home/Light/Dimmer8/val\", payload: msg.payload[23] },\n        { topic: \"Home/Light/Dimmer9/q\", payload: msg.payload[24] },\n        { topic: \"Home/Light/Dimmer9/command\", payload: msg.payload[25] },\n        { topic: \"Home/Light/Dimmer9/val\", payload: msg.payload[26] },\n        { topic: \"Home/Light/Dimmer10/q\", payload: msg.payload[27] },\n        { topic: \"Home/Light/Dimmer10/command\", payload: msg.payload[28] },\n        { topic: \"Home/Light/Dimmer10/val\", payload: msg.payload[29] }, \n        { topic: \"Home/Light/Dimmer11/q\", payload: msg.payload[30] },\n        { topic: \"Home/Light/Dimmer11/command\", payload: msg.payload[31] },   \n        { topic: \"Home/Light/Dimmer11/val\", payload: msg.payload[32] },\n        { topic: \"Home/Light/Dimmer12/q\", payload: msg.payload[33] },\n        { topic: \"Home/Light/Dimmer12/command\", payload: msg.payload[34] },\n        { topic: \"Home/Light/Dimmer12/val\", payload: msg.payload[35] },    \n        { topic: \"Home/Light/Dimmer13/q\", payload: msg.payload[36] },\n        { topic: \"Home/Light/Dimmer13/command\", payload: msg.payload[37] },    \n        { topic: \"Home/Light/Dimmer13/val\", payload: msg.payload[38] },\n        { topic: \"Home/Light/Dimmer14/q\", payload: msg.payload[39] },    \n        { topic: \"Home/Light/Dimmer14/command\", payload: msg.payload[40] },\n        { topic: \"Home/Light/Dimmer14/val\", payload: msg.payload[41] },    \n        { topic: \"Home/Light/Relay1/q\", payload: msg.payload[51] },\n        { topic: \"Home/Light/Relay1/command\", payload: msg.payload[52] },    \n        { topic: \"Home/Light/Relay2/q\", payload: msg.payload[53] },\n        { topic: \"Home/Light/Relay2/command\", payload: msg.payload[54] },\n        { topic: \"Home/Light/Relay3/q\", payload: msg.payload[55] },\n        { topic: \"Home/Light/Relay3/command\", payload: msg.payload[56] }, \n        { topic: \"Home/Light/Relay4/q\", payload: msg.payload[ [... cut ...] modbus-client","z":"","name":"PCL","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"192.168.1.110","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":false,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]`

biancode, 2020-05-24:
@mastrikGit try to work with the units in parallel on tcp in your case - how does it work for you in parallel?

biancode, 2020-05-24:
[image]
on TCP

biancode, 2020-05-24:
please test with the new 5.13.1

mastrikGit, 2020-05-24:
Updated, set setting as you suggest. Will write back tomorrow with results.

Thank you for fast replies!

mastrikGit, 2020-05-26:
@biancode Thanks for fast fix. Uptime is more than 24hrs after update and no errors appear

closed 2020-05-26 by biancode