BiancoRoyal/node-red-contrib-modbus #204: modbus-flex-getter Port Not Open issue still persist when in TCP

tjunussov, 2020-06-01:
## Problem
 - (x ) bug report

My setup is like 

1) ModbusFlexServer with 0.0.0.0 1502 port binding to host ( Docker ) 
2) Separate flow which polls data via ModbusFlexRead every 1 minute
3) Separate flow which sets data every 10 minutes from Node Red Websocket Client
3) External Modbus Poll utility which polls sever every 10 seconds ( Read Coils Add 107-110, Read Holding 111 - 113 ) 
 
 **Current behavior**
After couple of hours mobus-tcp-server begins to have port not open issue, 
Error: Client Not Ready To Read At State reconnecting, and if I manually restart flow, it begin to work.
   
### Your Environment
**Please tell us about your environment:**
Docker, Node-red, ModbusTCP Server

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
     
* **node-red-contrib-modbus:** v5.13.2

And here is following log
> 1 Jun 16:35:36 - [warn] [modbus-write:Set Register 111] Modbus Failure On State sending Get More About It By Logging
> 2020-06-01T16:35:36.045Z contribModbus:queue:core queue serial lock command node bibit modbus tcp client
> 2020-06-01T16:35:36.045Z contribModbus:getter Port Not Open
> 1 Jun 16:35:36 - [error] [modbus-getter:Read F01 Coils 107-110] Port Not Open
> 2020-06-01T16:35:44.113Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 2020-06-01T16:36:36.045Z contribModbus:queue:core sequential de-queue command
> 2020-06-01T16:36:36.046Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 2020-06-01T16:36:36.046Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 1 Jun 16:36:36 - [warn] [modbus-write:Set Coil 107] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:36:36 - [warn] [modbus-write:Set Coil 109] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:36:36 - [warn] [modbus-write:Set Coil 107 [Slave 2]] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:36:36 - [warn] [modbus-write:Set Register 111] Modbus Failure On State sending Get More About It By Logging
> 2020-06-01T16:36:36.047Z contribModbus:queue:core queue serial lock command node bibit modbus tcp client
> 2020-06-01T16:36:36.047Z contribModbus:getter Port Not Open
> 1 Jun 16:36:36 - [error] [modbus-getter:Read F01 Coils 107-110] Port Not Open
> 2020-06-01T16:36:44.112Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 2020-06-01T16:37:36.045Z contribModbus:queue:core sequential de-queue command
> 2020-06-01T16:37:36.046Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 2020-06-01T16:37:36.046Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 1 Jun 16:37:36 - [warn] [modbus-write:Set Coil 107] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:37:36 - [warn] [modbus-write:Set Coil 109] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:37:36 - [warn] [modbus-write:Set Coil 107 [Slave 2]] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:37:36 - [warn] [modbus-write:Set Register 111] Modbus Failure On State sending Get More About It By Logging
> 2020-06-01T16:37:36.047Z contribModbus:queue:core queue serial lock command node bibit modbus tcp client
> 2020-06-01T16:37:36.047Z contribModbus:getter Port Not Open
> 1 Jun 16:37:36 - [error] [modbus-getter:Read F01 Coils 107-110] Port Not Open
> 2020-06-01T16:37:43.234Z contribModbus:queue:core sequential de-queue command
> 2020-06-01T16:37:43.236Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:37:43.236Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 1 Jun 16:37:43 - [warn] [modbus-flex-getter:fe9f8884.78b2a8] Modbus Failure On State sending Get More About It By Logging
> 1 Jun 16:37:43 - [warn] [modbus-flex-getter:fa926cd.ba0959] Modbus Failure On State sending Get More About It By Logging
> 2020-06-01T16:37:43.237Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:37:43.237Z contribModbus:flex:getter Port Not Open
> 1 Jun 16:37:43 - [error] [modbus-flex-getter:fe9f8884.78b2a8] Port Not Open
> 2020-06-01T16:37:44.113Z contribModbus:queue:core queue serial unlock command node bibit modbus tcp client
> 2020-06-01T16:37:44.741Z contribModbus:flex:getter Client Not Ready To Read At State reconnecting
> 1 Jun 16:37:44 - [error] [modbus-flex-getter:fe9f8884.78b2a8] Error: Client Not Ready To Read At State reconnecting
> 2020-06-01T16:37:44.741Z contribModbus:basics Client Not Ready To Read At State reconnecting
> 2020-06-01T16:37:47.239Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:37:47.240Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:37:51.244Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:37:51.244Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:37:55.248Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:37:55.248Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:37:59.251Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:37:59.251Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:38:03.253Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:38:03.253Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:38:07.256Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:38:07.256Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 2020-06-01T16:38:11.259Z contribModbus:queue:core queue serial unlock command node Modbus Flex Server
> 2020-06-01T16:38:11.259Z contribModbus:queue:core queue serial lock command node Modbus Flex Server
> 
>

tjunussov, 2020-06-01:
I would like to have advice, where I need to dig, resolve bug ? Is it yacov side or nodered side ?

tjunussov, 2020-06-03:
I'm ok even if I restart Flex Server via script "schedule or catch exception for port not open" as temp solution, but how it is possible with NodeRed ?

biancode, 2020-06-03:
good questions ... sorry I've no time to dig here ...