BiancoRoyal/node-red-contrib-modbus #205: Modbus flex getter not getting server details in its output

Kartheek1512, 2020-06-03:
I'm submitting a Simple question to get some hint

 **Current behavior**
As of now, the output of the modbus flex getter are:
1. payload
2. msg.id
3. input (unit id, fc, address, quantity) 
 
 **Expected behavior**
   I would like get the Host IP and port number that I set in the server settings too. 
 Wondering if that is possible.

biancode, 2020-06-03:
simple answer :) 
should be possible ...

closed 2020-06-03 by biancode

Kartheek1512, 2020-06-03:
Can you please let me know how can i get that info from modbus flex getter?

biancode, 2020-06-03:
@Kartheek1512 then it is a feature request and you could open an issue for that. Please, use the template to start your feature request! ;)

Kartheek1512, 2020-07-21:
HI Klaus,

  I just opened a feature request. I followed the template.. Can you please
check if my request is in queue :)?

Thanks,
Kartheek