BiancoRoyal/node-red-contrib-modbus #206: Parallel unit ID handling in Modbus RTU

boraozgen, 2020-06-03:
**I'm submitting a ...**  (check one with "x")

## Problem
 - ( x ) unexpected behavior
 
 **Current behavior**
Modbus RTU (serial mode) encounters timeout issues when using multiple slaves and when "Unit ids in parallel" is enabled in modbus-client (which is enabled by default). Issue is solved when this option is disabled.
 
 **Expected behavior**
As far as I know, Modbus RTU slaves should be addressed sequentially, not in parallel. Therefore this feature should not be allowed in serial mode.
 
 **Minimal reproduction of the problem with instructions**
   1. Configure for RTU operation.
   2. Leave "Unit ids in parallel" checked.
   3. Try to address multiple slaves at the same time.
   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
    
* **node-red-contrib-modbus version:** v5.13.1

* **Node:** (min. 10.x)
   `node --version` = v12.16.3

biancode, 2020-06-03:
it should get switched off on change to serial, but it should be free to configure

biancode, 2020-07-01:
could reopen with a PR

closed 2020-07-01 by biancode

keshavvasudeva, 2022-08-01:
I am trying to use multiple solo temperature controllers in series with 1 rs485 - usb. I'm not getting any errors in node red but I'm also not getting outputs. I am trying to extract data using the correct unitids and the correct parameter. Any ideas why?

biancode, 2022-08-02:
a) do not use parallel unit-ids option - this just for TCP
b) try to start with one, then two etc. 
c) try to play with the delay of the queue to send the messages with some delay like 200 (ms)