BiancoRoyal/node-red-contrib-modbus #208: Failed Connection to Modbus Master Kills Node-red

Ikoth, 2020-06-11:
**I'm submitting a ...**  (check one with "x")

## Problem
 - ( X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( X) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : OpenPLC
     - Version: v3
     - Vendor :
     - Link   : https://github.com/thiagoralves/OpenPLC_v3
    
* **node-red-contrib-modbus version:** v5.13.2

* **Node:** (min. 10.x)
   `node --version` = v12.18

biancode, 2020-06-11:
It is a Node-RED concept to crash on unhandled errors. This package follows Node-RED in that case. What is the state without a valid address - unknown or complex - so it stops. Feel free to send a pull request with your ideas of handling such errors see core-client and reopen ! ;)   btw there is a safe mode in node-red for such errors if you did some wrong configuration.

closed 2020-06-11 by biancode

Ekristoffe, 2020-06-12:
@biancode , i am sorry but I don't agree with the way you describe the error handling system of Node-RED.
Here is a quote of the official documentation:

> **Handling errors**
> Node-RED provides two ways for a node to report an error. It can either just write a message to the log or it can notify the runtime of the error and cause a flow to be triggered.
> 
> If the error is only written to the log, you will see the message in the Debug sidebar and log output, but you will not be able to create a flow to handle it. These are uncatchable errors.
> 
> If it does notify the runtime properly, then it is a catchable error that can be used to trigger an error-handling flow.
> 
> There is a third sort of error that can cause the Node-RED runtime to shutdown. These uncaughtException errors cannot be handled in the flow and are caused by bugs in nodes.
> 
> This guide describes each of these error types in more detail and shows what can be done to handle them.

I personally think it is cleaner to return something in a console / error log instead of killing the whole machine.
Having Node-RED crash at every error mean we lose the context each time. 
(Not the best option).
Does the core/library return error ? or just throw and goodbye ?

biancode, 2020-06-12:
@Ekristoffe I discussed it with the core team and you can find that on slack. It was there address in use and here we have nearly the same situation. As I wrote I am open for pull requests.

biancode, 2020-06-12:
@Ekristoffe what I remind is that I tried to catch, but it wont and Node-RED is crashing because it is a system error and not just software handling, but maybe I missed some at this point.

biancode, 2020-06-12:
see https://nodejs.org/api/errors.html

Ikoth, 2020-06-12:
Very helpful. Thanks.

biancode, 2022-07-29:
With v3.0.x of node-red we decided to handle errors in our package from v5.22.x+