BiancoRoyal/node-red-contrib-modbus #209: [object Object] error

PapaNoyel, 2020-06-25:
**I'm submitting an ...** 

 - ( x) unexpected behavior
 
 **Current behavior**
 
I'm pulling multiple information from different equipements (meteo station, solar panel, probe) by Modbus with Node-Red. I've placed a catch node on those flows. They are sending errors but the error message is always [object Object]
 
 **Expected behavior**
 
Giving more hints about the problem which trigger the catch node.
 
 **Minimal reproduction of the problem with instructions**
 
I've attached the minimal flow I've got (it's the txt since json files aren't accepted, just rename it "flow.json"), and the msg obtained from the catch node
```json
{"topic":"2b64786a.8cfc68",
"messageId":"5ef470bc50ffc531ec5718e8",
"payload":{
    "unitid":2,
    "fc":4,
    "address":30775,
    "quantity":2,
    "emptyMsgOnFail":false,
    "keepMsgProperties":false,
    "messageId":"5ef470bc50ffc531ec5718e8"},
    "queueLengthByUnitId":{
        "unitId":2,
        "queueLength":0},
        "queueUnitId":2,
        "unitId":2,
        "error":{"message":"[object Object]",
            "source":{"id":"2b64786a.8cfc68",
            "type":"modbus-flex-getter",
            "name":"Modbus Flex Getter  - Panneaux solaires",
           "count":1
        }
    },
    "_msgid":"de05c35e.2e76d"
}```
   
### Your Environment
**Please tell us about your environment:**
<!-- system, Modbus server or client type, package manager, equipment, vendor info -->
Node-Red 0.20.5 running on Docker,
Node-red-contrib-modbus 5.13.2

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
 
  
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: v Ubuntu 18.04
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   (Doing the same with every equipment but here is one)
     - Name   : Humidity and Temperature Probe with Modbus Interface
     - Version : 
     - Vendor : E+E Elektronik
     - Link   : http://downloads.epluse.com/fileadmin/data/product/ee071/datasheet_EE071.pdf
    
* **node-red-contrib-modbus version:** v5.13.2
<!-- Check which is the hash of the last commit from node-red-contrib-modbus that you have locally -->

* **Node:** (min. 10.x)
   `node --version` = v12.16.3
[flows.txt](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/4832320/flows.txt)

biancode, 2020-07-01:
Hi, try the community slack forum, please

biancode, 2020-07-01:
https://nodered.org/slack/

closed 2020-07-01 by biancode