BiancoRoyal/node-red-contrib-modbus #211: Error: Client Not Ready To Read At State init

josepj1986, 2020-07-08:
**Problem**
 - (x) bug report 

**My setup is like**
Separate flow which polls data via ModbusFlexRead every 5 second
 
**Current behavior**
After six or eight hours mobus-tcp-server begins to have port not open issue,
Error: Client Not Ready To Read At State init, after changing the acquisition times and restarting the flow, it starts working again.

  *** Device :**
    - (x) Linux: vUbuntu

**And here is following log**

{"message":"Error: Client Not Ready To Read At State init","source":{"id":"35195cd4.d005d4","type":"modbus-flex-getter","name":"Modbus Flexible Read","count":1},"stack":"Error: Client Not Ready To Read At State init\n    at Object.<anonymous> (/home/administrador/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:441:14)\n    at Object.emit (events.js:198:13)\n    at Object.emit (/usr/local/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:185:20)\n    at Object._inputCallback (/home/administrador/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-flex-getter.js:146:21)\n    at Object.Node._emitInput (/usr/local/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:200:18)\n    at Immediate._onImmediate (/usr/local/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:179:33)\n    at runCallback (timers.js:705:18)\n    at tryOnImmediate (timers.js:676:5)\n    at processImmediate (timers.js:658:5)"}

   * Description of the connecting Modbus system:
   
     - Name   : Inverter SMA communicate Modbus TCP
     - Vendor : SMA
     - Link   :  https://www.sma.de/es/productos/inversor-fotovoltaico/sunny-highpower-peak3.html
    
* **node-red-contrib-modbus version:** v5.3.13

* **Node:**
   node --version= v 1.1.0

Thanks for your help beforehand

AimeeQiu, 2020-08-04:
Hi,How did you solve this problem,please help me

sensbo, 2020-08-24:
Same issue here. Please fix it.

biancode, 2020-08-24:
Hi,

the message "Client Not Ready To Read At State init" is new to see if a request in your flow is going wrong and to analyze such problems. It just tells you: You read but the connection was not ready and in initialization state. That is often if it starts to read without waiting for some time (start delay) to get connected to your device physically. It could happen on connection is lost states and while reconnecting is doing the work to get ready again. Sending a request in Node-RED is possible at every moment and is not to handle in every case and it does not include the readiness of the Modbus-Client (which is the config node).

short words: set up start delays (5 to 10 sec), do not connect too much nodes to a client (check configuration node-counter), watch your events and check if and why the connection gets maybe lost over time and why you get those information

biancode, 2020-08-24:
btw. some Modbus devices are closing the connection if talk to fast to them ...

rucksman, 2020-08-25:
Well, thats is not totally true:

> in earlier version you just don't see it ;)

I simpy do not get any data, so it is not only a message that was not shown before, because I used to get data for a long time and without changng anything in my setup all of a sudden there is no data anymore.

biancode, 2020-08-25:
> Well, thats is not totally true:
> 
> > in earlier version you just don't see it ;)
> 
> I simpy do not get any data, so it is not only a message that was not shown before, because I used to get data for a long time and without changng anything in my setup all of a sudden there is no data anymore.
> 
> And in another application I run I always get data out of the same modbus device, so clearly there is an issue with node red.

to get back to the context here - I wrote: the message "Client Not Ready To Read At State init" is new to see if a request in your flow is going wrong and to analyze such problems - From my side there was no doubt about the problem.

josepj1986, 2020-08-25:
Good night, finally the Modbus server of sma inverter stopped responding when you made requests in the same second, you have to make requests separated from each other a few seconds by means of a loop. 

 If you need help I can give you a code example, make a loop every 150 seconds and without problems.

Thank you @biancode for your time and job.

biancode, 2020-08-26:
@josepj1986 did you use the queue delay of the config node or give it a try? The queue holds your requests and send it just after the delay one by one with a pause between. Default is 1 ms (ASAP) in the config node (Modbus Client).

sensbo, 2020-08-26:
@josepj1986  Ok, as I have understood your issue, the problem comes from the receiver side and wasn't an issuee from the modbus node. You add a kind of a loop (?) which avoid that issue? Because at my setup is sporadically issue, possibly also initialized by the receiver / modbus server.... I am still investigating the root cause.
@biancode What "config node" do you mean? In node-red there is a "delay node" which can be used for delaying messages. Do you mean this?

biancode, 2020-08-26:
@sensbo I mean to modbus configuration node, which is a Modbus client 
[image]

sensbo, 2020-08-26:
Thank you, I found it! I will give it try.

biancode, 2020-12-18:
looks like it is done

closed 2020-12-18 by biancode

rucksman, 2020-12-21:
Unfortunately it is not done, at least not for me. Still getting this error. It works only for some days, and then it stops working throwing this error.

In node red a flex getter node says "init" and a read node "initialized". My settings in the client node:
Timeout (ms): 1000
Reconnect bei Timeouts: true
Reconnect Timeout (ms): 2000
UnitIds in parallel: false
Log State Wechsel: false
Queue Logging: false
Queue Einträge: true
Queue Sendeverzögerung (ms): 200

I played around quite a bit with "Queue Sendeverzögerung", but nothing helped. Is there maybe a limit für getting modbus data? In my flow I have 7 flex getter nodes and 2 read nodes. Too much?

AndyMuc72, 2021-07-01:
Hi,
I have a similar problem.
The error is the same as it is descriped in here.
I used to get data from a Solaredge inverter through Modbus TCP (port mirroring on a switch) to my NodeRed on a Raspberry.
It was working quite well for around 2 years with your node.
For some some months after the night the Modbus state of the Solaredge inverter is 'ready' but not 'connected'.
To connect I need to restart node-red or change something in the flow and press 'deploy'. 
After this the inverter is always connected through Modbus TCP until the next morning. In the night my inverter is in 'night mode'
So something happens in the starting process of your node, some reconnection that is not done by the flow itself.

Is the problem solved and how?

Thanx for your help

emmanuel-landrivon, 2021-08-02:
Hi,
I may fix this reconnect issue adding the following nodes in //  : 

[image]

Seems working so far.
regards.
Hope it help

rucksman, 2021-08-02:
@emmanuel-landrivon  Would you mind posting your flow in JSON format as code? Otherwise I would have to guess a lot ...

emmanuel-landrivon, 2021-08-03:
Hi, 
no prob.
The trick here is : 
1) detect the error (init state)
2) send some parameters (IP/connexion type) to the modbus server --> it will restart the modbus server... and leave the init state !
Regards.

`[{"id":"43ee6ada.9b5b64","type":"modbus-flex-connector","z":"46dd0227.50b9cc","name":"","maxReconnectsPerMinute":4,"emptyQueue":false,"showStatusActivities":true,"showErrors":true,"server":"376f57fc.d2bf58","x":890,"y":800,"wires":[["661243cf.f9ac0c"]]},{"id":"181e491d.3c6377","type":"inject","z":"46dd0227.50b9cc","name":"","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":"5","topic":"","payload":"{}","payloadType":"json","x":430,"y":760,"wires":[["2d395433.b4833c"]]},{"id":"661243cf.f9ac0c","type":"debug","z":"46dd0227.50b9cc","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":1150,"y":800,"wires":[]},{"id":"f481a786.fb3df8","type":"change","z":"46dd0227.50b9cc","name":"HOST : 192.168.20.2","rules":[{"t":"set","p":"payload.tcpHost","pt":"msg","to":"192.168.20.2","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":650,"y":800,"wires":[["43ee6ada.9b5b64"]],"info":"test"},{"id":"2d395433.b4833c","type":"change","z":"46dd0227.50b9cc","name":"connectorType : tcp","rules":[{"t":"set","p":"payload.connectorType","pt":"msg","to":"TCP","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":640,"y":760,"wires":[["f481a786.fb3df8"]],"info":"test"},{"id":"11e30246.5371fe","type":"catch","z":"46dd0227.50b9cc","name":"","scope":["6b0215c3.397c2c"],"uncaught":false,"x":290,"y":660,"wires":[["96ee1dc5.09895"]]},{"id":"e8b41b1b.7c5f08","type":"debug","z":"46dd0227.50b9cc","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"true","targetType":"full","statusVal":"payload","statusType":"auto","x":590,"y":600,"wires":[]},{"id":"e43e2b4a.6b6fd8","type":"trigger","z":"46dd0227.50b9cc","name":"","op1":"","op2":"{}","op1type":"nul","op2type":"json","duration":"1","extend":false,"overrideDelay":false,"units":"s","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":840,"y":660,"wires":[["2d395433.b4833c"]]},{"id":"96ee1dc5.09895","type":"switch","z":"46dd0227.50b9cc","name":"","property":"error.message","propertyType":"msg","rules":[{"t":"eq","v":"Error: Client Not Ready To Read At State init","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":420,"y":660,"wires":[["e8b41b1b.7c5f08","de0c601d.40743"]]},{"id":"de0c601d.40743","type":"delay","z":"46dd0227.50b9cc","name":"","pauseType":"rate","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"30","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":true,"x":620,"y":660,"wires":[["e43e2b4a.6b6fd8"]]},{"id":"376f57fc.d2bf58","type":"modbus-client","name":"2_Bronkhorst","clienttype":"tcp","bufferCommands":false,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"192.168.20.2","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"200","clientTimeout":"3000","reconnectOnTimeout":true,"reconnectTimeout":"6000","parallelUnitIdsAllowed":false}]`

AndyMuc72, 2021-08-03:
thank you @emmanuel-landrivon 

So far I restarted every morning Node-Red to fix the problem. I am very happy to test your solution.
Thank you again

emmanuel-landrivon, 2021-08-04:
Ok let me know, if it fix the prob.
I'm trying know performing benchmarking / stress my modbus tcp device (high requests, deconnections...).
Reflexion :
Node-red in production (industrial env.) seems ok for me, but i think is better to be in headless (command line/no display) if you have critical process with end-devices.
Regards

emmanuel-landrivon, 2021-08-05:
Hi,
I have about 0.02 % of error, with high sampling (10ms).
Fyi, my env : node red desktop with a TCP flowmeter as end-device
After 48h, it's still working well.
Regards.

biancode, 2021-08-08:
>  SMA Modbus - 

@rucksman The signal run time of SMA has sometimes a maximum of 200 ms.
Maybe you can just request every 250ms on that or better each second. Try to reduce your communication to the Modbus device. Some devices will close stress polling connections on the server side. ;)

en: https://files.sma.de/downloads/FSC11_Modbus-TB-en-10.pdf
de: https://files.sma.de/downloads/FSC11_Modbus-TB-de-10.pdf

emmanuel-landrivon, 2021-08-16:
Hi "Biancode",
Just say... thanks a lot for your advices and the modbus contrib dev  !
Regards.

rucksman, 2021-08-21:
> 1. detect the error (init state)

Unfortunately I do not have too much spare time at the moment, so I was not able to test earlier. My problem is that the Flex-Getter-Node says "Initialize" (and stays in this state), and error.message from the catch node is "[object Object]". So the error message is probably not suited to act as a trigger for your flow?

emmanuel-landrivon, 2021-08-21:
Hi,
If you cannot succeed to catch the error, you have another solution : Watchdog / Timeout.
After a timeout, for instance 2 minutes  without value --> trigger the modbus server to restart.

Regards

rucksman, 2021-08-24:
Brilliant! Sometimes it is so easy that one does not see the solution right in front of his eyes. Using a trigger node is probably the simpler solution as I can feed the parameters from the trigger node directly into the connector node. Thanks a lot for your help!