BiancoRoyal/node-red-contrib-modbus #213: Support for RTU “server”

rusirius84, 2020-07-14:
**I'm submitting a ...**

## Feature
 - ( x ) feature request
 
 **What is the motivation / use case for changing the behavior?**
  have been working with these nodes for some time now and everything works as expected. Currently bringing in data from multiple sources and assigning it to the desired registers. Able to retrieve those values from other third party devices as expected. Utilizing the modbus flex server via node red as it lends itself well to being able to gather data from multiple data sources via multiple protocols and consolidate them into a single modbus tcp server. That being said, while most people are not generally going this direction, there are some times where a local serial connection from a field device to node red would be useful. Looking to see if it would be possible to add a configuration to the modbus flex server node to accommodate rs485 requests. 

Node red SOAP / SNMP / MODBUS TCP -> modbus flex server -> modbus rtu master. Modbus RTU master being a third party device querying the flex server. 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( x ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( x ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( x ) Linux: Ubuntu 18.6
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X

* **Node:** (min. 10.x)
   `node --version` = vX.X.X

biancode, 2020-07-14:
Hi, like #123 ?

rusirius84, 2020-07-14:
[image]

[image]

Apologies I created this thread from my phone which made it difficult to add much clarity around my question. 

Attached are two pictures - 
full flow:
* records read from a .csv
* pulling out just the part required for the payload of the SOAP call
* results returned and scaled (for decimal place within modbus)
* joined back together with the corresponding description of the value being returned
* formatted to be sent to influxdb
* formatted for MODBUS Flex Server

Modus Flex Server Config:

My question is more around this node specifically.  Right now I am able to point modbus TCP requests at my instance of Node Red / device (unit-id) / register(s) I am storing the data in and get the correct response back.  No problem all good there.  

Now what I am curious about is if there is a way either within this same node or some combo of other nodes to have it also listen for requests on a serial port (usb to rs485). 

What I am after here is to be able to utilize Node-Red to collect data from multiple sources feed that to an instance of influx for my own purposes as well as expose that data to a third party MODBUS Master via rs485.  

In this case node red would need to act as a modbus slave which as mentioned it already does for TCP no problem.  What I cant seem to locate or wrap my head around is how to retrieve that same data locally over serial.  

Modbus Master initiate request for register 30006 over RS485 connected directly to Node-Red.  Node-Red respond with corresponding register value.

Thank you,

rusirius84, 2020-07-14:
quick aside for not only this project but others I see you are involved in... I have access to plenty of MODBUS (IP or Serial) / SOAP / BACNET (IP or MSTP) / SNMP / etc. devices if further testing or evaluation / troubleshooting is ever needed.  Have been looking for a way to contribute to these projects as I see great value in the flexibility node-red offers.

closed 2022-09-29 by github-actions[bot]