BiancoRoyal/node-red-contrib-modbus #215: Getting incorect values ModBus RTU

alexszilagyi, 2020-07-15:
I am adding this issue since I've tried to figure out what's the issue with my node-red setup & node-red-contrib-modbus. I have studies nearly all the documentation that I could find on the web related to this repo or modbus, still unclear how to overcome it.

## Problem
I have an industrial device which has a RS485 communication option. The problem I am facing is that I get different results when reading values from the modbus.

The reference software which I'm using (based on device documentation is: [ModScan64, Win10](http://win-tech.com/), if I'm reading the Read input register value (Function 04, Address 07) using ModScan64 I get the correct value, as the screen-shot below:

[image]

but when using the [node-red-contrib-modbus](https://github.com/BiancoRoyal/node-red-contrib-modbus/) package for node-red I'm getting the following values:

[image]

 **Current behavior**
Now I'm geting 32 insted of 20 as value.

 **Expected behavior**
**The value returned value should be 20**

 **Minimal reproduction of the problem with instructions**
1. Connect to a ModBus device over 485 via USB (`/dev/ttyUSB0`)
2. Specify serial type: RTU
3. Baund rate: 19200
4. Data bits: 8
5. Stop bits: 2
6. Parity: None
7. Try to read a value from the device (e.g. `04: Input register`, `address 07` - in my case)
   
### Your Environment

I am using a:
1. Raspberry Pi Zero W
2. USB 2.0 to TTL RS485 Serial Converter Adapter FTDI FT232RL FT232 SN75176 Double Function Protection Diy Kit Electronic PCB Board
3. Installed the following npm packages:

```
{
    "name": "node-red-project",
    "version": "0.0.1",
    "lockfileVersion": 1,
    "requires": true,
    "dependencies": {
        "@serialport/binding-abstract": {
            "version": "8.0.6",
            "resolved": "https://registry.npmjs.org/@serialport/binding-abstract/-/binding-abstract-8.0.6.tgz",
            "integrity": "sha512-1swwUVoRyQ9ubxrkJ8JPppykohUpTAP4jkGr36e9NjbVocSPfqeX6tFZFwl/IdUlwJwxGdbKDqq7FvXniCQUMw==",
            "requires": {
                "debug": "4.1.1"
            }
        },
        "@serialport/binding-mock": {
            "version": "8.0.6",
            "resolved": "https://registry.npmjs.org/@serialport/binding-mock/-/binding-mock-8.0.6.tgz",
            "integrity": "sha512-BIbY5/PsDDo0QWDNCCxDgpowAdks+aZR8BOsEtK2GoASTTcJCy1fBwPIfH870o7rnbH901wY3C+yuTfdOvSO9A==",
            "requires": {
                "@serialport/binding-abstract": "8.0.6",
                "debug": "4.1.1"
            }
        },
        "@serialport/bindings": {
            "version": "8.0.8",
            "resolved": "https://registry.npmjs.org/@serialport/bindings/-/bindings-8.0.8.tgz",
            "integrity": "sha512-xMJHr7CyOPq+wwC/S2RNI+tY+WZW4gXY3tE8QUOIRp0K7lSyLYOzKdyGUtk2uI0ohDMV3OcB+TEhhffT2S2DHQ==",
            "requires": {
                "@serialport/binding-abstract": "8.0.6",
                "@serialport/parser-readline": "8.0.6",
                "bindings": "1.5.0",
                "debug": "4.1.1",
                "nan": "2.14.1",
                "prebuild-install": "5.3.5"
[... 725 lines cut ...]
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Custom device
     - Version: N/A
     - Vendor : N/A
     - Link   : N/A
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v10.21.0

Why on Windows PC works with no issues using the [ModScan64](http://win-tech.com/) and the same device (USB 2.0 to TTL RS485 Serial Converter Adapter FTDI FT232RL FT232 SN75176 Double Function Protection Diy Kit Electronic PCB Board) ?

Ekristoffe, 2020-07-16:
hello,
In modscan, you are using hexadecimal based value or decimal based one ?
In Node-RED, the data are displayed in decimal base and the buffer is displayed in hexadecimal base.
In modscan you can change in setup > display option.

alexszilagyi, 2020-07-16:
Hi @Ekristoffe , 

Within the ModScan64 I have selected Integer, here's a capture:

[image].

I still not get it why the value is different for the same address, nearly in the same time.

Ekristoffe, 2020-07-16:
Indeed ...

alexszilagyi, 2020-07-16:
I did contacted the vendor of the device and support team of ModScan64 just to clarify the underlying aspects of the issue.

Happy to contribute after fixing this (even a PR).

alexszilagyi, 2020-08-24:
Hi @egdev !,

Thanks! Yeah, that was it and I figure it out by using another tool (desktop based, an open source one). Good to know for the future.

closed 2020-08-24 by alexszilagyi