BiancoRoyal/node-red-contrib-modbus #218: Modbus Flex Getter indicates a "active/connected" state but first modbus message lost.

sensbo, 2020-08-26:
I'm submitting a **problem.** 

## Problem

 **Current behavior**
The Modbus Flex Getter node indicates a "active/connected" state via node status. I trigger this state to ensure that everything is fine. If I start sending a message the state changes to "queueing", ("failure"), "reconnecting", "initialize" and then again changes to "connected". The message which I tried to send is gone and will be lost. If I now start again to send a message shortly afterwards (20sec or less), the state changes from "connected" to "queueing", "reading done" and  "active" and the message has been send.
I am loosing every time the first message. In the logs there is sometimes a  "Modbus Failure On State sending Get More About It By Logging" or/and "PortNotOpenError: Port Not Open" error during the first message send. 
After send the next message the connection establish, the connection state changes from "connected" to "active" and I can send/read messages via modbus. The question is: How can I avoid this lost of the first message?

My current workaround is a permanent read request to the modbus server every 10 sec. The getter node will be all the time in an "active" status. But this provides traffic and also do not avoid this behavior, if the connection breaks.

 **Expected behavior**
The user should be able to check if the modbus connection is really initiliazed and ready for sending data. The "active/connected" state is not valid all the time.

 **Minimal reproduction of the problem with instructions**
 
   1. Request data from a modbus server
   2. Send it to Modbus Flex Getter
   
### Your Environment
  * (x) I am using just the Node-RED package with node-red-contrib-modbus
  
  * Device :
       - (x) Linux, nodered docker container
   
   * Description of the connecting Modbus system:
      - Name   : Modbus Server as a part of the SPPA-T3000 Control System
     - Version:  7.4 SP1
     - Vendor : Siemens Energy
     - Link   : https://new.siemens.com/global/en/products/automation/distributed-control-system/sppa-t3000.html
    
* **node-red-contrib-modbus version:** v5.13.3
* **Node:**  v12.18.3

biancode, 2020-08-26:
Hi, please attach a log file - based on our Wiki about Debug logging - including the modbus-serial logs!
With that we should see the Modbus failures, too. ;)

sensbo, 2020-08-27:
@biancode Thank you for your answer.
Unfortunately, I have no access to a console. The environment is locked, node-red running in a docker container, I have no access to it. Are there any options to enable the debug information directly in node-red ui as starting point?

lehades, 2020-12-13:
I can report I've found this same issue in the modbus-flex-write node (v5.13.3). Unfortunately i'm in the same boat (docker container) and can't debug either.

All i can get from the docker logs is:

> 13 Dec 21:34:56 - [warn] [modbus-write:AC Command] Modbus Failure On State sending Get More About It By Logging
> 13 Dec 21:34:56 - [warn] [modbus-write:AC Command] Modbus Failure On State sending Get More About It By Logging
> 13 Dec 21:34:56 - [warn] [modbus-flex-write:2b67ecb4.eed904] Modbus Failure On State sending Get More About It By Logging
> 13 Dec 21:34:56 - [error] [modbus-flex-write:2b67ecb4.eed904] Port Not Open

Seems to me like the remote device closes the socket but the local node still thinks it's open. After the first attempt the socket reopens and any subsequent attempts work as intended. I can confirm too that having a Modbus-read node polling the server is a workarround.

Maybe the flex nodes need some kind of 'heartbeat' to keep the socket open.

biancode, 2020-12-16:
Hi you can set Docker env variables to activate the DEBUG global or you could set envs from the scripts to get more debug logs see the Wiki here about your options to set DEBUG. NODE_DEBUG should work as well via env settings to Docker ;)

biancode, 2020-12-18:
That is more in the context of the user not on the nodes from my side. active means it should be ready, from what the nodes can check so far. If the is not enough, then it is out of scope here. You can use the status node of node-red to take care about it a bit more, but at the end there is no real proof in all cases. The message which you tried to send is gone and will be lost, but you get informed by errors and you can handle those errors on your side. There are to much possibilities to take care about all those cases. ;)

closed 2020-12-18 by biancode