BiancoRoyal/node-red-contrib-modbus #219: Allow receive data length bigger than expected

E-NINA, 2020-08-28:
Hello

I'm trying read data from my company energy meter with your plugin but I'm getting problem with a data length bigger than expected "Error: Data length error, expected 7 got 9"
I don't know if this is a mistake of their modbus implementation or if in fact this is possible. Just to explain.
All registers are consecutive with some of them single others double:

Index  Address  data
1         0x0001      FF FF               (single)
2         0x0002      FF FF FF FF      (double)
3         0x0003      FF FF FF FF
4         0x0004      FF FF
5 ...

When I read single registers like 1 or 4 I've no problem
I get "Error: Data length error, expected 7 got 9" when I try read "double registers"

Is there a way to add possibility to overcome this limitation?

Best regards
ENina

E-NINA, 2020-08-30:
Hello, to complete information above

Meter information JANZ 3801 (national company meter)

Example of single register:
Voltage -> Index 108 (HEX -> 6C )
resquest:  00 01 00 00 00 06 01 04 **_00 6c 00 01_**
response: 00 01 00 00 00 05 01 04 02 _**08 E5**_  ->  (hex 08 e5  -> 2277  -> 227,7V)

Example of double register:
rate 1 total energy -> Index 38 (HEX -> 26 )
resquest:  00 01 00 00 00 06 01 04 **_00 26 00 01_**
response: 00 01 00 00 00 07 01 04 04 **_00 01 92 B1_**  ->  (hex 00 01 92 B1  -> 103089  -> 103,089 kwh)

As you can see request is exactly the same but response output is different
It make sense add the possibility to add a switch to not compare data lengths, or a function to mention expected size of response?

Best regards
ENina

Ekristoffe, 2020-08-31:
Hello ENina,
I am not sure this possibility is in the standard.
I have dissected the modbus request and response and those are correct (not expected but still correct):

Example of single register:
Voltage -> Index 108 (HEX -> 6C )
request: 
  Transaction Identifier: 00 01 
  Protocol Identifier: 00 00 
  Length: 00 06 (only the * are counted)
 * Unit Identifier: 01 
 * Function Code: 04 
 * Starting Address: 00 6c 
 * Quantity of Registers: 00 01
response: 
  Transaction Identifier: 00 01 
  Protocol Identifier: 00 00 
  Length: 00 05 (only the * are counted)
 * Unit Identifier: 01 
 * Function Code: 04 
 * Byte count: 02 
 * Input Registers: 08 E5 -> (hex 08 e5 -> 2277 -> 227,7V)

Example of double register:
rate 1 total energy -> Index 38 (HEX -> 26 )
request: 
  Transaction Identifier: 00 01 
  Protocol Identifier: 00 00 
  Length: 00 06 (only the * are counted)
 * Unit Identifier: 01 
 * Function Code: 04 
 * Starting Address: 00 26  
 * Quantity of Registers: 00 01
response: 
  Transaction Identifier: 00 01 
  Protocol Identifier: 00 00 
  Length: 00 07 (only the * are counted)
 * Unit Identifier: 01 
 * Function Code: 04 
 * Byte count: 04 
 * Input Registers: 00 01 92 B1 -> (hex 00 01 92 B1 -> 103089 -> 103,089 kwh)

Whet if you ask in the request not 1 but 2 register for the address 0x26  ?
Will your device give an error ?

E-NINA, 2020-08-31:
Hello,

yes it give error.
Them error message is:

Error: Data length error, expected 9 got 13

Replay data is:

resquest: 00 01 00 00 00 06 01 04 00 26 00 **_02_**
response: 00 01 00 00 00 0B 01 04 08 **_00 01 A2 CA, 00, 00, 58, 58_** 

00 01 A2 CA are the data for register 0x26 
     ->    rate 1 total energy - 107,210 kwh

 00, 00, 58, 58 are the data for register 0x27 
     ->    rate 2 total energy - 22,616 kwh

E-NINA, 2020-08-31:
error messages are from Node Red

E-NINA, 2020-08-31:
To complete my statements before, i can communicate with my device trough windows program and no problem at all (data/examples I provide are taken from that), issue is with Node-red plugin node-red-contrib-modbus, that brake with the messages "Data length error, expected XX got YY" and no data are show/retained.
It's possible add a switch to not compare data lengths, or a function to mention expected size of response?
If not, can give me some hints how to delete/baypass this check on the code of plugin?

Ekristoffe, 2020-09-01:
Hello I have done some digging in the code. It seem this error is not from the code made by christian but from the base library :
https://github.com/yaacov/node-modbus-serial/
Maybe you can try to ask there.
Unfortunately we can disable this check in the node, this must be done in the library.

biancode, 2020-12-18:
We will check for new updates for serial here.

closed 2022-09-10 by github-actions[bot]