BiancoRoyal/node-red-contrib-modbus #220: Help - Writing Floats to 16bit integers

vigalent, 2020-09-03:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - (x) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * (x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v12.X.X

Help - I'm using the server as a Modbus slave and populating the registers via various systems (some fixed values) which is then queried by a Tridium Jace8000 (acting as Modbus Master). One of the values is a large number (1766229), so I have the flow below converting it to 16bit and writing to 2 registers.

When I read these registers in the Modbus master, the resulting float value is incorrect. The value drifts as it gets larger. For instance, at 10, it is correct, at 100, the Modbus master reads 100.4 and at 1776229 it reads 1767680.

I cant figure out if the problem is at the node-red end or at the Modbus master.

Thanks

`[{"id":"1dd9ce49.b1647a","type":"function","z":"80b83cac.1c4e7","name":"","func":"function float32ToUint16(value) {  \n  var buffer = new ArrayBuffer(4);\n  var intView = new Uint16Array(buffer);\n  var floatView = new Float32Array(buffer);\n\n  floatView[0] = value;\n \n\n  return [intView[0], intView[1]];\n}\n\n//var incoming = JSON.parse(msg.payload);\n\n//Compose ModBus message (Holding Registers) READ/WRITE\nvar modbusHR = {\n payload : {\n 'value' : [\n float32ToUint16(msg.payload)[0], \n float32ToUint16(msg.payload)[1],\n ],\n 'fc' : 16,\n 'unitid' : 1,\n 'address' : 4,\n 'quantity' : 2\n }\n};\n\nnode.send([modbusHR]);","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":180,"wires":[["4c60cf8d.36e49","447eebe8.03b0fc"]]},{"id":"447eebe8.03b0fc","type":"debug","z":"80b83cac.1c4e7","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":610,"y":240,"wires":[]},{"id":"4c60cf8d.36e49","type":"modbus-flex-write","z":"80b83cac.1c4e7","name":"","showStatusActivities":true,"showErrors":true,"server":"1dd4b889.c286f7","emptyMsgOnFail":false,"keepMsgProperties":false,"x":710,"y":180,"wires":[[],["ee02d018.f56d8"]]},{"id":"ee02d018.f56d8","type":"modbus-response","z":"80b83cac.1c4e7","name":"","registerShowMax":20,"x":970,"y":180,"wires":[]},{"id":"da2856fb.2af07","type":"inject","z":"80b83cac.1c4e7","name":"1766229","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"1766229","payloadType":"num","x":260,"y":180,"wires":[["1dd9ce49.b1647a"]]},{"id":"1dd4b889.c286f7","type":"modbus-client","z":"","name":"local","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectOnTimeout":false,"reconnectTimeout":2000,"parallelUnitIdsAllowed":false}]`

JorgeAlberto91MS, 2020-12-04:
// Use Uint16 Registers to Write your Variables.
var dim     = 10;  // Size of Registers you want to Write on Modbus Buffer
var Uint16  = new Uint16Array(dim);
var Uint8   = new Uint8Array(dim*2);

// Uint16 Assignation
Uint16[0] = Var1;
Uint16[1] = Var2;
Uint16[2] = Var3;

// Pass UInt16 Register to Uint8 Registers
for(var i=0; i<dim*2; i+=2){
    Uint8[i]    = Uint16[parseInt(i/2)]>>8; 
    Uint8[i+1]  = Uint16[parseInt(i/2)]; 
}
// Assign payload
msg.payload = { 
    'value': Uint8,
    'register': 'holding', 
    'address': 0
};
return msg;

biancode, 2020-12-16:
Hi, please give Buffers a try to fill them first and then create y new UINT16Array from that Buffer ;)
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Typed_arrays
https://nodejs.org/en/knowledge/advanced/buffers/how-to-use-buffers/

closed 2020-12-18 by biancode