BiancoRoyal/node-red-contrib-modbus #222: Uninstallation issue from Palette Manager

royaldevbrat, 2020-10-06:
**Problem**
 - I am trying uninstall this node from Node-RED palette manager

**Current behavior**
 Its throwing error during uninstallation 
 
[image]

**Expected behavior**
It should get uninstall properly without error
 
**Minimal reproduction of the problem with instructions**
 
   1. Install node-red-contrib-modbus node from palette manager
   2. Drag and Connect to any server from TCP/Serial and Deploy the node-red
   3. Delete the node and configuration node and Deploy the node-red
   4. Try to uninstall the node from palette manager

   
### Your Environment
  
  * Device :

    - ( ) Windows: v 10
   
   * Description of the connecting Modbus system:
   
     - Name   : Modbus Simulator

* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v10.15.0

biancode, 2020-12-16:
Hi, please take care on Win10, that username has no special character in it like a whitespace. That is sometimes not working from nodejs install and uninstall. Did you get that issue while install and uninstall from your CMD as well? Please check that points! ;)

royaldevbrat, 2020-12-18:
Yes, the username does not has any special character. And I am getting this error from CMD as well if Node-RED is running.

biancode, 2020-12-18:
ERRNO 4048 is out of scope here and looks like a problem with your NPM installation. You could try:
1.)
`npm cache verify`
2.) follow the instructions to re-install https://www.npmjs.com/package/npm-reinstall

closed 2020-12-18 by biancode