BiancoRoyal/node-red-contrib-modbus #223: Timeout results in: "TypeError: u.pushToQueueByUnitId(...).then(...).catch(...).finally is not a function "

windkh, 2020-10-27:
**I'm submitting a bug**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 Running the flow in a docker container on a QNAP NAS results in a TypeError:

`
7 Oct 19:06:11 - [info] Flows gestartet                                                                                                                     
27 Oct 19:06:11 - [warn] [modbus-client:Fronius] Client -> fsm connected after state init Get More About It By Logging TCP@fronius.fritz.box:502 default Unit
-Id: 1                                                                                                                                                       
27 Oct 19:06:11 - [warn] [modbus-read:MPPT I160] Client -> initializeReadingTimer delay timer node 37b24.875344dc8undefined                                  
27 Oct 19:06:21 - [warn] [modbus-read:MPPT I160] Client -> startIntervalReading node 37b24.875344dc8undefined                                                
27 Oct 19:06:33 - [red] Uncaught Exception:                                                                                                                  
27 Oct 19:06:33 - TypeError: u.pushToQueueByUnitId(...).then(...).catch(...).finally is not a function                                                       
    at Object.<anonymous> (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:436:7)                                                    
    at emitThree (events.js:136:13)                                                                                                                          
    at Object.showStatusActivities [as emit] (events.js:217:7)                                                                                               
    at Timeout.a.modbusPollingRead [as _onTimeout] (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-read.js:165:16)                            
    at ontimeout (timers.js:498:11)                                                                                                                          
    at tryOnTimeout (timers.js:323:5)                                                                                                                        
    at Timer.listOnTimeout (timers.js:290:5)                                                                                                                 
npm ERR! code ELIFECYCLE      
`

 **Expected behavior**
 
I do not expect an Uncaught Exception on timeout. 

 **Minimal reproduction of the problem with instructions**
 
   1.  Node-Red 0.20.8 inside docker on a QNAP NAS
   2. add flow from here https://flows.nodered.org/flow/a2c26a868e97339e869630e003d3dde1
   3. Start flow
   Problem does not occur when I run the flow on my local node red instance (Windows 10)

### Your Environment
**Please tell us about your environment:**

    Fronius Modbus RTU (PV Inverter)

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (X) Linux: v Linux 4.14.24-qnap x64 LE 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v8.16.1 

If the node version is the problem, then just close this issue. Thanks in advance.

windkh, 2020-10-28:
Upgrading to node-red docker image v1.1.3-12-AMD64 from https://github.com//node-red/node-red-docker helped solving the problem. I guess it was the node version.

closed 2020-10-28 by windkh

biancode, 2020-10-28:
It is NodeJS 10 instead of 8 if I understood your information right, which is required see the minimum requirements of the package ;)