BiancoRoyal/node-red-contrib-modbus #224: Bug after restart of Modbus-flex-server with active modbus-tcp connection

tadzil, 2020-11-01:
There is a problem with at least Modbus-flex-server, when after restarting flows, server values in node-red flow are reset and flow is working with new memory, but earlier connected master, which had active connection when flow was restarted, still haves old register values, can change and read changed registers and these values can't be seen anywhere in node-red flow.
This is serious problem, because neither server side nor modbus master side are not having any error, but are working with completely different values.

## Problem
 - (x) bug report 
 
 **Minimal reproduction of the problem with instructions**
 
   1. Import Modbus-Server example flow.
   2. Connect ModbusTCP master to this server and change some values (ModbusPoll as master, for example). Leave master opened and reading registers.
   3. Restart flows from node-red
   
### Your Environment
**Please tell us about your environment:**

  
  * Device :
    - (x) Raspbian: latest for 2020.10.31 (fresh install)
   * Description of the connecting Modbus system:
   
     - Name   :ModbusPoll, windows 10
    
* **node-red-contrib-modbus version:** latest for 2020.10.31 (fresh install)

* **Node:** (min. 10.x)
   `node --version` = latest for 2020.10.31 (fresh install)

SietzeKoonstra, 2020-11-23:
Hi,
I got the same error and debugged it. The problem is as follow: after restarting the flows, the TCP socket is not destroyed. So, when you read the modbus slave, you get a response of the old socket. 
### Solution
The used modbus-serial module in modbus-flex-server.js has a close function. This close function will destroy all the open sockets.
When the code in the `node.on('close', function..` is replaced by:
`node.on('close', function () {
      node.modbusServer.close();
      mbBasics.setNodeStatusTo('closed', node)
      node.modbusServer = null
    })` the problem is solved.

I know the solution for this problem, but I'm not sure how to do a pull request..

Zeddy-, 2021-01-04:
This problem also happends with the non flex server, but only if you use Flex write on it after it started it seems.

mixarrpg, 2021-04-04:
same problem here...and opc client dosent update the registers values when node red update the project (and the values changed in the runtime)

biancode, 2021-08-09:
also #255 
should be done from PR in v5.14.1