BiancoRoyal/node-red-contrib-modbus #225: Node-Red Uncaught Exception due Modbus Server node.

JorgeAlberto91MS, 2020-12-04:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 Uncaught Exception on Node-RED.
 After a lost connection from Modbus master, the Modbus Server node shows an error that makes Node-Red to forced stop.

 **Expected behavior**
 
 After a lost connection from Modbus master, the Modbus Server must show the connection error but Node-Red should remain active.

 **Minimal reproduction of the problem with instructions**
 
   1. Load data into Modbus Server Buffer.
   2. Start a Modbus Request from another device.
   3. Start / Stop the connection from remote device.
   
### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name     : Modbus Simulator
     - Version  : 3.53
     - Vendor   : Calta Computer Systems Limited
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v12.19.1

GlennMatthys, 2022-01-19:
I've encountered this error myself, this is the output from the logs:

```
18 Jan 13:25:23 - [red] Uncaught Exception:
18 Jan 13:25:23 - [error] Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
```

rudam, 2022-05-16:
Hi.
I see a similar error:

RPi 3B
Debian 10
node-red-contrib-modbus v5.21.2

```
16 May 18:53:39 - [info] Node-RED version: v2.2.2
16 May 18:53:39 - [info] Node.js  version: v12.22.12
16 May 18:53:39 - [info] Linux 5.10.103-v7+ arm LE
16 May 18:53:43 - [info] Loading palette nodes
16 May 18:53:49 - [info] Dashboard version 3.1.7 started at /ui
16 May 18:53:49 - [info] Settings file  : /home/unipi/.node-red/settings.js
16 May 18:53:49 - [info] Context store  : 'default' [module=memory]
16 May 18:53:49 - [info] User directory : /home/unipi/.node-red
16 May 18:53:49 - [warn] Projects disabled : git command not found
16 May 18:53:49 - [info] Flows file     : /home/unipi/.node-red/default_flow.json
16 May 18:53:49 - [info] Server now running at http://127.0.0.1:1880/
16 May 18:53:49 - [info] Starting flows
16 May 18:53:49 - [warn] [modbus-read:Read Coils] Client -> open node ba6e396b6556405eundefined
16 May 18:53:49 - [warn] [modbus-client:Modbus Server] Client -> fsm init state after new
16 May 18:53:49 - [warn] [modbus-client:Modbus Server] Client -> first fsm init in 500 ms TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:49 - [warn] [modbus-read:Read Registers] Client -> open node f57fa64bb5e43561undefined
16 May 18:53:50 - [info] Started flows
16 May 18:53:50 - [warn] [modbus-client:Modbus Server] Client -> fsm connected after state init Get More About It By Logging TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:50 - [warn] [modbus-read:Read Coils] Client -> initializeReadingTimer delay timer node ba6e396b6556405eundefined
16 May 18:53:50 - [warn] [modbus-read:Read Registers] Client -> initializeReadingTimer delay timer node f57fa64bb5e43561undefined
16 May 18:53:51 - [warn] [modbus-flex-server:12f9c7873761a4da] Error: This socket has been ended by the other party
    at Socket.writeAfterFIN [as write] (net.js:456:14)
    at sockWriter (/home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp.js:284:30)
    at cb (/home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp.js:113:16)
    at /home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp_handler.js:228:17
    at _handlePromiseOrValue (/home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp_handler.js:55:9)
    at Object._handleReadMultipleRegisters [as readMultipleRegisters] (/home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp_handler.js:300:21)
    at Socket._parseModbusBuffer (/home/unipi/.node-red/node_modules/modbus-serial/servers/servertcp.js:160:22)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
16 May 18:53:51 - [warn] [modbus-client:Modbus Server] Client -> fsm failed state after sending Get More About It By Logging TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:51 - [warn] [modbus-read:Read Coils] Client -> resetDelayTimerToRead node ba6e396b6556405eundefined
16 May 18:53:51 - [warn] [modbus-read:Read Coils] Modbus Failure On State sending Get More About It By Logging
16 May 18:53:51 - [warn] [modbus-read:Read Registers] Client -> resetDelayTimerToRead node f57fa64bb5e43561undefined
16 May 18:53:51 - [warn] [modbus-client:Modbus Server] Client -> fsm broken state after failed Get More About It By Logging TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:51 - [warn] [modbus-client:Modbus Server] Client -> fsm reconnect state after broken Get More About It By Logging TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:53 - [warn] [modbus-client:Modbus Server] Client -> fsm init state after reconnecting TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:53 - [warn] [modbus-client:Modbus Server] Client -> fsm init in 2000 ms TCP@127.0.0.1:18502 default Unit-Id: 1
^C16 May 18:53:55 - [info] Stopping flows
16 May 18:53:55 - [warn] [modbus-read:Read Coils] Client -> close node ba6e396b6556405eundefined
16 May 18:53:55 - [warn] [modbus-client:Modbus Server] Client -> stopped state without reconnecting TCP@127.0.0.1:18502 default Unit-Id: 1
16 May 18:53:55 - [warn] [modbus-read:Read Registers] Client -> close node f57fa64bb5e43561undefined
16 May 18:53:55 - [info] Stopped flows
```

R-LopezV, 2022-07-27:
Good day.
It's been 2 years and the same bug is present.

Raspberry Pi Zero W (Raspbian Lite 32-bit 2022-04-04)
Node-RED 3.0.1 using NodeJS 14.20.0
node-red-contrib-modbus v5.21.2

`27 Jul 22:32:35 - [info] Server now running at http://127.0.0.1:1880/
27 Jul 22:32:35 - [warn] Encrypted credentials not found
27 Jul 22:32:35 - [info] Starting flows
27 Jul 22:32:35 - [info] Started flows
27 Jul 22:36:11 - [info] Installing module: node-red-contrib-modbus, version: 5.21.2
27 Jul 22:39:35 - [info] Installed module: node-red-contrib-modbus
27 Jul 22:39:39 - [info] Added node types:
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-client
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-response
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-read
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-getter
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-flex-getter
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-write
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-flex-write
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-server
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-flex-server
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-queue-info
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-flex-connector
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-io-config
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-response-filter
27 Jul 22:39:39 - [info]  - node-red-contrib-modbus:modbus-flex-sequencer
27 Jul 22:44:25 - [info] Installing module: node-red-contrib-modbus-tcp-ip, version: 1.1.5
27 Jul 22:45:02 - [info] Installed module: node-red-contrib-modbus-tcp-ip
27 Jul 22:45:02 - [info] Added node types:
27 Jul 22:45:02 - [info]  - node-red-contrib-modbus-tcp-ip:modbus-tcp-ip
27 Jul 22:45:02 - [info] Installing module: node-red-contrib-modbustcp, version: 1.2.3
27 Jul 22:45:45 - [info] Installed module: node-red-contrib-modbustcp
27 Jul 22:45:46 - [info] Added node types:
27 Jul 22:45:46 - [info]  - node-red-contrib-modbustcp:modbustcp-server
27 Jul 22:45:46 - [info]  - node-red-contrib-modbustcp:modbustcp-write
27 Jul 22:45:46 - [info]  - node-red-contrib-modbustcp:modbustcp-read
27 Jul 22:47:28 - [info] Stopping flows
27 Jul 22:47:28 - [info] Stopped flows
27 Jul 22:47:28 - [info] Updated flows
27 Jul 22:47:28 - [info] Starting flows
27 Jul 22:47:28 - [info] Started flows
27 Jul 22:47:44 - [info] Stopping flows
27 Jul 22:47:44 - [info] Stopped flows
27 Jul 22:47:44 - [info] Updated flows
27 Jul 22:47:45 - [info] Starting flows
27 Jul 22:47:45 - [info] Started flows
27 Jul 22:47:45 - [red] Uncaught Exception:
27 Jul 22:47:45 - [error] Error: listen EADDRNOTAVAIL: address not available 192.168.100.12:10502
    at Server.setupListenHandle [as _listen2] (net.js:1314:21)
    at listenInCluster (net.js:1379:12)
    at doListen (net.js:1516:7)
    at processTicksAndRejections (internal/process/task_queues.js:83:21)
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Failed with result 'exit-code'.
nodered.service: Consumed 4min 58.991s CPU time.
nodered.service: Scheduled restart job, restart counter is at 1.
Stopped Node-RED graphical event wiring tool.
nodered.service: Consumed 4min 58.991s CPU time.
Started Node-RED graphical event wiring tool.
`
Everything from the Raspbian installation to this palette has been freshly installed.

Is there any chance we'll get a bug fix soon? A project I'm working on needs this library, and the whole program crashes due to this bug. Thank you in advance!

RedShift1, 2022-07-28:
As a workaround you can use an external Modbus/TCP server, and writing the necessary values using Node-Red to it

biancode, 2022-07-28:
Yes - two years - the last two years were and are not easy with all the issues from our real life. 
@R-LopezV it is Open-Source and the community could contribute or users could sponsor for fixing issues faster ;)  

It will be fixed with the next v5.22+ - but keep in mind, then the flow should handle the error state of the node

R-LopezV, 2022-07-28:
> Redshift1:
As a workaround you can use an external Modbus/TCP server, and writing the necessary values using Node-Red to it 

Yeah, we're working on that right now, thanks for the idea.

@biancode I hope all of you have had a better year so far. I appreciate all of you working on this in spite of your personal situations.

> biancode:
it is Open-Source and the community could contribute or users could sponsor for fixing issues faster ;)
It will be fixed with the next v5.22+ - but keep in mind, then the flow should handle the error state of the node

To be honest I'm not super savvy on JS, but I could give it a try... As for the flow handling the error, it's more than enough personally. That would make fixing the mistake on the node's config much faster.

closed 2022-10-28 by github-actions[bot]