BiancoRoyal/node-red-contrib-modbus #226: Modbus TCP stops communicating after a few days

nygma2004, 2020-12-08:
**Current behavior**
I have a flow reading data from a Modbus TCP client. I am using 4 Modbus-Read nodes two with FC1 and two with FC3 reading different ranges. I also added Modbus queue info watching these 4 nodes.
All are configured with Unit ID 1, all set to 5 second update.
 
 **Expected behavior**
All works fine, but after a few days or a week no update is read any more. The Modbus nodes show status "active (5sec)" and nothing happens. And in the queue info I see thousands of messages. I added the catch node and used the example to reset the queue on very high, but that does not seem to make a difference either. Btw, in the Modbus Queue Info node the documentation says msg.resetQueue= True, but the example code below uses msg.payload.resetQueue= True;
So this is misleading and I have no idea how it is supposed to work.
I need to redeploy the flow to make it work.

I am sorry, but I did not find any useful information in the Wiki for this scenario.
 
 **Minimal reproduction of the problem with instructions**
I can give you a demo flow, but I would not post the code here with working Modbus addresses included.
   
### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - () Linux: v 
    - ( ) Windows: v 
    - (X) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Saia PLC
     - Version:
     - Vendor : Saia-Burgess
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v10.22.0

biancode, 2020-12-16:
Hi, send some log file(s) to my E-Mail with Debug enabled for *modbus* to see all states until the stop. I'll give it a short check if there was a problem.

nygma2004, 2020-12-16:
Hi Klaus,
I have never done this before. Can I just clarify how it is done? Shall I enable "Queue Logging" and "Log state changes" on the modbus-client configuration? Does it get saved along with the standard debug file in var/log? Is there a way I can separate the log of this component to a separate file?
FYI, the modbus client is on the "open internet", so it could be due to connection problems, but it always seem to recover when I re-deploy the flow.

biancode, 2020-12-16:
Hi, that is out of scope here. Please, use Slack to ask the community for more detailed help. If you have access you could try to find your way to get more logs see https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG

nygma2004, 2020-12-16:
I managed to get the log and emailed it to you.

nygma2004, 2021-01-12:
I updated my flow, and changed all my Modbus read nodes, to Modbus Getter. There are 4 getters on a Modbus queue, and I and sending the requests in every 5 seconds. I am not sure when I made the change, probably about 2 weeks ago. I just checked now and it was running fine until yesterday 9pm. So for some reason it appears to be more stable.
Now there are 40K entries in the queue stuck, and again it is not responding to resetQueue messages.

biancode, 2021-11-13:
reopen if it still existing with v5.15+

closed 2021-11-13 by biancode