BiancoRoyal/node-red-contrib-modbus #233: Limited amount of registers in read node?

rucksman, 2021-01-20:
AS #211 is still not solved for me, I tried a different approach: I want to read all the necessary registers at once and split the buffer afterwords in Node-RED to select the few values I need.

The necessary amount is 334 (registers starting at 30535 up to 30869). But when I try this with a read node, there is no result. Only after entering 96 instead of 334 as an amount there is a result.

Is there really a limit for the amount, or is this problem related to my solar inverter?

Another question, if this does not work properly: can I use a flex getter node to get more than one register back? At the moment I feed the flex getter node with just one register like so:
```
msg.payload = { 
'fc': 3,
'unitid': 3, 
'address': 30867 , 
'quantity': 2 }; 
return msg;
```

Is there a syntax to feed in more objects then this single register?

biancode, 2021-02-10:
Hi @rucksman - it depends on the vendor and Modbus Specs. Some can just provide 50 Registers per read - 96 is a good support. We do those reads as you can find in the examples in steps á 50 Registers (most supported) and collect the data to the context inside Node-RED - from there you will have no limits anymore. ;)

pseudo-code is:

```
for x-times do:
msg.payload = { 
'fc': 3,
'unitid': 3, 
'address': 30867, 
'quantity': 50 }; 
node.send(msg);
```

for the results use:

```
msg.payload transform to data structure
flow.set('registers_x_y', data);
or set up some more complex information structure to the context
```

use:

```
new msg

modbusData = flow.get('registers_x_y')

msg.payload = modbusData[3]
or
msg.payload = { temperature: modbusData[3] }

return msg;
```

rucksman, 2021-02-13:
Hi,

thanks for the explanation.

Just one more question: any idea about #211? I still run into this error every few days which makes it hard to use it in other flows like load my eletric car when my solar systems produces enough power...

closed 2021-08-09 by biancode