BiancoRoyal/node-red-contrib-modbus #235: Act as a Slave in a Serial RS485 System

Ion3, 2021-02-08:
**I'm submitting a feature request**

 **What is the motivation?**

I would like to use Node-RED (via Modbus-Nodes) as a slave, to give an external master the possibility to pull data. The Node should be able to emulate a slave device.

Practical example: A heat pump is pulling the energy surplus of an energy meter connected to a photovoltaic system to manage the energy consumption. Node-RED could by emulating the energy meter influence the energy consumption. This would be a useful option in a complex setup.

 **Current behavior**
This seems to be possible in a client/server setup of a TCP/IP system.
 
 **Expected behavior**
 It should also be possible in a master/slave setup of a serial RS485 system by using a simple USB-RS485-Dongle.
 
### Your Environment
**Please tell us about your environment:**
Simple System using an Raspberry Pi with USB-RS485-Dongle.
German issue description: https://www.viessmann-community.com/t5/Waermepumpe-Hybridsysteme/Energiezaehler-fuer-Eigenstromnutzung-der-Vitocal-200-A-einbinden/m-p/147571
Second German source: https://openwb.de/forum/viewtopic.php?p=19611#p19611

 **Device:**

- (X) Raspbian: v 

**Description of the connecting Modbus system:**
   
- Name   : SBC ALE3 energy meter
- Version: ALE3D5F
- Vendor : sbc
- Link   : https://sbc-support.com/uploads/tx_srcproducts/26-527_ENG_DS_EnergyMeter-ALE3-with-Modbus_01.pdf

and

- Name   : Viessmann Vitocal 200-A
- Version: AWO-E-AC 201.A13
- Vendor : Viessmann 
- Link   : https://www.viessmann.co.uk/products/heat-pumps/air-source-heat-pumps/vitocal-200-a

    
**node-red-contrib-modbus version:** v5.13.3

industrialinternet, 2021-02-08:
Ion3 I support this we have used the Viessmann with the KNX option.
So you could Node-RED KNX ultimate.

The other option is to create your own in a function node we've done this once or twice using node lib modcr which you have to add to  functionGlobalContext:  and can then ref in function node.

Ion3, 2021-02-10:
Hi industrialinternet, thank you for your input. KNX is no option, as I have no KNX devices and KNX does not support the transfer of the surplus energy available.

My idea was not to develop a workaround but to submit a feature request which would be of general benefit. 

I didn’t really get your second option. I have read and write nodes which both actively acts in the serial RTU network. I am missing a node which is listening an reacts on inputs from the serial network.

biancode, 2021-02-10:
@Ion3 I'm sorry, but we have no time to do that at the moment. It is on the roadmap 2021.

Ion3, 2021-02-10:
Wow Klaus, thank you so much! This satisfies me more then enough! I’m crossing fingers it will be soon in 2021. Greetings, Timo

BTW: Where is the best place to discuss troubleshootings? I have still some reading issues which works fine with the CAS Modbus Scanner but not on my raspberry setup. I’m willing to learn. As your shared roadmap, this will be my preferred tool. It will give a huge benefit for me. Thank you so much.

industrialinternet, 2021-02-10:
Hi that's what we do.
We have a LoRaWAN (wireless iot) gateway it runs a very old version off
node-red.
It has a ubs port on it so we use rs454-conv-usb-serial [[ node-red:
(serial-in)-(function node)-(serial-out) ]]
Decoding the modbus input request is easy and the response a little harder
but works.

Ion3, 2021-02-10:
Thank you for your advertisement but I believe this is not the right place.

Ion3, 2021-04-07:
> @Ion3 I'm sorry, but we have no time to do that at the moment. It is on the roadmap 2021.

Hi Klaus, I really do not want to push. I know it is voluntary work and I appreciate your project so much. Are you already able to say something about your release planning for 2021?

biancode, 2021-04-08:
We plan to start with new updates end of Q2 (June) and next end of Q3 (Sept.)

biancode, 2021-04-08:
next major should be ready with next Nodejs LTS in October/November see https://nodejs.org/en/about/releases/

KuBaum, 2021-08-18:
Hello, I am wondering if this feature is already available?

LiquidOcelot29, 2021-11-11:
Hi, is there any chance to update package in this year? Operating as slave device in RTU mode will be very helpful for me... ;)
Best regards

eLabsOy, 2021-12-10:
I would also be interested in slave RTU mode. This would greatly extend the possibilities of this nice package.

americanium, 2021-12-26:
Hello!

I am also waiting for slave RTU mode. An news about that ?

Thanks a lot!

JaagupAverin, 2022-04-11:
Bump!
We may be considering adding the RTU server ourselves, but want to assure first that this isn't already being implemented?
Additionally, what are the caveats that could have been delaying this feature, other than manhours? Thanks.

biancode, 2022-04-11:
@JaagupAverin there is no implementation for now or in development, because we have not enough hours to work on new features, right! Feel free to send your PR! Happy programming!

biancode, 2022-04-21:
see also  #213

closed 2022-09-02 by github-actions[bot]

maga-1, 2022-12-10:
I think, implementing a modbus server for node red is not so simple, since the underlying JS modbus library does not support it. RS485 seems to involve a lot of timing issues and such things, so development would be a bigger endeavour. 

Since I have the same problem as the initial requester, I have programmed a python script that consumes a power value via mqtt and servs it for my Viessmann heat pump via RS485, emulating the power meter mentioned above. The pymodbus library has a fully featured modbus server for RS485. Via mqtt this integrates well with node red.

In case it helps, here is the program: https://github.com/maga-1/mqtt2vitocal

Broadsens, 2024-07-22:
Interested in Modbus RTU slave function too. Surprised that this feature is not implemented.