BiancoRoyal/node-red-contrib-modbus #236: Client unable to reconnect after Modbus Server goes offline and back online

MountainKing91, 2021-02-13:
## Problem
Modbus client (read/write/flex-getter) does not reconnect if modbus server is powered off/disconnected from the network and  then powered on/connected on the network again. I have checked the option "Reconnect on timeout" with the default time of 2000 ms. The only way to reconnect the client is to restart the flows or the entire node-red system. 

Thank you for any advice!

**node-red -v logs:**
13 Feb 15:50:31 - [warn] [modbus-client:Modbus Server] Client -> fsm connected after state init Get More About It By Logging TCP@127.0.0.1:5020 default Unit-Id: 1
13 Feb 15:51:20 - [warn] [modbus-client:Modbus Server] Client -> fsm failed state after sending Get More About It By Logging TCP@127.0.0.1:5020 default Unit-Id: 1
13 Feb 15:51:20 - [warn] [modbus-client:Modbus Server] Client -> fsm broken state after failed Get More About It By Logging TCP@127.0.0.1:5020 default Unit-Id: 1
13 Feb 15:51:20 - [warn] [modbus-client:Modbus Server] Client -> fsm reconnect state after broken Get More About It By Logging TCP@127.0.0.1:5020 default Unit-Id: 1
13 Feb 15:51:22 - [warn] [modbus-client:Modbus Server] Client -> fsm init state after reconnecting TCP@127.0.0.1:5020 default Unit-Id: 1
13 Feb 15:51:22 - [warn] [modbus-client:Modbus Server] Client -> fsm init in 2000 ms TCP@127.0.0.1:5020 default Unit-Id: 1

### Environment
I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it. Version of node-red-contrib-modbus: 5.13.3 (latest). I experienced this issue with various servers and using different devices as client (Windows 10 PC, Raspberry Pi with latest buster release). Node-RED version: 1.2.9 (latest).
    
* **node-red-contrib-modbus version:** v5.13.3

* **nodejs version:** v15.8 for Windows, v12.x for Pi

Zeddy-, 2021-02-26:
Same exact issue, has been going on for a while. Have to stop using this module for now.

MountainKing91, 2021-02-26:
After various tests with differente machines and OS the only way I could solve the issues was to:
1 - unistall nodejs and reinstall version 12.20.2 (which by the way is the version automatically installed with the "official" Pi script from nodered.org)
2 - reinstall node-red-contrib-modbus (latest version) - simply reinstalling the module without touching nodejs didn't work out
Note that upgrading nodejs could break again the modbus nodes. I think that's what could have generated the issue for me so I didn't take the chance again.

seb976, 2021-03-11:
hi,
i have the same problem, but with only 1 modbus connexion (i have 8 other modbus connexion working fine).
i'm running nodered on ubuntu 20.04
I upgraded nodejs to 12.20.3 (10.19.0 before), nodered to 1.2.9 (1.2.7 previously), and reinstall node-red-contrib-modbus but the problem is still here. do you have additionnal informations to communicate about the way you resolve the problem ?
do you have a idea of what is going on ? 
thx

MountainKing91, 2021-03-11:
Hi, Did the 7 other connections work with the previous nodejs version? have you tried to delete the client configuration node and recreate it? If one connection works and the others don't I think you should be able to fix them by cancel the relative configuration nodes and recreate them from zero.

seb976, 2021-03-12:
Yes the connexion work with the previous version, exept the last one, and that is why I try to resolve it by changing the version of nodejs.
Yes i try to delete client configuration  many time and recreate with many configurations. increasing timeout, reconnect timeout , queue delay, uncheck or check boxes but nothing resolve the problem.
All connexion work exept 1. I have deleted all modbus configuration , uninstall node-red-contrib-modbus, reinstall it and reconfigure only the connexion that didn't work and  i always have the problem.
I did some test today with modbus and nodered. i copy/paste my project into a nodered under Windows10 and it's working fine with it. 
An other thing  that i have in ubuntu  is on every flow restart, i have the following :
[image]
this is not happenning on Windows10.
Do you have the same thing ? 
When you test it on W10, do you find a configution where it's working fine ? or the only way you find is  to change nodejs version on your Raspberry.
thx.

MountainKing91, 2021-03-12:
In the screenshot you posted I see logs of the various client operations, 8 different clients as you already said. I think you see the logs in the debug window because you checked the relative option in the client nodes (probably in windows it is not checked). With that option checked you get to see what the finite state machine is doing.
I'm not sure about Ubuntu but in windows10 I have no issues and I am using nodejs 12.20.2 and the latest modbus nodes version. With later versions of nodejs I experienced issues.

sniicker, 2021-03-17:
Same Probleme here - no reconnect!

seb976, 2021-03-17:
You're right @MountainKing91  , my problem is not solved on W10, the logs are just hidden. so I enable log errors and activities on Node configation and I found an othe aspect of my problem : a "PortNotOpenError: Port Not Open" message when i send a request.
After a few days of testing, I found the solution. Nodered is not the cause of this problem, it's come from a PLC which disrupts all modbus communication to another PLC. 
@sniicker, try to activate log errors and activities on your modbus node, maybe the problem is just hidden.

MountainKing91, 2021-03-17:
Glad u guys were able to solve. At this point the bottom line is: do NOT upgrade nodejs past version 12.2.20. I couldn't get it done with the latest nodejs version.

sniicker, 2021-03-18:
@seb976 Yes i also get this msg port not open, but this is the truth. When i unplug the lan cable there is no Port open :) But after i reconnect - the status stays at init. I think not thats because of the port.

biancode, 2021-03-29:
nodejs V14.15.4 is not supported yet

sgorlon, 2021-08-04:
Boa tarde. Eu desenvolvi uma solução temporária para o problema, utilizando os seguintes nós: Catch , Delay, Function e Modbus Flex Connector.

[{"id":"166e9137.258aaf","type":"tab","label":"Siren Act","disabled":false,"info":""},{"id":"49cc8c84.19a0f4","type":"inject","z":"166e9137.258aaf","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"1","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":110,"y":80,"wires":[["9e27f0d3.a686d"]]},{"id":"9e27f0d3.a686d","type":"modbus-getter","z":"166e9137.258aaf","name":"","showStatusActivities":true,"showErrors":true,"logIOActivities":false,"unitid":"1","dataType":"Coil","adr":"5","quantity":"1","server":"9896a43e.0450d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":false,"x":320,"y":80,"wires":[["7e59772f.8888c8"],[]]},{"id":"7c86b41f.80ba1c","type":"inject","z":"166e9137.258aaf","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"10","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":110,"y":160,"wires":[["1461453.ca1e9bb"]]},{"id":"1461453.ca1e9bb","type":"modbus-getter","z":"166e9137.258aaf","name":"","showStatusActivities":true,"showErrors":false,"logIOActivities":false,"unitid":"1","dataType":"HoldingRegister","adr":"4","quantity":"1","server":"9896a43e.0450d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":false,"x":320,"y":160,"wires":[["2f51bb91.9f9c04"],[]]},{"id":"7f9b3178.ed9c9","type":"modbus-flex-connector","z":"166e9137.258aaf","name":"","maxReconnectsPerMinute":4,"emptyQueue":false,"showStatusActivities":false,"showErrors":false,"server":"9896a43e.0450d8","x":890,"y":300,"wires":[[]]},{"id":"77400a24.a29a24","type":"catch","z":"166e9137.258aaf","name":"","scope":["9e27f0d3.a686d"],"uncaught":false,"x":130,"y":300,"wires":[["65bceb85.711204"]]},{"id":"d7d4c279.34d04","type":"modbus-write","z":"166e9137.258aaf","name":"","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"Coil","adr":"3","quantity":"1","server":"9896a43e.0450d8","emptyMsgOnFail":false,"keepMsgProperties":false,"x":880,"y":220,"wires":[[],[]]},{"id":"5f906dda.dfd2f4","type":"eth-ip out","z":"166e9137.258aaf","endpoint":"c51db702.7f4008","variable":"LG1","program":"","name":"","x":910,"y":80,"wires":[]},{"id":"822457c8.c97238","type":"eth-ip in","z":"166e9137.258aaf","endpoint":"c51db702.7f4008","mode":"single","variable":"RTN1","program":"","name":"","x":130,"y":220,"wires":[["8e6c2bad.109c78"]]},{"id":"7e59772f.8888c8","type":"function","z":"166e9137.258aaf","name":"","func":"if(msg.payload[0] == true)\n{\n    msg.payload = 1\n}\nif(msg.payload[0] == false)\n{\n    msg.payload = 0\n}\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":580,"y":80,"wires":[["5f906dda.dfd2f4"]]},{"id":"8e6c2bad.109c78","type":"function","z":"166e9137.258aaf","name":"","func":"if(msg.payload == true)\n{\n    msg.payload = 0\n}\nelse\nif(msg.payload == false)\n{\n    msg.payload = 1\n}\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":580,"y":220,"wires":[["d7d4c279.34d04"]]},{"id":"2f51bb91.9f9c04","type":"eth-ip out","z":"166e9137.258aaf","endpoint":"c51db702.7f4008","variable":"RSSI","program":"","name":"RSSI","x":910,"y":160,"wires":[]},{"id":"65bceb85.711204","type":"delay","z":"166e9137.258aaf","name":"","pauseType":"rate","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"20","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":true,"x":380,"y":300,"wires":[["482badee.aae8d4"]]},{"id":"482badee.aae8d4","type":"function","z":"166e9137.258aaf","name":"","func":"msg.payload = \n{ 'connectorType': 'TCP','unitId': 1 } \nreturn msg","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":640,"y":300,"wires":[["7f9b3178.ed9c9"]]},{"id":"a1ef58fe.777a78","type":"debug","z":"166e9137.258aaf","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":900,"y":380,"wires":[]},{"id":"9896a43e.0450d8","type":"modbus-client","name":"ESP01","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"192.168.1.80","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectOnTimeout":true,"reconnectTimeout":2000,"parallelUnitIdsAllowed":true},{"id":"c51db702.7f4008","type":"eth-ip endpoint","address":"10.1.7.75","slot":"0","cycletime":"500","name":"","vartable":{"":{"LG1":{"type":"BOOL"},"RTN1":{"type":"BOOL"},"RSSI":{"type":"REAL"}}}}]

MountainKing91, 2021-08-04:
Should be fine now, without workarounds.
From node-red-contrib-modbus homepage: Node.JS 14.x is supported with v5.14.+

sgorlon, 2021-08-04:
in my application the problem persists

sgorlon, 2021-08-04:
Node.JS v14.16.1 with npm v6.14.12 and NR v1.3.1, maybe the npm not compatible in my case.

sniicker, 2021-10-05:
I gave it a retry with NR V2.0.6 and JS V14.18.0 but reconnect wont work
so i go back to JS V12.20.2

benfiglin, 2021-10-05:
You mean you gave a try to the PR above? Can you share debug logs?

sniicker, 2021-10-06:
Did i understand anything wrong? Is the PR above not already merged in the actual Version 5.14.1 ?

bcifaulkner, 2021-11-16:
> From what I see, this PR was merged into dev branch. There hasn't been a release yet since then. You'll have to wait for the next release for it to be included.

Do you know if this fix was included in version 5.15 just released? Thanks.

benfiglin, 2021-11-16:
It should be included

ad490, 2021-11-25:
I updated the node to version 5.15.0 and at the moment it works fine and it no longer stays disconnected.
I have:
25 Nov 16:49:39 - [info] Node-RED version: v2.1.3
25 Nov 16:49:39 - [info] Node.js version: v14.18.1

closed 2022-09-01 by github-actions[bot]