BiancoRoyal/node-red-contrib-modbus #237: Using environment variables for connection parameter in modbus client configuration

flodan, 2021-02-18:
**I'm submitting a ...**  (check one with "x")

## Feature
 - (x) feature request
 
 **What is the motivation / use case for changing the behavior?**

I created a subflow that represents a single modbus device. In my main flow, I want to use multiple instances of this modbus device, each of the device having another IP address. Currently, I cannot use a subflow for this because the IP address is fixed inside the subflow itself and cannot be configured, e.g. using an environment variable.

 **Current behavior**
 
The IP address of the the modbus client inside a subflow is fixed and cannot be edited (e.g. using an environment variable).
 
 **Expected behavior**

The connection parameter (IP address, port) of the the modbus client inside a subflow should be editable, e.g. using environment variables.
 
 **Minimal reproduction of the problem with instructions**
 
   1. Create a subflow including a Modbus Flex Getter
   2. Configure the modbus client for the Modbus Flex Getter
   3. The IP address and port cannot be configured using environment variables
   
### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x) Raspbian: Buster 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v14

biancode, 2021-03-29:
Hi @flodan,

you should be able to set variables to the subflow or you could use the flex connector to control the connection settings of the node:

https://nodered.org/docs/user-guide/environment-variables

https://discourse.nodered.org/t/sample-flow-on-how-to-setup-environment-variable-in-a-sub-flow/25341

> Since 0.20, Subflows can be configured with instance properties. These appear as environment variables within the Subflow and can be customised for individual instances of the subflow.
> 
> For example, given a REST API that provides access to different types of record, a subflow could be created to access the API and handle the response, using an environment variable to identify which type of record should be accessed. Individual instances of the Subflow can then be customised for those particular types.

rwrnet, 2021-05-16:
Hi,
it took me a while to find a solution for subflows, so I wanted to share it here.

1. The `modbus-client` connection must be defined and used inside the subflow.

2. An input parameter (e.g. `HOST`) should be defined in the subflow.

3. Inside the subflow, a function node builds the config, using the HOST env 
```
msg.payload = { 
    'connectorType': 'TCP', 
    'tcpHost': env.get("HOST")
}
return msg;
```
and feeded into the `modbus-flex-connector`. If then inside the subflow, this connection needs to be used in an interval activity, the trigger needs to sit inside the subflow (as only one input is available to subflows).

[image]

4. In a flow, this subflow can then be used multiple times, setting separate HOST inputs.

[image]

5. If results (e.g. metrics from modbus devices) needs to be handled, feed them out of the subflow (output).

@biancode if you find this useful, let me know, whether it would make sense to put it in the docs, and if so, where.

biancode, 2021-05-16:
Hi @rwrnet the Wiki should be open for articles here. A Wiki article or blog to link here is the best way for text and you could add an example template via a Pull Request to the master or develop branch and to the [node-red flows library](https://flows.nodered.org/), too.

rwrnet, 2021-05-16:
Makes total sense for the entire flow. Sure sure.

Out of curiosity, how would you use the `modbus-flex-connector` node outside of subflows. I was trying to sequenzialise a flow with using the connector node several times in between, but beside the fact that it is suboptimal anyway, the connector also do not allow for synchronous reaction like "start pulling data, once connection is ready".

biancode, 2021-05-16:
It depends on the use case to answer your last question. The flexible connector has an impact to the client (Modbus-Master) state machine. Each node is effected by the state machine of the client node, too. You get informed from the client if it was not ready to read or write and that's it. The rest is in your hands. Modbus has a wide range of what it controls or represents. The user has to handle the use case of the connection change in detail. The user or flow creator knows if and how the data can or should be handled. The nodes should be a toolbox to create, not a ready to use end-user product ;)

biancode, 2021-05-16:
For me the best way is to be flexible and to use blue-prints and here you can save yours: [Node-RED Flows Modbus](https://flows.nodered.org/search?term=modbus)

anxuae, 2021-12-15:
Hi everybody,

Just to be sure how it's working, what to you put in the "Server" field of the` Modbus Flex Connector` and `Modbus Flex Getter` ?
I put a default configuration which is obviously the same for all instances of my sublow template.
For each instance, I have set an instance parameter with a different IP address. It seems that the ` Modbus Flex Connector` is dynamically editing this default connection config, and thus all sublows instances takes the same IP. It resulting a melting of all read registers from different devices at the output or each subflow.

bernhardkurz, 2021-12-15:
Hello anxuae,
i setup my Modbus Flex Connector in the same way.

I also saw behaviour almost like this, i use the Modbus Flex Connector to read/write Pokeys 57E devices and all looked good. i only had one productive device and some testing devices until now. today i attached a new productive device and found that the register reading from "old" device looked the same as from "new" device.

after some testing i think that somehow all the Flex Connector Instances use latest host setup...

hope someone does have any suggestion how to fix this problem
br bernhard

rwrnet, 2021-12-26:
@anxuae @bernhardkurz 
Are you guys using a subflow as discussed here? 
If so, **in the subflow**, you define a server node (not in the outside flow) and the `Modbus-Flex-Connector` as well as the `Modbus-Getter` are using this server node in the subflow.
To trigger the `Modbus-Flex-Connector` with the dynamic IP of your device, in the subflow you put a one-time trigger (for me it needed a 1sec delay) and cascade it with a function node. Inside the function node, you read the desired IP from an ENV variable like this:
```
msg.payload = { 
    'connectorType': 'TCP', 
    'tcpHost': env.get("HOST")
}
return msg;
```
and this function needs to be cascaded with the `Modbus-Flex-Connector`.
It should looke like this:
[image]

In your outside flow, once you inserted a subflow node, you can then edit its ENV variables, as seen on the following screen:
[image]

That setup works perfectly fine since months now for me. I'm pulling data from 4 inverter instances in parallel without collisions that way.

bernhardkurz, 2021-12-26:
hi, i setup my subflow basically as you describe (of course, as i've started with your flow, thanks for that). Just added my own logic to collect different mb register areas flexible. this is my subflow...
[image]

this is the server setup in modbus flex connector:
[image]

and this is the server setup in modbus flex getter:
[image]

As described i am perfectly able to collect data from the host i setup in ENV variable of subflow. Also when changing host in ENV it collects the data from new host. But when i add new "instance" of this subflow and set different (new) host in specific ENV var, all other instances collect data from latst established instance.

below i have attached my flow, i would appreciate some help.

`[
    {
        "id": "01a2e4156aae3a99",
        "type": "subflow",
        "name": "Pokeys Modbus Connection",
        "info": "",
        "category": "",
        "in": [
            {
                "x": 50,
                "y": 30,
                "wires": []
            }
        ],
        "out": [
            {
                "x": 920,
                "y": 120,
                "wires": [
                    {
                        "id": "9e1678e7c0c86c03",
                        "port": 0
                    }
                ]
            },
            {
                "x": 1040,
                "y": 160,
                "wires": [
                    {
                        "id": "18a36e6a169a62b8",
                        "port": 0
                    }
                ]
            },
            {
                "x": 1040,
                "y": 200,
                "wires": [
                    {
                        "id": "18a36e6a169a62b8",
                        "port": 2
                    }
                ]
            }
        ],
        "env": [
            {
                "name": "rd55io",
                "type": "json",
                "value": "{\"en\":true,\"strt\":0,\"len\":55,\"ts\":0,\"rd\":{\"FC\":1,\"poll\":true,\"interval\":0,\"wrOnChange\":true},\"wr\":{\"FC\":5}}",
                "ui": {
                    "label": {
                        "de": "Read 55 Di / Do"
                    },
                    "type": "input",
                    "opts": {
[... 354 lines cut ...]
        "id": "5876d9d3.3456c8",
        "type": "modbus-client",
        "name": "test",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "tcpHost": "10.0.0.202",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "unit_id": 1,
        "commandDelay": 1,
        "clientTimeout": 1000,
        "reconnectOnTimeout": false,
        "reconnectTimeout": 2000,
        "parallelUnitIdsAllowed": false
    }
]`

rwrnet, 2021-12-26:
In your attached flow, you unfortunately spared out the flow, which uses the subflow. Some differences I spottet:
- for the server node use a non-responding default IP (in my case 127.0.0.1)
- try `Modbus-Getter` instead of `Modbus-Flex-Getter`
- try to minimize your example, e.g. remove the HOST env variable in the subflow definition and only define them on the subflow instances in the main flow. I double checked my setup and its working that way.

bernhardkurz, 2021-12-26:
hi rwrnet,
i tried your suggestions but same result.
i create new Subflow as you suggested very limited function.
just flex connector with standard modbus getter.

i implemented 2 clients. below the main flow (just 2 instances of the subflow)

`[
    {
        "id": "73d11aba5e48f09b",
        "type": "subflow",
        "name": "MBFlex_StdINput",
        "info": "",
        "category": "",
        "in": [],
        "out": [
            {
                "x": 700,
                "y": 160,
                "wires": [
                    {
                        "id": "48bd9af1031388d6",
                        "port": 0
                    }
                ]
            }
        ],
        "env": [
            {
                "name": "HOST",
                "type": "str",
                "value": "10.0.0.105"
            }
        ],
        "meta": {},
        "color": "#DDAA99"
    },
    {
        "id": "a2c4008814fa43c2",
        "type": "inject",
        "z": "73d11aba5e48f09b",
        "name": "Startup Trigger",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": "3",
        "topic": "",
        "payloadType": "date",
        "x": 200,
        "y": 80,
        "wires": [
            [
                "6d3d3f48557719f5"
            ]
        ]
    },
    {
        "id": "6d3d3f48557719f5",
        "type": "function",
        "z": "73d11aba5e48f09b",
        "name": "SetLocalINV",
        "func": "node.warn(env.get(\"HOST\"));\nmsg.payload = { \n    'connectorType': 'TCP', \n    'tcpHost': env.get(\"HOST\"),\n    'unitId':1\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 410,
        "y": 80,
        "wires": [
            [
                "91c8a4bc03c5921e"
            ]
        ]
    },
    {
        "id": "91c8a4bc03c5921e",
        "type": "modbus-flex-connector",
        "z": "73d11aba5e48f09b",
        "name": "Pokeys_Connector",
        "maxReconnectsPerMinute": 4,
        "emptyQueue": false,
        "showStatusActivities": true,
        "showErrors": true,
        "server": "5876d9d3.3456c8",
        "x": 650,
        "y": 80,
        "wires": [
            []
        ]
    },
    {
        "id": "48bd9af1031388d6",
        "type": "modbus-getter",
        "z": "73d11aba5e48f09b",
        "name": "GetSensors",
        "showStatusActivities": true,
        "showErrors": true,
        "logIOActivities": false,
        "unitid": "1",
        "dataType": "HoldingRegister",
        "adr": "400",
        "quantity": "12",
        "server": "5876d9d3.3456c8",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "x": 480,
        "y": 180,
        "wires": [
            [],
            []
        ]
    },
    {
        "id": "af560b5433e9e7c9",
        "type": "inject",
        "z": "73d11aba5e48f09b",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "1",
        "crontab": "",
        "once": true,
        "onceDelay": "3",
        "topic": "",
        "payloadType": "date",
        "x": 190,
        "y": 180,
        "wires": [
            [
                "48bd9af1031388d6"
            ]
        ]
    },
    {
        "id": "5876d9d3.3456c8",
        "type": "modbus-client",
        "name": "test",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": false,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": false
    },
    {
        "id": "1dd75c59506e1098",
        "type": "tab",
        "label": "MB_Flex_stdInput",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "aabed4c9ea564a77",
        "type": "subflow:73d11aba5e48f09b",
        "z": "1dd75c59506e1098",
        "name": "10.0.0.205",
        "env": [
            {
                "name": "HOST",
                "value": "10.0.0.205",
                "type": "str"
            }
        ],
        "x": 760,
        "y": 160,
        "wires": [
            [
                "9db3caeb095d9b5a"
            ]
        ]
    },
    {
        "id": "9db3caeb095d9b5a",
        "type": "debug",
        "z": "1dd75c59506e1098",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 1000,
        "y": 140,
        "wires": []
    },
    {
        "id": "095b8e068fae6946",
        "type": "subflow:73d11aba5e48f09b",
        "z": "1dd75c59506e1098",
        "name": "10.0.0.201",
        "env": [
            {
                "name": "HOST",
                "value": "10.0.0.201",
                "type": "str"
            }
        ],
        "x": 760,
        "y": 240,
        "wires": [
            []
        ]
    }
]`

rwrnet, 2021-12-26:
There is still ein ENV variable directly in the subflow. 

What is the log output from your `SetLocalINV` node saying?

And if I'm not mistaken, your modbus-client (Server) is not defined in the subflow but globally. See excerpt from my example config and note the "z" parameter (which is missing in yours).

```
    {
        "id": "ee92c24f1c328aef",
        "type": "modbus-client",
        "z": "74a9110216f958c5",
        "name": "LocalInv",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": true,
        "queueLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
```

bernhardkurz, 2021-12-27:
@rwrnet 
finally made it work now! thank you very much. it was all about the modbus-client which was defined globally.
So just for record: the flex-getter-node as also the env-var are not problematic.
Another experience i've made, no matter how much client connections are allowed in the server (in my case 4 simultaneously read connections), it is not possible to connect more than one at a time with this kind of setup, but thats fairly ok.

many thanks for your effort, it was a great help

rwrnet, 2021-12-27:
YWC - Lets cherish @biancode for his plugin.

bernhardkurz, 2021-12-28:
definitely many thanks to @biancode as also to the community

biancode, 2022-02-12:
Thanks to all for the great discussion! Maybe that issue could be closed now?

rwrnet, 2022-02-14:
@biancode as it seems to generate confusion, you might think about being more clear about the importance of the place for the server node in the short documentation of the client node. That might prevent further tickets like this. And as @flodan hasn't participated since a year now, it should be safe to close it.

closed 2022-08-31 by github-actions[bot]