BiancoRoyal/node-red-contrib-modbus #238: FC43 / 14 (0x2B / 0x0E) Read Device Identification not supported

GrtVHecke, 2021-03-04:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( x) feature request
 
Since 2006 FC 43 / 14 is present in the Modbus specification. So device manufacturers are meanwhile putting devices on the market where their device info like serial number is retrieved by FC 43 / 14. But the current implementation of the Node-RED nodes does not support it.

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( x) Linux: Debian GNU/Linux 10 (buster) on 64 bit ARM
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : HMP110
     - Version: I can not tell because of the issue with FC 43 / 14
     - Vendor : Vaisala
     - Link   : https://www.vaisala.com/en/products/instruments-sensors-and-other-measurement-devices/instruments-industrial-measurements/hmp110
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v14.15.3

closed 2022-08-31 by github-actions[bot]

Meloman-zz, 2024-07-10:
I need this feature for my Schneider Electric devices.
Any news abot it ?

Meloman-zz, 2024-07-10:
**Additional info for SE users**
Here is a link to documentation about Schneider Electric devices : https://product-help.schneider-electric.com/ED/PowerTag/Smartlink_SIB/EDMS/DOCA0123EN/DOCA0123xx/Details_of_Modbus_Functions/Details_of_Modbus_Functions-5.htm

biancode, 2024-07-10:
Flex-FC can do this and our Leanpub book explains how.